realsenseai/librealsense #13230: Optimal Camera Settings for D455 / D455f Depth Quality @D400 series(D455, D455f)

oliveyoung123, 2024-08-03:
Hello,

I am currently using the D455 and D455f to detect obstacles in an indoor factory environment and am striving to obtain high-quality point clouds. Despite referencing multiple documents and adjusting my camera settings accordingly, I am not satisfied with the quality of the depth frames and point clouds I am obtaining.

Could you advise me on what should be prioritized in terms of camera settings in the following environment? I would greatly appreciate any tips and potential solutions you could offer!

**Setup Environment**:

- Due to computational cost concerns on the PC, the input video resolution is set to 480x270.
- The camera is set up in an indoor factory environment with no natural light and only fluorescent lighting.
- The floor is glossy, causing noise due to reflections from fluorescent lights and other light sources. I need settings to optimize for this.
- Methods to improve depth quality on a perforated plate (600mm x 600mm).
[image]
[image]

**Additional Point Cloud Questions**:
- Currently, the point cloud exhibits a wavy quality. I use the DBSCAN algorithm for clustering the floor and non-floor areas. Are there other clustering algorithms I could try?
- Besides the D455 / D455f, are there any other RealSense models that might be more suitable for the described environment?

Thank you in advance for your assistance!

MartyG-RealSense, 2024-08-03:
Hi @oliveyoung123  The D455f is the best model for this environment, as its filter helps it to negate problems caused by light reflections and repetitive patterns such as floor / ceiling tiles.

For models such as D455 that are not equipped with the filter, it can be purchased separately and attached externally to the front of the camera over the sensors.  The filter used on the D455f is the **CLAREX NIR-75N**.

https://astraproducts.com/info-acrylic-nir-filters.asp

If it is not possible for you to use an external filter then Intel have a guide to negating repetitive patterns at the link below.

https://dev.intelrealsense.com/docs/mitigate-repetitive-pattern-effect-stereo-depth-cameras

************

The optimal depth resolution to use for accuracy is 848x480.  As resolution is reduced, inaccuracy increases.  Rather than reducing resolution to 480x270, instead reducing **FPS** to 15 may help to reduce computational load.

***********

Fluorescent lights can cause disruption to RealSense images due to the heated gas inside them flickering at rates that are difficult to see with the human eye.  The nearer that the operating frequency of the camera is to the operating frequency of the fluorescent lights, the less disruption that will occur.

*************

RealSense has a feature called **Power Line Frequency** that can be configured to a frequency value that best matches the operating frequency in Hz of the lights in your location.  The settings available are 30, 50 and 60.  The 60 Hz frequency is typically used in American regions, whilst 50 Hz is used in European regions. 

***********

If the perforations on the plate are not deep then the whole of the flat top surface will typically have the same depth color because the performations are too shallow to be distinguished as separate elements.  As you can see on your depth image, the main hole that stands out on the image is the deeper central hole.

*************

Point cloud waviness can be reduced if the camera is positioned closer to the objects that it is observing.

Setting the **Default** camera configuration preset can provide cleaner edges and less pointcloud 'spray'.

*********

If you increase the **Laser Power** setting to its maximum value of '360' instead of the default of '156' then the invisible infrared dot pattern projection cast onto objects in the scene by the camera's projector component will be easier for the camera to see.  This should improve the camera's depth analysis of the scene.

******

I do not know of a clustering algorithm that could be recommended for a RealSense camera.  However, using a **domain transform edge preserving filter** may help with the waviness.  RealSense's **Spatial** post-processing filder can provide edge-preserving.

Navifra-Brad, 2024-08-03:
Hello @MartyG-RealSense,

Thank you so much for your detailed response. I have a few additional questions and would appreciate your further assistance.

**Plate Environment and Light Reflection Issues**:
- The plates we are using have perforations, and beneath these perforations, there is an underground space with a depth of 1-3 meters. (The previously attached image was from our test environment, which differs significantly from the actual application environment.) Additionally, the material of the plates causes light reflections, resulting in the depth data not being captured correctly. This leads to holes appearing in the point cloud despite the presence of the plates.
- While achieving perfect depth quality in such an environment is challenging, I am curious if adjusting the 'Power Line Frequency' and 'domain transform edge preserving filter' settings, as you mentioned, would be the best approach in this situation.

**Parameter Adjustment Expertise**:
- Connected to the previous question, there are numerous parameters in the RealSense Viewer that can be adjusted. I am looking to learn how to optimize these parameters based on the environment. Could you provide some tips or best practices for achieving the best quality?
- Additionally, are there any specific recipes for different lighting conditions such as natural light + fluorescent, natural light + incandescent, fluorescent, and incandescent lighting?
 
**Depth Data Holes Issue**:
- When there are holes in the depth data despite the presence of objects, is light reflection the primary cause? I am curious about the extent to which the material affects the depth quality.

Your responses have been incredibly helpful, and I sincerely appreciate it. I also welcome insights from others in the community!

oliveyoung123, 2024-08-03:
Hello @MartyG-RealSense,

Thank you so much for your detailed response. I have a few additional questions and would appreciate your further assistance.

**Plate Environment and Light Reflection Issues**:
- The plates we are using have perforations, and beneath these perforations, there is an underground space with a depth of 1-3 meters. (The previously attached image was from our test environment, which differs significantly from the actual application environment.) Additionally, the material of the plates causes light reflections, resulting in the depth data not being captured correctly. This leads to holes appearing in the point cloud despite the presence of the plates.
- While achieving perfect depth quality in such an environment is challenging, I am curious if adjusting the 'Power Line Frequency' and 'domain transform edge preserving filter' settings, as you mentioned, would be the best approach in this situation.

**Parameter Adjustment Expertise:**
- Connected to the previous question, there are numerous parameters in the RealSense Viewer that can be adjusted. I am looking to learn how to optimize these parameters based on the environment. Could you provide some tips or best practices for achieving the best quality?
- Additionally, are there any specific recipes for different lighting conditions such as natural light + fluorescent, natural light + incandescent, fluorescent, and incandescent lighting?

**Depth Data Holes Issue:**
- When there are holes in the depth data despite the presence of objects, is light reflection the primary cause? I am curious about the extent to which the material affects the depth quality.

Your responses have been incredibly helpful, and I sincerely appreciate it. I also welcome insights from others in the community!

MartyG-RealSense, 2024-08-03:
You are very welcome.  I'm pleased that I could help.  :)

Metal reflections can be dampened by covering a reflective surface with a fine powder such as foot powder or baby powder.  There are are professional reflection damping aerosol sprays available.  

If you cannot apply something to the plate then another method of dampening glare from reflections is to attach a thin-film **linear polarization filter** over the lenses on the outside of the camera.  Because any polarization filter will work as long as it is linear (except for the round ones in 3D glasses), they can be purchased inexpensively from stores such as Amazon by searching them for the term 'linear polarizing filter sheet'.

Using power line frequency and a Spatial filter can improve the depth image quality, but applying powder / aerosol or using a linear polarization filter will have the most effect on dealing with reflections specifically.

Defining a **Region of Interest (ROI)** on the lower half of the image can also help to deal with strong light or strong reflections from the light cast onto a suface.  See point 3 of the link below for more information about configuring a ROI.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#make-sure-image-is-properly-exposed

Using a Visual Preset is the easiest way to configure multiple settings, as it stores all the settings in a file and automatically applies all the settings simultaneously when the Visual Preset is activated.  If you want to improve overall depth image quality and accuracy and not just clean up the pointcloud then the **Medium Density** preset will be a recommendable choice as it provides a good balance between accuracy and the amount of detail on the depth image (less gaps / holes).

I note on the depth image that the walls of the room are mostly missing.  This could be because the plain wall and strong light from the window is making the projected infrared dot pattern harder for the camera to see.  Increasing the Laser Power value to its maximum of 360 should make it easier for the camera to see the dot pattern that is cast onto the walls and use the dots to aid depth analysis.

Fluorescent is the main light type that needs compensating for with the Power Line Frequency because of how the gas-based nature of fluorescent lights makes them flicker and cause depth noise.  Other types of light should be fine.

Light reflections make sufaces more difficult for the camera to read.  Black and dark gray surfaces are also difficult to read because they absorb light and so appear on the depth images as plain black areas with no depth.  

Different types of material can affect how visible the dot pattern projection is on the surface of a material, and the visibility of the pattern may become stronger or weaker depending on the lighting level during certain times of day (though this is not such a problem with an indoor artificial lighting location).

The dot pattern projection is not absolutely required though if the scene has strong light, as the camera can alternatively use any type of visible light in the scene to aid depth analysis.

oliveyoung123, 2024-08-03:
Based on the advice provided so far, I will conduct some tests and then proceed with further questions!

Thank you sincerely.

I would also appreciate any additional advice from others!

MartyG-RealSense, 2024-08-03:
I look forward to your next update.  Good luck!

closed 2024-08-23 by MartyG-RealSense