realsenseai/librealsense #13238: What is threshold_filter

1005088h, 2024-08-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |   Ubuntu 22  | 
| Kernel Version (Linux Only)    | 5.15                             | 
| Platform                                 | PC  |
| SDK Version                            | 2.55.1                       | 
| Language                            |  C              | 
| Segment			|  Robot }                          | 

### Issue Description

There are very less information about threshold_filter.  https://dev.intelrealsense.com/docs/post-processing-filters does not mention it.
Is there any difference between threshold_filter and clip distance.  clip distance will still contain the noise cause by the background. Does threshold_filter remove the noise from  background?

MartyG-RealSense, 2024-08-06:
Hi @1005088h  The threshold filter and clip distance are similar. Clip distance defines a maximum distance in meters from the camera at which depth data will be rendered, with depth values greater than that value being excluded from the depth image.

The threshold filter allows both a maximum **and** a minimum distance to be defined.  For example, if there was an object close to the camera at 20 cm away but you did not want it to be included on the depth image then you could set a minimum threshold filter distance of 0.25 meters (equal to 25 cm) so that depth values lower than 0.25 meters are excluded from the depth image.

The threshold filter does not affect noise on the image.  A **temporal** filter would have a beneficial effect on depth noise reduction.

MartyG-RealSense, 2024-08-13:
Hi @1005088h  Do you require further assistance with this case, please?  Thanks!

1005088h, 2024-08-14:
> Hi @1005088h Do you require further assistance with this case, please? Thanks!

thanks, let's close for now

closed 2024-08-14 by 1005088h