realsenseai/librealsense #13242: d435i motion module not working. error message: messenger-libusb.cpp:42 - control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

topyihao, 2024-08-07:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|244222073637|
|**Firmware Version**|5.16.0.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|302643060088|
|**Firmware Update Id**|302643060088|
|**Dfu Device Path**||

Please provide a description of the problem

MartyG-RealSense, 2024-08-07:
Hi @topyihao  Are you accessing the Motion Module (IMU) through the RealSense Viewer tool, or through a programming script such as Python or C++ please?

topyihao, 2024-08-07:
> Hi @topyihao Are you accessing the Motion Module (IMU) through the RealSense Viewer tool, or through a programming script such as Python or C++ please?

Hi MartyG, I'm accessing the motion module (IMU) through the RealSense viewer tool.

MartyG-RealSense, 2024-08-07:
Do you have JetPack 6 installed on your Jetson Nano?  If you are then when using JP6, a **MIPI driver** at the link below needs to be installed to add support for the IMU on RealSense camera models equipped with an IMU.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

If JetPack 6 is downgraded to JetPack 5.1 then the IMU should work.

If you are unable to downgrade from JetPack 6 to 5.1 then a workaround that enables you to keep JP6 is to build the librealsense SDK from source code using the **libuvc backend** procedure's 'libuvc_installation.sh' build script.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

topyihao, 2024-08-09:
I'm using jetpack 4.6. 

I tried https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

I still got same error and the IMU still not working.

topyihao, 2024-08-09:
when I follow this instruction: 
https://dev.intelrealsense.com/docs/nvidia-jetson-tx2-installation
the rgb camera is working, the IMU is not working, giving me error: 
messenger-libusb.cpp:42 - control_transfer returned error. 

when I follow this instruction: 
https://jetsonhacks.com/2019/12/22/install-realsense-camera-in-5-minutes-jetson-nano/
and choose to build from source
The rgb camera is not working, IMU is working with error message: 
ERROR [547096600944] (synthetic-stream.cpp:58) Exception was thrown during callback: aMalloc failed status

MartyG-RealSense, 2024-08-09:
Are you using Ubuntu 18.04, please?  That is the Ubuntu version that should be used with JetPack 4.6.

topyihao, 2024-08-12:
> Are you using Ubuntu 18.04, please? That is the Ubuntu version that should be used with JetPack 4.6.

I think I'm using Ubuntu 20.04 and the jetpack is 4.6. Is that the reason why I got this kind of issue?

We make the color sensor and IMU work together by using the SDK v2.50.0. But the data from the gyro looks wired.

MartyG-RealSense, 2024-08-12:
I cannot say for sure that your problem is related to use of 20.04.  However, Nvidia advises to use JetPack 4.6 only with Ubuntu 16.04 or 18.04, and use JetPack 5 for 20.04.

https://forums.developer.nvidia.com/t/frustrations-with-the-ecosystem-looking-for-clarification-jetpack-sdk-version/258836/2

https://forums.developer.nvidia.com/t/how-to-install-jetpack-4-6-2-using-ubuntu-20-04/218051

topyihao, 2024-08-12:
Thanks for your help. I'll try the correct version.

closed 2024-08-26 by MartyG-RealSense