realsenseai/librealsense #13246: USB port power instability with D435 Camera

ky047, 2024-08-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 5.15.1.0 | 
| Operating System & Version |   {Win (8.1/10) /  | 
| Kernel Version (Linux Only)    |       2.54.2                             | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {}                          | 
| Segment			|  others             | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello
I'd like to ask you a few questions about the D435 camera
When I connect the D435 camera to the PC cable, I get a warning message
MESSAGE

**/USB port power instability
Unknown USB devices require more power than the port can supply /**

If you need a picture, please let me know BUT It's in Korean, 

I found the same symptoms when I connected it to a different cable and a different PC
Is there any way we can take action?

Genozen, 2024-08-08:
You probably already tried it, but make sure you are using the cable that's USB 3.0+   or USB-C to USB-C.
I had this problem when plugged into USB 2.0

ky047, 2024-08-08:
> 아마 이미 시도해 보셨겠지만, USB 3.0+ 또는 USB-C to USB-C 케이블을 사용하고 있는지 확인하세요. 저는 USB 2.0에 꽂았을 때 이 문제가 있었습니다.

> You probably already tried it, but make sure you are using the cable that's USB 3.0+ or USB-C to USB-C. I had this problem when plugged into USB 2.0

Thank you for your answer. 
I connected through USB 3.0 cable.. 
Even if I connect through another 3.0 cable, I get the same symptoms

MartyG-RealSense, 2024-08-08:
Hi @ky047  Are you using the official 1 meter USB cable supplied with the camera or a cable of your own choice, please?

It is important to use a high quality cable that is designed for data transfer.  A cable with quality problems or one designed only for charging devices instead of transferring data could have stability problems.

It sounds though as if the problem is more related to the USB port than the cable.  Which computer / computing device are you using with the camera, please?  For example, small single board computing devices may have a USB port that cannot supply all of the power that the camera needs to operate correctly.  

In those situations, attaching the camera to a mains electricity powered USB 3 hub instead of plugging it directly into the computing board's USB port can help the camera to work, as the mains power socket is able to provide the camera with enough power to work.

ky047, 2024-08-08:
> 안녕@ky047 카메라와 함께 제공된 공식 1m USB 케이블을 사용하고 계신가요, 아니면 본인이 원하는 케이블을 사용하고 계신가요?
> 
> 데이터 전송을 위해 설계된 고품질 케이블을 사용하는 것이 중요합니다. 품질 문제가 있는 케이블이나 데이터 전송 대신 장치 충전 전용으로 설계된 케이블은 안정성 문제가 있을 수 있습니다.
> 
> 하지만 문제는 케이블보다는 USB 포트와 더 관련이 있는 것 같습니다. 카메라와 함께 어떤 컴퓨터/컴퓨팅 장치를 사용하고 계신가요? 예를 들어, 작은 단일 보드 컴퓨팅 장치에는 카메라가 제대로 작동하는 데 필요한 모든 전원을 공급할 수 없는 USB 포트가 있을 수 있습니다.
> 
> 이런 상황에서는 카메라를 컴퓨팅 보드의 USB 포트에 직접 연결하는 대신, 주 전원 소켓에서 카메라가 작동하는 데 필요한 충분한 전력을 공급받을 수 있는 USB 3 허브에 연결하면 카메라 작동에 도움이 될 수 있습니다.

Thank you for your answer
First of all, I connected it through the regular 3.0 USB sold in the market
When the cable is connected to another camera (D435), an error message does not appear
So, it is judged that there is no problem with the cable
I tried connecting through the laptop USB port, but I couldn't through the hub I'll check it out and let you know

MartyG-RealSense, 2024-08-08:
Thanks very much, @ky047 - I look forward to your next report.  Good luck!

ky047, 2024-08-11:
> Thanks very much, @ky047 - I look forward to your next report. Good luck!
Hello, I'll let you know the test results
When I connected the camera through the hub
I don't get a message that I'm connected
Perhaps the camera is suspected to be out of order, is there more cases I can check?

ky047, 2024-08-11:
> 정말 고맙습니다.@ky047- 다음 보고서를 기대하겠습니다. 행운을 빌어요!

Hello, I'll let you know the test results
When I connected the camera through the hub
I don't get a message that I'm connected
Perhaps the camera is suspected to be out of order, is there more cases I can check?

MartyG-RealSense, 2024-08-12:
As you are using Windows, please try launching the Windows **Camera** app.  You can find this by typing 'camera' into the text box at the bottom of the Windows screen.  When it has opened, click the 'change camera' button in the top corner of the Camera app to change to using your D435 camera.

Is the RGB color from the D435 able to be displayed?

[image]

If the Windows Camera app can display the RGB from the D435 then it could indicate that the problem is with the RealSense software program rather than a fault with the camera.

***********

Please also check the camera privacy settings in Windows to ensure that permission for applications to access the camera is set to On.  Instructions for doing so are at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

ky047, 2024-08-12:
> Windows를 사용 중이므로 Windows **카메라** 앱을 실행해 보세요. Windows 화면 하단의 텍스트 상자에 '카메라'를 입력하면 찾을 수 있습니다. 열리면 카메라 앱의 상단 모서리에 있는 '카메라 변경' 버튼을 클릭하여 D435 카메라를 사용하도록 변경하세요.
> 
> D435의 RGB 색상을 표시할 수 있나요?
> 
> ![영상](https://private-user-images.githubusercontent.com/41145062/356968126-00cfaa89-cfa1-4599-a0b4-066bf250d8be.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM0NTE1MzIsIm5iZiI6MTcyMzQ1MTIzMiwicGF0aCI6Ii80MTE0NTA2Mi8zNTY5NjgxMjYtMDBjZmFhODktY2ZhMS00NTk5LWEwYjQtMDY2YmYyNTBkOGJlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODEyVDA4MjcxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAzYjgwNDUwNWM1ZTZjOWUxZjE5NGY5ZDA5ZmU3MGVjNjNjODYzZGI1ZWUxMTM5Y2FjMWM3MGUzNTQ2NjhmMzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.HxQr9l-6E_5sZV7dbZrpTbKhy3Va3r-fN2AYrdLL9uM)

Thank you for your answer
Isn't it only possible if the D435 camera is connected to the USB port?
The D435 camera is not currently connected to the USB port
An unstable power port message occurs
D435 camera is not displayed in Device Manager

MartyG-RealSense, 2024-08-12:
If the D435 camera is not currently connected to the USB port then it would not be displayed in the Device Manager.  It only appears in the Device Manager when inserted into the USB port and disappears when unplugged from the port.

When the camera is detected, there should be two RealSense drivers (Depth and RGB) listed under the 'Cameras' category of the Device Manager.  'Cameras' might be called something in Korean, like 카메라

If there is a problem with these drivers then the D435, when it is inserted, might appear under the 'Universal Serial Bus controllers' category - 범용 직렬 버스 컨트롤러 - as an Unknown USB Device

[image]

ky047, 2024-08-12:
> D435 카메라가 현재 USB 포트에 연결되어 있지 않으면 장치 관리자에 표시되지 않습니다. USB 포트에 삽입했을 때만 장치 관리자에 나타나고 포트에서 분리하면 사라집니다.
> 
> 카메라가 감지되면 장치 관리자의 '카메라' 범주에 두 개의 RealSense 드라이버(Depth 및 RGB)가 나열되어야 합니다. '카메라'는 한국어로 카메라와 같이 불릴 수 있습니다.
> 
> 이러한 드라이버에 문제가 있는 경우 D435를 삽입하면 '범용 직렬 버스 컨트롤러' 범주에 알 수 없는 USB 장치로 나타날 수 있습니다.
> 
> ![영상](https://private-user-images.githubusercontent.com/41145062/357002798-6d78b58e-b93a-47b6-8c65-f98bfcf1bdd7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM1MDQzOTMsIm5iZiI6MTcyMzUwNDA5MywicGF0aCI6Ii80MTE0NTA2Mi8zNTcwMDI3OTgtNmQ3OGI1OGUtYjkzYS00N2I2LThjNjUtZjk4YmZjZjFiZGQ3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODEyVDIzMDgxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQzYzY0ZDVjMTBhNjc1YzQxYjBlZjAwODRmMjg2ZDA1OTBjYTc3YmFmZDVjNmY0NWZhNjNmYjE3YmVmMzgzNzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bdX7HbgfivUyXQmZ_QTsrD5_P4cOQS3bEV_BkPN9EIY)

Device Manager does not display the device when you connect the D435 camera to USB
Just in case it's a USB port problem, if i connect another camera, it'll connect normally
It's not a USB port issue
Nothing appears in Device Manager Even i connect the USB
What I suspect is a faulty camera, what do you think?

(This is the device manager picture when the camera is connected to the USB port)
[image]

MartyG-RealSense, 2024-08-13:
Sometimes a particular camera will have problems with a specific computer.  For example, I have a D455 that is always detected on one Windows laptop but is often not detected on another Windows laptop when other RealSense cameras work fine on that computer.

Right now though that D455 is working normally on the affected laptop.  This could suggest that the issue is related to Windows rather than a problem with the camera or the RealSense SDK.  If the camera is not detected in Device Manager then it is not an SDK issue since installing the SDK is not required for the camera and its drivers to be detected.

Let's focus on the message that you reported, 'USB port power instability: Unknown USB devices require more power than the port can supply'.  If a particular USB device's power needs are not being met by the USB port on the computer for some reason then a way to test if it is indeed an insufficient power issue would be to obtain a mains electricity powered USB 3 hub and plug the camera into the hub instead of the computer.  Mains powered hubs have greater power stability than the ports on computers because their USB ports draw their power from the mains plug socket instead of from the power supply of the computer.

ky047, 2024-08-13:
> 때때로 특정 카메라는 특정 컴퓨터에서 문제가 발생합니다. 예를 들어, 저는 D455를 가지고 있는데, 한 Windows 노트북에서는 항상 감지되지만 다른 Windows 노트북에서는 다른 RealSense 카메라가 그 컴퓨터에서 잘 작동하는 경우 감지되지 않는 경우가 많습니다.
> 
> 하지만 지금은 D455가 영향을 받은 노트북에서 정상적으로 작동하고 있습니다. 이는 문제가 카메라나 RealSense SDK의 문제가 아니라 Windows와 관련이 있음을 시사할 수 있습니다. 장치 관리자에서 카메라가 감지되지 않으면 SDK 문제가 아닙니다. SDK를 설치하지 않아도 카메라와 해당 드라이버를 감지할 수 있기 때문입니다.
> 
> 보고하신 'USB 포트 전원 불안정: 알 수 없는 USB 기기가 포트에서 공급할 수 있는 것보다 더 많은 전원을 필요로 합니다'라는 메시지에 집중해 보겠습니다. 어떤 이유에서인지 특정 USB 기기의 전원 요구 사항이 컴퓨터의 USB 포트에서 충족되지 않는다면 실제로 전원 부족 문제인지 테스트하는 한 가지 방법은 주 전원으로 구동되는 USB 3 허브를 구해서 카메라를 컴퓨터가 아닌 허브에 꽂는 것입니다. 주 전원으로 구동되는 허브는 컴퓨터의 포트보다 전원 안정성이 더 높습니다. USB 포트가 컴퓨터의 전원 공급 장치가 아닌 주 전원 플러그 소켓에서 전원을 공급받기 때문입니다.

Thank you for your answer. 
Unfortunately, the same symptoms appear when i connect the camera through a powered hub 
Even other cameras connect normally to the hub and USB ports

ky047, 2024-08-18:
> 때때로 특정 카메라는 특정 컴퓨터에서 문제가 발생합니다. 예를 들어, 저는 D455를 가지고 있는데, 한 Windows 노트북에서는 항상 감지되지만 다른 Windows 노트북에서는 다른 RealSense 카메라가 그 컴퓨터에서 잘 작동하는 경우 감지되지 않는 경우가 많습니다.
>
> 하지만 지금은 D455가 영향을 받은 노트북에서 정상적으로 작동하고 [...]

Is there anything else I need to check?
Please reply as soon as possible.

MartyG-RealSense, 2024-08-19:
You said at the beginning of this issue:

> When I connect the D435 camera to the PC cable, I get a warning message
> MESSAGE
> /USB port power instability
> Unknown USB devices require more power than the port can supply /

Do you mean that the error happened when you inserted the cable into the small USB port on the side of the camera?

If the error happened when inserting into the camera's USB port then it might indicate that there is a hardware problem with the camera's port.  If there is a USB port problem on the camera then it would not be able to be repaired, unfortunately.  

If the camera was purchased less than a year ago then you could enquire to the retailer that you purchased it from about returning it.

ky047, 2024-08-19:
> 당신은 이 호의 시작 부분에서 이렇게 말했습니다:
> 
> > D435 카메라를 PC 케이블에 연결하면 경고 메시지 
> > MESSAGE 
> > /USB 포트 전원 불안정 
> > 알 수 없는 USB 장치가 포트에서 공급할 수 있는 것보다 많은 전원을 필요로 합니다./
> 
> 카메라 옆면에 있는 작은 USB 포트에 케이블을 꽂았을 때 오류가 발생했다는 뜻인가요?
> 
> 카메라의 USB 포트에 삽입할 때 오류가 발생한 경우 카메라 포트에 하드웨어 문제가 있음을 나타낼 수 있습니다. 카메라의 USB 포트에 문제가 있는 경우 불행히도 수리할 수 없습니다.
> 
> 카메라를 구매한 지 1년이 지나지 않았다면, 카메라를 구매한 소매점에 반품에 관해 문의하세요.

I think it's right to say that the camera is dead
Is there any way I can recover my camera with software or hardware?

MartyG-RealSense, 2024-08-19:
If the camera's USB port has a hardware fault then there is no recovery method available.

The only options would be to:

1.  Enquire to the retailer that you purchased the camera from about an 'RMA return' if the camera was purchased less than a year ago.  Or

2. Consider purchasing a replacement 'Vision Processor D4 V1' circuit board, opening up the camera's casing and swapping the old D4 V1 board with the new one.

ky047, 2024-08-19:
> 카메라의 USB 포트에 하드웨어 오류가 발생한 경우 유일한 옵션은 다음과 같습니다.
> 
> 1. 카메라를 구매한 지 1년이 지나지 않았다면 카메라를 구매한 소매업체에 'RMA 반품'에 대해 문의하세요. 또는
> 2. '비전 프로세서 D4 V1' 회로 기판을 교체용으로 구매하고, 카메라 케이스를 열고 오래된 D4 V1 보드를 새 보드로 교체하는 것을 고려해보세요.

I see
Do you happen to have a manual or reference to disassemble and replace the D4 V1 board?
May I know the new D4 V1 price as well?

MartyG-RealSense, 2024-08-20:
There is not an official "teardown" disassembly reference, and most RealSense teardown guides created by others cost a lot of money.  Below are the only free ones that I could find.  The second link is Chinese but the web browser should offer to translate the page when it is visited.

https://www.robotsforroboticists.com/intel-realsense-d435-teardown-and-cleaning/

https://ek21.com/news/tech/17657/

Price will depend on where the D4 V1 is purchased from.  Its part code is 952019.  As an example of prices from Intel authorized distributors, Rutronik24 offers it for 48 euros whilst DigiKey stock it for the similar price of £53.79 British pounds.

https://www.rutronik24.com/product/intel/952019/10164045.html

https://www.digikey.co.uk/en/products/detail/intel-realsense/82635DSASMDLPRQ/9926010

ky047, 2024-08-20:
> 공식적인 "해체" 분해 참조는 없으며, 다른 사람이 만든 대부분의 RealSense 분해 가이드는 많은 비용이 듭니다. 아래는 내가 찾을 수 있는 유일한 무료 가이드입니다. 두 번째 링크는 중국어이지만 웹 브라우저는 방문 시 페이지를 번역하도록 제공해야 합니다.
> 
> https://www.robotsforroboticists.com/intel-realsense-d435-teardown-and-cleaning/
> 
> https://ek21.com/news/tech/17657/
> 
> 가격은 D4 V1을 어디에서 구매하느냐에 따라 달라집니다. 부품 코드는 952019입니다. Intel 공인 유통업체의 가격의 예로, Rutronik24는 48유로에 제공하는 반면 DigiKey는 비슷한 가격인 53.79파운드(GBP)에 재고로 보유하고 있습니다.
> 
> https://www.rutronik24.com/product/intel/952019/10164045.html
> 
> https://www.digikey.co.uk/en/products/detail/intel-realsense/82635DSASMDLPRQ/9926010
> 
> 화폐 가치에 대한 대략적인 기준으로 1파운드는 한국 원화로 약 ₩1,730, 1유로는 약 ₩1,475에 해당합니다.

I think it helped me a lot thanks to you
I think we need to check with the customer whether the camera is replaced or not
I think you can close the case

MartyG-RealSense, 2024-08-20:
You are very welcome.  I'm pleased that I could help.  As you suggested, I will close this case.  Thanks again!

closed 2024-08-20 by MartyG-RealSense