realsenseai/librealsense #13247: Advanced mode isn't required to set the visual preset?

sidy3d, 2024-08-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |  Windows 11  | 
| Platform                                 | PC |
| SDK Version                            |  Intel® RealSense™ SDK 2.0 (v2.55.1) }                          | 
| Language                            |  C++                           | 
| Segment			|  other                           | 

### Issue Description
Hi,

I am currently using the sensor API as documented in the [rs-sensor-control](https://dev.intelrealsense.com/docs/rs-sensor-control) example to set the `RS2_OPTION_VISUAL_PRESET` option:
```
context.query_devices()[0]->query_sensors()[0] // for brevity first sensor is the stereo module
sensor->set_option(RS2_OPTION_VISUAL_PRESET, RS2_RS400_VISUAL_PRESET_HIGH_ACCURACY);
```
Using this method there are no errors reported in the RS2 log which suggest that the option was successfully applied.

#### Questions:
* The issue [here](https://github.com/IntelRealSense/librealsense/issues/3710#issuecomment-481676831) says that advanced mode needs to be enabled before setting the visual preset. Is this still the case? I'm a bit confused as my code seems to be working.
* Are the preset values listed here: [https://dev.intelrealsense.com/docs/d400-series-visual-presets](https://dev.intelrealsense.com/docs/d400-series-visual-presets) and the presets listed here [rs2_rs400_visual_preset](https://github.com/IntelRealSense/librealsense/blob/e1688cc318457f7dd57abcdbedd3398062db3009/include/librealsense2/h/rs_option.h#L196) using the same values? Does this mean if I am using a built-in preset with no modifications I can set the `RS2_OPTION_VISUAL_PRESET` option instead of loading JSON file using advanced mode?

Thanks,

MartyG-RealSense, 2024-08-08:
Hi @sidy3d  Advanced Mode is enabled by default in the camera hardware when the camera is purchased, so you do not need to enable it in code.

Yes, you can set built-in presets in code instead of loading a json file with Advanced Mode.  The link below has a guide for doing so in C++.

https://www.intel.com/content/www/us/en/support/articles/000028416/emerging-technologies/intel-realsense-technology.html

I checked the Advanced Mode values that the built-in presets apply against the values stored in the equivalent downloadable json files on the Visual Presets web-page (for example, Medium Density vs MedDensityPreset.json), and they do seem to set the same values for options when applied.

sidy3d, 2024-08-08:
Hi Marty,

I appreciate your response. Then it seems that the code in that link and my code do the same thing so I guess I can close the issue.

Thanks,

closed 2024-08-08 by sidy3d

MartyG-RealSense, 2024-08-08:
You are very welcome.  I'm pleased that I could help.  :)