realsenseai/librealsense #13248: Why do I get a bunch of 0's？ QAQ

CCCCCCCT, 2024-08-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515| 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   {Win (10)   | 
| Kernel Version (Linux Only)   
| Platform                                 | PC |
| SDK Version                            | 2.50.0                     | 
| Language                            |  Python                     | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Hello, I'm currently developing a function to obtain the depth values of keypoints using the L515 and OpenPose. I've noticed that although the coordinates are correct, my program often gives me depth values of 0, and these frequent 0 values prevent me from proceeding with the next steps.

I’ve carefully reviewed the part of the code where OpenPose captures the keypoints, and it seems correct. However, the L515 frequently returns a depth value of 0.

I would like to ask for advice: Is there something wrong with my code or L515?

Here is the output
PersonID 0: Depth value at coordinates (x=291, y=253): 0.0000 m
PersonID 0: Depth value at coordinates (x=374, y=231): 0.0000 m
PersonID 0: Depth value at coordinates (x=381, y=231): 0.0000 m
PersonID 0: Depth value at coordinates (x=384, y=236): 0.0000 m
PersonID 0: Depth value at coordinates (x=374, y=236): 0.0000 m
PersonID 0: Depth value at coordinates (x=376, y=243): 3.2178 m
PersonID 0: Depth value at coordinates (x=376, y=243): 0.0000 m
PersonID 0: Depth value at coordinates (x=376, y=241): 3.1893 m
PersonID 0: Depth value at coordinates (x=374, y=238): 0.0000 m
PersonID 0: Depth value at coordinates (x=371, y=236): 0.0000 m
PersonID 0: Depth value at coordinates (x=371, y=233): 0.0000 m
PersonID 0: Depth value at coordinates (x=371, y=231): 0.0000 m
PersonID 0: Depth value at coordinates (x=369, y=231): 0.0000 m
PersonID 0: Depth value at coordinates (x=366, y=231): 0.0000 m

```python
import cv2
import sys
import os
import pyrealsense2 as rs
import numpy as np
import bin.pyopenpose as op

def get_depth_value(depth_frame, x, y, personid):
    if not depth_frame:
        print("No depth frame available")
        return None

    depth_value = depth_frame.get_distance(int(x), int(y))
    print("PersonID {}: Depth value at coordinates (x={}, y={}): {:.4f} m".format(personid ,int(x), int(y), depth_value))
    return depth_value

def main():
    try:
        # Openpose參數
        params = dict()
        params["model_folder"] = "./models/"
        params["net_resolution"] = "256x256"  # 以準確率為代價提升處理速度
        params["model_pose"] = "BODY_25"

        # 初始化OpenPose
        opWrapper = op.WrapperPython()
        opWrapper.configure(params)
        opWrapper.start()

        # 打開相機，連接L515
        cap = cv2.VideoCapture(0,cv2.CAP_DSHOW)
        if not cap.isOpened():
            print("Error: Unable to open camera")
            return
        
        # 初始化RealSense pipeline
        # 做一次就好
        pipeline = rs.pipeline()
        config = rs.config()
        config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30)
        # config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

        pipeline.start(config)
        
        # device = profile.get_device()
        # depth_sensor = device.first_depth_sensor()
        # device.hardware_reset()
        
        while True:
            ret, frame = cap.read()
            if not ret:
                break

            #L515拿深度
            frames = pipeline.wait_for_frames()
            depth_frame = frames.get_depth_frame()
            
            #openpose開始處理
            datum = op.Datum()
            datum.cvInputData = frame #cvInputData 是 OpenPose 用於接收輸入影像的屬性。
            datums = op.VectorDatum([datum])
            opWrapper.emplaceAndPop(datums)

            keypoints = datum.poseKeypoints #取得姿態估計的結果，即關鍵點資訊。
            if keypoints is not None:
                for person_id, person in enumerate(keypoints):
                    i = 8  #髖關節
                    x, y, confidence = person[i]
                    if confidence > 0.1:
                        #output_point(i, person_id, x, y, frame)
                        get_depth_value(depth_frame, x, y,person_id)
                    
            pose_frame = datum.cvOutputData
            frame = cv2.addWeighted(frame, 0.7, pose_frame, 0.3, 0)
            #顯示
            cv2.imshow("OpenPose Camera", frame)

            if cv2.waitKey(1) & 0xFF == ord('q'):
                break

        #release
        cap.release()
        pipeline.stop()
        cv2.destroyAllWindows()

    except Exception as e:
        print(f"Error: {e}")
        sys.exit(-1)

if __name__ == "__main__":
    main()

```

CCCCCCCT, 2024-08-08:
Sometimes it can detect the depth value, but most of the time it fails to detect it.

MartyG-RealSense, 2024-08-08:
Hi @CCCCCCCT  Would it be possible to provide an RGB color image of the scene that the L515 camera is observing in order to see what elements in the scene could be making it difficult for the camera to read depth information from surfaces?

For example, if the surface that is being analyzed is the color black then it will be seen as an area of empty depth (zeroes) unless a strong light source is cast onto that area, as described at https://github.com/IntelRealSense/librealsense/issues/6757

CCCCCCCT, 2024-08-09:
[image]

The results below the image are generated by my code, and it should have captured my coordinates correctly.

CCCCCCCT, 2024-08-09:
[image]
Sometimes it detects the coordinates, but sometimes it doesn't."

MartyG-RealSense, 2024-08-09:
The RGB color images in the situations where the depth is detected and when it is not detected are very similar, so it does not seem to be a case of the detection being disrupted by elements in the scene.

It also appears that the body's skeleton landmark joints are being detected correctly even on the image where depth is not detected.

I notice that in the image where distance detection is successful you are standing near to the center of the camera view, whilst in the image where distance values are zero you are standing at the far side of the camera view.  Being off to the side could make depth detection more difficult for the camera.  I recommend trying again whilst in the center of the camera view and see if you have consistent success.

CCCCCCCT, 2024-08-10:
Thank you for your help. I'll try again.

CCCCCCCT, 2024-08-14:
Hello, I followed your suggestion and changed that line of code to the one you specified. However, I was already using that line when running my code until I came across some articles and changed it to config_rs.enable_stream(rs.stream.depth). But even after changing it back to what you specified, I'm still encountering the issue where the depth value shows up as 0. Perhaps I can try modifying my code to avoid this problem.

CCCCCCCT, 2024-08-14:
When I use config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30), I notice that the distance in front of the camera always remains around 4.7 meters, whether I move forward or backward, it stays around 4.6 to 4.7 meters. However, when I move to the left side of the screen, the distance starts to decrease, but when I move to the right side, the distance still remains around 4.6 to 4.7 meters. This is clearly incorrect.
[image]
[image]

CCCCCCCT, 2024-08-14:
[image]
[image]
I also noticed that with config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30) and config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30), at the same position, I can get depth values in some cases, but not in others.

In the bottom right corner of the screen, I can't get depth values with 640x480, but with 1024x768, I can get them, although they are incorrect. However, they are detected.

MartyG-RealSense, 2024-08-14:
Although the L515 camera can depth-sense up to 9 meters, it is best suited to use at 4 meters distance and below.

Are you able to test the camera in the RealSense Viewer tool?  If you enable the **Stereo Module** (the depth stream) and move the mouse cursor over a coordinate on the pointcloud, the distance of that point in meters is displayed at the bottom of the Viewer window.  If depth is measured correctly in the Viewer then that will validate whether or not the depth problem is in your project rather than an issue with the camera.

CCCCCCCT, 2024-08-15:
I tried using the RealSense Viewer tool with the resolution set to 640x480, and it was able to detect the depth. Maybe there is an issue with my code. QAQ

MartyG-RealSense, 2024-08-15:
It will be difficult to diagnose the code, given the few number of RealSense OpenPose projects to refer to.

It may be worth focusing on the depth value printing line first.

`print("PersonID {}: Depth value at coordinates (x={}, y={}): {:.4f} m".format(personid ,int(x), int(y), depth_value))
`
What function does the **{:.4f}** part of the above line perform, please?  Might it be preventing the depth value from updating correctly?

You could also try disabling the code that deals with zeroes to see whether it could be breaking something.

CCCCCCCT, 2024-08-16:
It is a function to output depth values, and it only takes four decimal places.

[image]

I discovered a new issue: I found that the coordinates from the images detected by OpenPose do not align with those detected by the L515. So, I want to try aligning them before performing detection.

Additionally, I found that there is an error when I try to specify the size of my color_frame.
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
"Error: frame didn't arrive within 5000"

CCCCCCCT, 2024-08-16:
I found out that the issue was indeed because my depth image and RGB image were not properly aligned. I've successfully resolved it. Thank you! ╰(*°▽°*)╯

```python
import cv2
import sys
import os
import pyrealsense2 as rs
import numpy as np
import bin.pyopenpose as op

 
def get_depth_value(depth_image, x, y,person_id):
    #座標在圖像裡面
    if x < 0 or y < 0 or x >= depth_image.shape[1] or y >= depth_image.shape[0]:
        print("Coordinates out of bounds")
        return None

    depth_value = depth_image[y, x]
    
    if depth_value == 0:
        print(f"Person_id:{person_id},Depth value at coordinates (x={x}, y={y}): 0 m (no depth information)")
    else:
        depth_in_meters = depth_value * 0.00025  #轉成公尺
        print(f"Person_id:{person_id},Depth value at coordinates (x={x}, y={y}): {depth_in_meters:.4f} m")

    return depth_value

def main():
    try:

        # 初始化RealSense pipeline
        pipeline = rs.pipeline()
        config = rs.config()
        config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
        config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

        
        align_to = rs.stream.color
        align = rs.align(align_to)
        
        pipeline.start(config)
        #datum = op.Datum()
        
        
        # 初始化OpenPose
        params = dict()
        params["model_folder"] = "./models/"
        params["net_resolution"] = "256x256"
        params["model_pose"] = "BODY_25"

        opWrapper = op.WrapperPython()
        opWrapper.configure(params)
        opWrapper.start()
        
        while True:

            # L515拿深度
            frames = pipeline.wait_for_frames()
            
            
            aligned_frames = align.process(frames) 
            
            aligned_depth_frame = aligned_frames.get_depth_frame()
            aligned_color_frame = aligned_frames.get_color_frame()

            if not aligned_depth_frame or not aligned_color_frame:
                continue
            
            color_image = np.asanyarray(aligned_color_frame.get_data())
            depth_image = np.asanyarray(aligned_depth_frame.get_data())
            
            
            #depth_image = cv2.applyColorMap(cv2.convertScaleAbs(depth_image,alpha=0.03),cv2.COLORMAP_JET)
            
            
            datum = op.Datum()
            datum.cvInputData = color_image
            datums = op.VectorDatum([datum])
            opWrapper.emplaceAndPop(datums)
            
            keypoints = datum.poseKeypoints
            
            if keypoints is not None:
                #for person in keypoints:
                for person_id, person in enumerate(keypoints):
                    #i = 0
                    nose_x, nose_y = int(person[8][0]), int(person[8][1])
                    #x,y,confidence = person[i][0]
                    #print("x:",nose_x,",y:",nose_y)
                    get_depth_value(depth_image, nose_x, nose_y,person_id)
                    
            
            #blend_image = cv2.addWeighted(color_image,0.7,depth_image,0.5,0)
            
            #images = np.hstack((color_image,depth_image))
            
            
            pose_frame = datum.cvOutputData
            frame = cv2.addWeighted(color_image, 0.7, pose_frame, 0.3, 0)
            
            cv2.imshow('window',frame)
            if cv2.waitKey(1) & 0xFF == ord('q'):
                break
            
            cv2.waitKey(1)

        # Release resources
        
    finally:
        pipeline.stop()

if __name__ == "__main__":
    main()

``` 

[image]

MartyG-RealSense, 2024-08-16:
It's great to hear that you succeeded.  Thanks very much for the update and for sharing your solution!

CCCCCCCT, 2024-08-19:
[image]
Hello, I would like to test longer distances now. I want to ask, why can't it capture the far end of the wall even though the distance isn't very far?

MartyG-RealSense, 2024-08-19:
The RealSense Viewer has a post-processing filter called **Threshold Filter** enabled by default that limits the depth image to 4 meters.  If you go to 'Stereo Module > Post' Processing, expand open the list of filters and click on the blue icon (On) beside Threshold Filter to turn the icon to black (Off) then the camera will be able to display depth to its full distance range.

MartyG-RealSense, 2024-08-19:
As you have light entering the room through the windows, please try changing the Preset at the top of the Viewer side-panel from 'Max Range' to 'Low Ambient Light' to see if it makes a difference.

CCCCCCCT, 2024-08-19:
I would like to ask if it's possible to use Python to enable this mode?

MartyG-RealSense, 2024-08-19:
To set the preset in your Python project, you could try the L515 script provided at https://github.com/IntelRealSense/librealsense/issues/8619 which sets the built-in preset to the Low Ambient Light preset.

`depth_sensor.set_option(rs.option.visual_preset, 3) # 5 is short range, 3 is low ambient light`

CCCCCCCT, 2024-08-19:
Thank you very much for your assistance. I will try again.

closed 2024-09-01 by MartyG-RealSense