realsenseai/librealsense #13249: Query on compressed image camera ros2 topics

girishsmurthyIntel, 2024-08-08:
Hi @MartyG-RealSense , I would need to generate the compressed image ros2 topics. Would suggest any ros2 plugin for the same or create a node to subscribe to the image topics & compress independently using the cv2bridge?

girishsmurthyIntel, 2024-08-08:
Thank you. Can you kindly let know the steps to be followed after installing the plugin for a specific ROS2 build?

MartyG-RealSense, 2024-08-08:
After installing the plugin, simply perform the rs_launch.py launch process and then check the list of published topics with the command **ros2 topic list** to confirm that the compressed topics are being published.

https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#published-topics

girishsmurthyIntel, 2024-08-09:
Dear @MartyG-RealSense , I wanted to update you that for ROS2 foxy, we need to use the **sudo apt install ros-humble-compressed-image-transport** to get the compressed streams. I understand that foxy is EOL. But for some application we were using the same and this was required.
Between, I had another question, about how these compressed topics are generated from the plugin? The camera is connected via USB3.0 cable/port to the NUC. Here we have the realsense ros package which creates the raw image/ camera info topics using the data from the camera. Will the image-transport plugin subscribe to these raw images and generate the compressed topics? I mean can we have only the ros2 compressed topics in the ros2 topic list and not the raw images?

MartyG-RealSense, 2024-08-10:
The compressed topics plugin for a specific ROS version should ideally be installed.  For Foxy that would be:

`sudo apt install ros-foxy-compressed-image-transport`

However, Foxy is only supported by the RealSense ROS2 wrapper up until version 4.51.1, which should be used with librealsense 2.51.1. 

An explanation of how image_transport works can be found at the link below, which states that "image_transport itself provides only "raw" transport so as not to impose unnecessary dependencies on client packages".

https://wiki.ros.org/image_transport

It is possible to **unsubscribe** from specific topics during runtime after launch has completed so that they are not included in the topic list.  Usually though the raw topics are active when publishing compressed topics.  

https://answers.ros.org/question/409553/how-can-i-remove-active-topic/

MartyG-RealSense, 2024-08-17:
Hi @girishsmurthyIntel  Do you require further assistance with this case, please?  Thanks!

girishsmurthyIntel, 2024-08-19:
Hi @MartyG-RealSense I was able to get the compressed topic working. Appreciate your kind support.

MartyG-RealSense, 2024-08-19:
You are very welcome, @girishsmurthyIntel - I'm pleased that I could help.  As you were able to get the compressed topic, I will close this issue.  Thanks very much for the update!

closed 2024-08-19 by MartyG-RealSense

Atharva-05, 2024-12-05:
> Hi @MartyG-RealSense I was able to get the compressed topic working. Appreciate your kind support.

Hello! @girishsmurthyIntel
Could you please let me know how you got the compressed topic working, and the details of the machine you are running your system on? Could you also please specify the librealsense version and realsense ros wrapper version that you used? 
Thanks!

girishsmurthyIntel, 2024-12-05:
We had to install the sudo apt install ros-foxy-compressed-image-transport (for ROS2 foxy) and when you launch the rs_launch.py, we already start to get the compressed topics. You can check this by running ros2 topic list on another terminal.  The libRealSense supported version: v2.51.1  We had built the realsense ros2 workspace from the commit 86ab0c8358214ae9a526395396c1619f9488a195 (HEAD -> ros2-development, origin/ros2-development, origin/HEAD)

Atharva-05, 2024-12-05:
Thank you for your prompt reply! @girishsmurthyIntel