realsenseai/librealsense #13251: pybackend build will always fail

gfx73, 2024-08-09:
I tried to build python bindings with 2.55.1 release. However for now it is [only possible](https://github.com/IntelRealSense/librealsense/pull/13079) if pybackend is built. 

At the same time pybackend won't build in 2.55.1, it also won't on dev branch. Look at [this](https://github.com/IntelRealSense/librealsense/blob/e1688cc318457f7dd57abcdbedd3398062db3009/wrappers/python/pybackend.cpp#L17). And also notice there is no `options.h` in `src/core`.

I guess for now there is no workaround to build python bindings with latest release.

MartyG-RealSense, 2024-08-10:
Hi @gfx73  As the pybackend fix was merged into the **development** branch of librealsense on June 26 2024, would it be possible for you to build from the development branch instead of the master branch until the next master SDK release (which is planned for around the end of September 2024).

https://github.com/IntelRealSense/librealsense/tree/development

gfx73, 2024-08-10:
Hi @MartyG-RealSense! Thank you for your reply. 

What is the fix are you talking about? Could you please provide a PR link? Because up to my understanding pybackend is not build able with current dev (I provided the evidence in the issues description). Basically `core/options.h` does not exist.

The PR that I mentioned fixes pyrealsense build, but not pybackend. To be honest I don't know why one would build pybackend. But if there is such functionality, it should work.

Regarding building from dev branch. How can I get tar.gz asset for it? I am dealing with docker build, and I need this.

Thank you for your attention!

MartyG-RealSense, 2024-08-10:
The fix that I am referring to is PR https://github.com/IntelRealSense/librealsense/pull/13079 where changes were made to the CMakeLists.txt file (highlighted in green in the image below) and then the updated CMakeLists.txt file was saved into the 'development' version of the librealsense SDK.

[image]

So if the development version of librealsense is built then it will include these CMakeLists.txt file changes that address the pybackend issue, so that you do not have to make CMakeList.txt edits yourself as eufrizz suggested to you today at https://github.com/IntelRealSense/librealsense/pull/13079#issuecomment-2278859539

When it is time for the next master 'latest release' of librealsense to come out, the development branch becomes the new master branch version.

The edited CMakeLists.txt adds an 'else' statement to deal with pybackend installation.

target_include_directories(pybackend2 PRIVATE ../../src)
**else() # not BUILD_LEGACY_PYBACKEND**

On the development branch of librealsense's GitHub page, clicking on the green CODE button reveals a download button for a zip file, but not a tar.gz.  You could possibly download the zip and extract the librealsense source code folder from it, and then manually re-compress the folder as a tar.gz.

[image]

gfx73, 2024-08-10:
@MartyG-RealSense the issues that I want to highlight is that if one would try to build pybackend, he would not be able to. I agree that pyrealsense is now build able. The issue is about pybackend

MartyG-RealSense, 2024-08-10:
I note that in the PR discussion you mention at https://github.com/IntelRealSense/librealsense/pull/13079#issuecomment-2278413109 that you are setting the CMake build flag **-DBUILD_LEGACY_PYBACKEND=true**.  This flag is not usually used when building the pyrealsense2 wrapper from source.  Normally, only **-DBUILD_PYTHON_BINDINGS:bool=true** is needed, plus **-DPYTHON_EXECUTABLE** if you have both Python 2 and Python 3 installed on the same computer.

gfx73, 2024-08-10:
Please look [here](https://github.com/IntelRealSense/librealsense/blob/e1688cc318457f7dd57abcdbedd3398062db3009/wrappers/python/pybackend.cpp#L17) again. This line of code will always fail. I just want to highlight this for realsense team. 

If pybackend is never supposed to be built, it should at least be mentioned somewhere.

MartyG-RealSense, 2024-08-10:
The best place to discuss this issue is likely to be on the PR where RealSense developers (maloel and nir_az) are so that they can comment on the problem that you are having.

gfx73, 2024-08-10:
Thank you @MartyG-RealSense

MartyG-RealSense, 2024-08-17:
Case closed due to discussion being moved to PR.

closed 2024-08-17 by MartyG-RealSense