realsenseai/librealsense #13257: RealSense R200 Camera Not Detected on Windows 10

tangyipeng100, 2024-08-13:
I have a RealSense R200 camera, and when I connect it to my Windows 10 machine, it appears in the Device Manager like this:
[image]

I'm not entirely sure about the exact model of the camera, but here's a photo of it:
[image]

I tried using the following Python code to access the camera streams, but it returns "No device found":
```python
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()

config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

pipeline.start(config)
```
Additionally, the RealSense Viewer v2.34.0 is unable to connect to the camera.

Could you please advise on how I can successfully access the camera streams?

MartyG-RealSense, 2024-08-13:
Hi @tangyipeng100  The RealSense SDK 2.0 (which 2.34.0 is a version of) is not compatible with the RealSense R200 camera unfortunately, as this camera model dates from 2015.  You can use the camera in Windows in the 'librealsense1' SDK or in a Windows SDK called '2016 R2'.  

pyrealsense2 will also not work with R200.  The original 'pyrealsense' should be used with it.

https://github.com/toinsson/pyrealsense

More information about the R200-compatible SDKs can be found in a recent discussion at https://github.com/IntelRealSense/librealsense/issues/13244

tangyipeng100, 2024-08-13:
Thank you for your response. I would like to ask if the 'librealsense1' SDK supports Android? Or whether the R200 support Android?

tangyipeng100, 2024-08-13:
By the way, when I try to install `pyrealsense`, I encounter the following error:

```
(test) C:\Users\51745>pip install pyrealsense
Collecting pyrealsense
  Using cached pyrealsense-2.2.tar.gz (60 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: numpy in c:\users\51745\.conda\envs\test\lib\site-packages (from pyrealsense) (1.24.4)
Requirement already satisfied: cython in c:\users\51745\.conda\envs\test\lib\site-packages (from pyrealsense) (3.0.11)
Collecting pycparser
  Downloading pycparser-2.22-py3-none-any.whl (117 kB)
     ---------------------------------------- 117.6/117.6 kB 1.1 MB/s eta 0:00:00
Collecting six
  Downloading six-1.16.0-py2.py3-none-any.whl (11 kB)
Building wheels for collected packages: pyrealsense
  Building wheel for pyrealsense (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [4 lines of output]
      rsutilwrapper.cpp
      C:\Users\51745\.conda\envs\test\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(14) : Warning Msg: Using deprecated NumPy API, disable it with #define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
      pyrealsense/rsutilwrapper.cpp(1278): fatal error C1083: Cannot open include file: 'rs.h': No such file or directory
      error: command 'D:\\install software\\vs2022\\VC\\Tools\\MSVC\\14.30.30705\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyrealsense
  Running setup.py clean for pyrealsense
Failed to build pyrealsense
Installing collected packages: six, pycparser, pyrealsense
  Running setup.py install for pyrealsense ... error
  error: subprocess-exited-with-error

  × Running setup.py install for pyrealsense did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      C:\Users\51745\.conda\envs\test\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      rsutilwrapper.cpp
      C:\Users\51745\.conda\envs\test\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(14) : Warning Msg: Using deprecated NumPy API, disable it with #define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
      pyrealsense/rsutilwrapper.cpp(1278): fatal error C1083: Cannot open include file: 'rs.h': No such file or directory
      error: command 'D:\\install software\\vs2022\\VC\\Tools\\MSVC\\14.30.30705\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> pyrealsense

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

What could be the cause of this error, and how can I resolve it?

MartyG-RealSense, 2024-08-13:
librealsense1 did not have built-in Android support.

In 2015 the R200 was used in an Android smartphone project with a tool called **Google Project Tango**.

https://www.tomshardware.com/news/project-tango-realsense-smartphone-prototype,29915.html

Project Tango was cancelled by Google though and replaced with **ARCore**.

https://developers.google.com/ar

So whilst it is technologically possible to use R200 technology with Android, I am not aware of any ready-made software solutions that you could use in the present day to interface an R200 with Android.

MartyG-RealSense, 2024-08-13:
Regarding your pyrealsense installation issue: I have no knowledge of this subject due to the age of the original pyrealsense, but a discussion at the link below provides advice about the error that you encountered.

https://stackoverflow.com/questions/73257839/setup-py-install-is-deprecated-warning-shows-up-every-time-i-open-a-terminal-i

MartyG-RealSense, 2024-08-23:
Hi @tangyipeng100  Do you require further assistance with this case, please?  Thanks!

tangyipeng100, 2024-08-23:
> Hi @tangyipeng100 Do you require further assistance with this case, please? Thanks!

Thank you for your kind review. After further consideration, I realized that the R200 is somewhat outdated, so I have decided to switch to the D435 instead.

closed 2024-08-23 by tangyipeng100