realsenseai/librealsense #13260: Multiple camera connection / What do I need to setup a multiple camera environment?

kickserver, 2024-08-14:
Hello, I bought a Realsense D435i and now consider to buy two more. 

I want to setup a synchronized multicamera configuration.

I've alread read the guide in the following link,  but not sure how many connectors to buy. 
[https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration]

Could you please give me a guide how many connectors and housing do I have to buy?
I am thinking of 3-camera system to take a 3 photos simultaneously. 

Thank you in advance.

MartyG-RealSense, 2024-08-14:
Hi @kickserver  There is information at the link below about working about how many sync cable components you need.

https://support.intelrealsense.com/hc/en-us/community/posts/360050894574/comments/360013337454

********

If you are able to use C++ code though then you might not need sync cabling for the photo capture application that you have in mind.  https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468 has a C++ script for capturing an image simultaneously from all attached cameras.

The original script was designed to check if the camera model was a D415 and prevent capture if it was not, so you will need to edit this line of the code to change 'D415' to 'D435i' or remove the camera name check from the script.

`if (strcmp(dev.get_info(RS2_CAMERA_INFO_NAME), "Intel RealSense D415") == 0) // Check for compatibility`

MartyG-RealSense, 2024-08-26:
Hi @kickserver  Do you require further assistance with this case, please?  Thanks!

kickserver, 2024-08-26:
Hi MarthG, 

Two new D435i cameras just delivered, so I have 3 cameras now. 

Thanks to your information, I am thinking of using C++ code to get the simultaneous images. 

My concern is even I do not need sync cabling, but cameras should be attached. 

What do I need to attach the cameras?

Thank you for your help.

MartyG-RealSense, 2024-08-26:
Yes, a USB 3 hub would be fine.  I would recommend a USB 3.1 or 3.2 hub rather than USB 3.0.  There are USB 3,0 hubs available if there is a reason why you need USB 3.0 specifically, otherwise go for 3.1 or 3.2.

Amazon's **Amazon Basics** own-brand hubs are highly reliable with RealSense cameras.

You just plug the cameras into the hub and plug the hub into a USB port on the computer.  If you are using a C++ program that is designed for automatic detection of multiple cameras like the simultaneous capture one that I highlighted earlier then the cameras attached to the hub will be detected when the program is run.

kickserver, 2024-08-27:
Hi @MartyG-RealSense,

Thank you for your answer. I just tested the C++ code and got 2 RGB images.

I have few more questions. 

In the code of #2219 , what does "9.json" file do? I erased this lines to run the code.
And, is there another code that I can get depth images together with RGB images?

MartyG-RealSense, 2024-08-27:
It looks as though 9.json is simply the name of a custom camera configuration preset file that the script's creator was using to import and apply a list of values for camera settings.  It would not be required in order for the script to work successfully.

The SDK C++ example **rs-save-to-disk** demonstrates saving both depth and RGB to .PNG file.

https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk

kickserver, 2024-09-04:
@MartyG-RealSense 
Thank you for your help!

closed 2024-09-04 by kickserver

MartyG-RealSense, 2024-09-04:
You are very welcome, @kickserver - I'm pleased that I could help.  Thanks very much for the update!