realsenseai/librealsense #13263: Permission denied when attempting to activate the motion module

radu-iordache-ADI, 2024-08-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455  | 
| Firmware Version                   | (5.16.0.1) | 
| Operating System & Version |  Linux (Ubuntu 22.04)   | 
| Kernel Version (Linux Only)    |  (6.5.0-44-generic)                                         | 
| Platform                                 | PC |
| SDK Version                            |   SDK 2.0 (v2.55.1.0)     | 
| Language                            |  python                          | 
| Segment			|  Robot                        | 

### Issue Description
I am using ROS2 to launch a node that is used to start up my RealSense 455. Everything works perfectly if I don't pass `enable_gyro: True` and  `enable_accel: True` as the parameters. The moment I pass one of those, the launch fails with this error: 

**Backend in rs2_open_multiple(sensor:0x747d3801df20,
profiles:0x747d383eab88, count:2):
Failed to open scan_element
/sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.5/0003:8086:0B5C.0096/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_x_en
Last Error: Permission denied**

The same error i get if I try to activate the Motion Module inside Realsense Viewer.

Worth mentioning that I am start the node inside a Docker container where i install the realsense sdk as such:

```
# install realsense sdk 
RUN apt-get install -y software-properties-common && apt update
RUN apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
RUN add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo focal main" -u
RUN apt-get install -y \
    librealsense2-dkms \
    librealsense2-utils \
    librealsense2-dev \
    librealsense2-dbg
```
This node worked flawlessly until last Thursday but since Monday I've been getting this issue. I haven't made any significant updates or changes in between.

radu-iordache-ADI, 2024-08-15:
> Can you confirm please if you are using a PC or an Nvidia Jetson computing board?

I am indeed using a PC to connect to the Realsense camera through a Docker container, due to using a ROS distribution that's not compatible to Ubuntu 22.04

MartyG-RealSense, 2024-08-15:
If you run Docker with a 'docker run' command, are you launching it in sudo admin permissions mode with **sudo docker run**

Does running Docker in sudo mode make any difference to your Motion Module access problem?

radu-iordache-ADI, 2024-08-15:
Unfortunately, I am running docker using the docker extension that I have installed on VScode. As far as I am aware, the container already runs under root by default, right?

MartyG-RealSense, 2024-08-16:
My knowledge of Docker is admittedly limited.

The **Failed to open scan_element** error can be caused by a problem with the **udev** device handling rules, such as the rules not having been installed, as advised at https://github.com/IntelRealSense/librealsense/issues/3242#issuecomment-464775017 and https://github.com/IntelRealSense/librealsense/issues/5049#issuecomment-542175378

MartyG-RealSense, 2024-08-30:
Hi @radu-iordache-ADI Do you require further assistance with this case, please?  Thanks!

radu-iordache-ADI, 2024-09-02:
Hi @MartyG-RealSense ,

I don't, I wish I knew what's causing the problem, but as of now, everything I've tried has had no success.

MartyG-RealSense, 2024-09-02:
As the IMU worked until recently and everything tried has had no success, there may be something hidden on your computer that is preventing it from working.   Removing librealsense and the wrapper and then reinstalling them may therefore help, unless the hidden problem is in Ubuntu.

The ideal would be to wipe everything including Ubuntu and reinstall the computer from the beginning in order to be certain that hidden issues have been eliminated, though this may not be practical if you would have to reinstall Docker too.

MartyG-RealSense, 2024-09-16:
Case closed due to no further comments received.

closed 2024-09-16 by MartyG-RealSense

JeffMag10, 2025-01-08:
I am using a Nvidia Jetson computing board and I am try to access the IMU features, but am getting this error - 

Traceback (most recent call last):
  File "realsense_imu_test.py", line 88, in <module>
    cam_util_console()
  File "realsense_imu_test.py", line 17, in cam_util_console
    p, conf = initialize_camera()
  File "realsense_imu_test.py", line 12, in initialize_camera
    p.start(conf)
RuntimeError: Failed to set frequency 2. device path: /sys/devices/platform/3610000.xhci/usb2/2-3/2-3.1/2-3.1:1.5/0003:8086:0B5C.0005/HID-SENSOR-200076.3.auto/iio:device1/in_anglvel_sampling_frequency Last Error: Permission denied

I believe I need to reinstall librealsense, is that correct?

Jeff

MartyG-RealSense, 2025-01-09:
Hi @JeffMag10  A RealSense user at https://github.com/IntelRealSense/realsense-ros/issues/2982 who experienced the **Failed to set frequency** error was able to resolve it by their particular case by reinstalling librealsense and installing the **udev** device handling rules.  It is not clear whether it was the SDK reinstall or the rules reinstall that fixed it, so it would be worth trying the easier of the two first - reinstalling the udev rules with the command below.

`sudo apt-get install librealsense2-udev-rules`

JeffMag10, 2025-01-27:
Thanks, I will try this.