realsenseai/librealsense #13264: D430 module not detected by realsense viewer

mprat, 2024-08-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |  Raspbian   | 
| Kernel Version (Linux Only)    | 6.1.21-v8+                                     | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            |  2.55.1                          | 
| Language                            |  C                           | 
| Segment			|  Robot                          | 

### Issue Description

Is there any documentation available for bringing up a D430 module using librealsense? When opening `realsense-viewer`, I see "device detected" but then the device does not appear in the sidebar as I am expecting. The error I see is below. Of note, the camera module is connected through a USB 2.0 interface, but my understanding is that this should not affect the ability of the viewer to see the camera. I was able to update the firmware to the latest firmware using `rs-fw-update` no problem, but I cannot find any information on error messages I am seeing below  online:

```
librealsense version: 2.55.1
ds-active-common.cpp:74 - Projector capacity is overrided and disabled by FW
Device PID = 0xad6, Capabilities Vector = []
context.cpp:116 - ... 1-1.2-5
context.cpp:129 - Found 1 RealSense devices (0xfe requested & 0xff from device-mask in settings)
rs.cpp:237 - [rs2_get_option( options:0xec50f820, option_id:28 ) Invalid Value] hwmon command 0x2c( 9 0 0 0 ) failed (response -9= No expected user action)
rs.cpp:237 - [rs2_get_option_range( options:0xec50f820, option:28, min:-1.27913e+33, max:-1.2792e+33, step:-3.12296e+24, def:-3.12296e+24 ) Invalid Value] hwmon command 0x2c( 9 1 0 0 ) failed (response -9= No expected user action)
rs.cpp:3106 - Couldn't refresh devices - hwmon command 0x2c( 9 1 0 0 ) failed (response -9= No expected user action)
```

MartyG-RealSense, 2024-08-15:
Hi @mprat  Which version of the Vision Processor D4 Board are you using with your D430 module, please?  A D430 without an RGB sensor should be used with the **V1** version of the D4 board.

 
The other versions of the D4 board are:

V2: used in D435i cameras with the D430 module and an RGB sensor

V3: Used in D455 cameras with the D450 module

V4: Used in D405 cameras with the D401 module

V5: Used in D457 cameras with the D450 module

mprat, 2024-08-15:
Another thing I'm noticing is that I would expect based on the datasheet for a D430 module to have a device ID of `0x0AD4`, but I'm seeing the device ID of `0x0AD6`, and can't find mention of that in the D400 datasheet.

mprat, 2024-08-15:
Additionally, the image formats I'm seeing from USB 2.0 don't match the documentation in the datasheet. I clearly have something physically wrong with the setup, but I'm not sure how I would know which piece is in the incorrect piece!

[image]

MartyG-RealSense, 2024-08-15:
If your D4 board has a sticker on it, the V1 version should have the code number **J32139-120**

[image]

I see that your serial number is listed by rs-enumerate-devices as **ffffffffffff**.  This is a certain sign of a problem.  In an example of this issue at https://github.com/IntelRealSense/librealsense/issues/7337#issuecomment-693120103 the RealSense user in that case identified a bent connector on the USB cable as the cause of their particular serial number problem.

The PID **0AD6** tends to appear when there is a problem with identifying the camera name, like the Intel RealSense USB2 case (like your camera ID) at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360033237894/comments/360008292933

In regard to rs-enumerate-devices, the Device Info comes from the D4 board.  The supported profiles comes from querying the sensor components, which are the left and right IR sensors on the D430 module in this case.  The RGB sensor would also be queried if one was present.

The firmware driver identifies the camera and provides the appropriate camera name based on the components that the firmware is able to detect.  For example, a D430 with a D4 V2 board would identify as D430i because D4 V2 had an IMU component but there was not an RGB sensor present (so could not be identified as D435i because of the lack of RGB).

mprat, 2024-08-15:
Thanks for the quick response - we'll confirm that the D4 board is in fact a V1, and do some checking on the cables and see if that resolves our issue.

MartyG-RealSense, 2024-08-15:
You are very welcome.  If the D4 board is a V1, it is worth checking the interposer (the join between the two boards) in particular.  If you constructed your own interposer cable then check the wiring, and if it is a pre-made one purchased from the RealSense Store then confirm that both ends of the cable are fully seated in the 50-pin connectors on the boards.  

On the D4 side of the connection, the cable needs to be in the connector on the upper edge and not the one on the side, which is designed for a tracking module or RGB sensor and not a depth module.

[image]

MartyG-RealSense, 2024-08-23:
Hi @mprat  Do you have an update about this case that you can provide, please?  Thanks!

mprat, 2024-08-25:
> Hi @mprat Do you have an update about this case that you can provide, please? Thanks!

This ended up being a cable (or two) that were not properly seeded! Thank you for the quick reply; I'll close the issue since it has been resolved.

closed 2024-08-25 by mprat

MartyG-RealSense, 2024-08-25:
It's great to hear that you found the cause of your problem.  Thanks very much for the update!