realsenseai/librealsense #13269: Issue using realsense-color stream on Raspberry Pi 4

hshorty, 2024-08-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D456 | 
| Firmware Version                   | 5.16.0.1| 
| Operating System & Version | Linux (Ubuntu 22.04) | 
| Kernel Version (Linux Only)    |  5.15.0-1060-raspi    | 
| Platform                                 |Raspberry Pi4 B 4GB  |
| SDK Version                            |  2.55.1.0
| Language                            | python                         | 
| Segment			|  Robot                    | 

### Issue Description

Issue using realsense-color stream on Raspberry Pi 4

#### Installation
I followed the instructions for installing `librealsense` as outlined in the [libuvc installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md).
(also tried: https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide)

- `realsense-viewer` runs fine.

#### Working Examples

- `rs-distance` example works correctly.
- Python example `python-tutorial-1-depth.py` works fine.

#### Non-Working Examples

1. **`rs-capture` Example:**
   - Error:
     ```
     16/08 22:59:29,271 ERROR [281473513734368] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success
     16/08 22:59:49,781 ERROR [281473513734368] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: Success
     16/08 22:59:52,447 ERROR [281473382478048] (d400-thermal-monitor.cpp:97) Error during thermal compensation handling: get_xu(id=15) failed! Last Error: Success
     ```

2. **Python Example `opencv_viewer_example.py`:**
   - Error:
     ```
     frames = pipeline.wait_for_frames()
     RuntimeError: Frame didn't arrive within 5000
     ```

#### Specific Observations
- The depth stream works fine on its own.
- Enabling the color stream (`config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)`) causes the following error:

```
RuntimeError: Frame didn't arrive within 5000
```

- I have tried various configurations for the color stream, 
- changed the timeout settings
but none have resolved the issue.

MartyG-RealSense, 2024-08-16:
Hi @hshorty  The D455-type camera models (D456 is essentally a D455 with a protective casing) support a feature called 'RGB from left infrared sensor', where an RGB image can be obtained from the left-side IR sensor instead of the RGB sensor.  Can you test whether an RGB image is displayed if you edit the config line to access RGB via the IR sensor, please?

`config.enable_stream(rs.stream.infrared, 640, 480, rs.format.bgr8, 30`

hshorty, 2024-08-17:
Hi @MartyG-RealSense, 
Thank you very much for you help. Once I edited the config line to access RGB via IR sensor,  it works.

MartyG-RealSense, 2024-08-17:
You are very welcome, @hshorty - I'm pleased to hear that it worked for you!

hshorty, 2024-08-17:
Hi @MartyG-RealSense, thanks again! Is there any way we can access the RGB sensor, or is this a limitation when using a Pi 4?

MartyG-RealSense, 2024-08-17:
RealSense cameras typically have problems with Pi boards, unfortunately.  Usually the best that Pi can provide is basic depth and color streams without any extras such as alignment and post-processing, and sometimes the dedicated RGB sensor will not stream frames and RGB From Left Infrared has to be used instead if the camera model supports it.

hshorty, 2024-08-17:
@MartyG-RealSense, thank you so much for your help and the detailed explanation.

MartyG-RealSense, 2024-08-23:
Hi @hshorty  Do you require further assistance with this case, please?  Thanks!

hshorty, 2024-08-26:
Hi @MartyG-RealSense, 

No, the issue was sorted with your assistance.

closed 2024-08-26 by hshorty

MartyG-RealSense, 2024-08-26:
Thanks very much for the update.  I'm pleased that I could help!

amrit7321, 2024-09-10:
@MartyG-RealSense as per above discussion, depth is working, infrared is working after the following code 
config.enable_stream(rs.stream.infrared, 640, 480, rs.format.bgr8, 30) 

but rgb color stream is not working, 

error RuntimeError: Frame didn't arrive within 5000  , could you assist with this

MartyG-RealSense, 2024-09-10:
Hi @amrit7321  Are you attempting to stream depth, RGB from infrared and also RGB from the RGB sensor on a Raspberry Pi, please?

Usually, the best that can be obtained from a Pi is two streams (such as depth + RGB).  If you are attempting to use three streams then it is likely hitting the limit of what can be obtained from a Pi when using a RealSense camera with it, unfortunately.