realsenseai/librealsense #13271: Failing to run/launch D435i while D435 works fine

akupferb, 2024-08-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 22  | 
| Kernel Version (Linux Only)    |  5.15.136-tegra  | 
| Platform                                 |  NVIDIA Jetson   |
| SDK Version                            |  v2.55.1                         | 

### Issue Description
Within docker on an NVIDIA Jetson Orin NX with realsense packages installed 
RealSense ROS v4.55.1
Built with LibRealSense v2.55.1

When connecting a D435 camera with firmware version 5.13.00.50 and running either 
`ros2 run realsense2_camera realsense2_camera_node` / `ros2 launch realsense2_camera rs_launch.py `
the node runs/launches correctly without issue.
However, when doing the same thing with a D435i with the same firmware version connected, I get the following error:

```:~$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [1723966735.515221692] [camera.camera]: RealSense ROS v4.55.1
[INFO] [1723966735.515402496] [camera.camera]: Built with LibRealSense v2.55.1
[INFO] [1723966735.515435521] [camera.camera]: Running with LibRealSense v2.55.1
 18/08 10:38:55,537 ERROR [281473072294112] (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
 18/08 10:38:55,537 ERROR [281473072294112] (rs.cpp:237) [rs2_create_device( info_list:0xffff800112e0, index:0 ) Backend] xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
 18/08 10:38:55,538 ERROR [281473072294112] (rs.cpp:237) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
 18/08 10:38:55,538 WARNING [281473072294112] (rs.cpp:373) null pointer passed for argument "device"
[WARN] [1723966735.538083617] [camera.camera]: Device 1/1 failed with exception: xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
[ERROR] [1723966735.538194724] [camera.camera]: The requested device with  is NOT found. Will Try again.
```

Runninng `rs_enumerate_devices` also errors:
```:~$ rs-enumerate-devices 
 18/08 10:39:50,003 ERROR [281472868743808] (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
 18/08 10:39:50,004 ERROR [281472868743808] (rs.cpp:237) [rs2_create_device( info_list:0xaaaaf327d4f0, index:0 ) Backend] xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
 18/08 10:39:50,004 ERROR [281472868743808] (rs.cpp:237) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
Could not create device - xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument . Check SDK logs for details
No device detected. Is it plugged in?
```

MartyG-RealSense, 2024-08-18:
Hi @akupferb  Are you using JetPack 6 on your Jetson, please?  If you are then the **MIPI driver** at the link below needs to be installed in order to add support for HID devices (RealSense cameras equipped with an IMU).  This is because JP6 does not support **hidraw**.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

If you are using JP6 and are able to downgrade your JetPack, using JetPack 5.13 should resolve the issue.

If you need to use JP6 then an alternative solution that avoids the need to install the MIPI driver is to install the librealsense SDK with a build script using the **libuvc backend** procedure at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

akupferb, 2024-08-18:
Thanks for the quick response!
I am indeed using Jetpack 6 as a requirement for my system and cannot downgrade, so I will try out the SDK installation from source with libuvc backend to see if that solves it
Thanks

akupferb, 2024-08-18:
I first purged my instances of realsense packages for a clean install of the SDK using `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge` and then ran the install commands for the libuvc_installation.
After the installation was completed successfully, running `rs-enumerate-devices` outputs the following:
```
$ rs-enumerate-devices 
 18/08 12:03:39,516 ERROR [281473118367968] (handle-libusb.h:53) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 18/08 12:03:39,517 ERROR [281473159297632] (uvc-sensor.cpp:412) acquire_power failed: failed to set power state
 18/08 12:03:39,517 ERROR [281473159297632] (rs.cpp:237) [rs2_create_device( info_list:0xaaab0f2cfdb0, index:0 ) UNKNOWN] failed to set power state
 18/08 12:03:39,517 ERROR [281473159297632] (rs.cpp:237) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
Could not create device - failed to set power state . Check SDK logs for details
```
This occurred for both my D435 and D435i... 
As the debian install of the Realsense ROS wrapper (ros-humble-realsense2_camera...) has a dependency on 'ros-humble-librealsense2', I re-installed the wrapper from source and made sure to build and source it, but that did not help.

MartyG-RealSense, 2024-08-18:
If you are accessing the camera within a Docker container, have you tried launching Docker in **sudo** admin permissions mode with **sudo docker run** and including the **--privileged** flag?  For example:

`sudo docker run -it --rm  --privileged -v $HOME/.ros:/root/.ros`

akupferb, 2024-08-18:
I am already running the Docker container with sudo permissions and using --privileged.

I will check out the other solution of installing the MIPI driver...

akupferb, 2024-08-22:
@MartyG-RealSense Is there a method to complete the MIPI driver installation from [this page](https://github.com/IntelRealSense/realsense_mipi_platform_driver)'s instructions _without_ have the "RealSense De-Serialize board" to connect the camera? 
Can the `Enable d4xx overlay:` step be accomplished while the camera is plugged in using a regular USB 3?  The "jetson-io" interface only utilizes the pin connectors and M.2 slot...

akupferb, 2024-08-22:
Note also that I am using a Jetson Orin NX, not the Jetson AGX Xavier...

akupferb, 2024-08-25:
Thanks for the explanation. I fill follow up on their page.

closed 2024-08-25 by akupferb