realsenseai/librealsense #13276: Showing a reduced FoV for RS 435

monajalal, 2024-08-20:
Is there a way to programmatically using Python to control (or reduce) FoV of the current image shown from the video feed? I want to show a customized view but I am not sure how I can do this programmatically. Currently the FoV is very wide and shows way more than expected.

https://github.com/IntelRealSense/realsense-ros/issues/1692#issuecomment-778796892

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-08-21:
Hi @monajalal  Reducing the width of the RGB view by lowering its resolution to 640x480 may be the easiest solution.  

**1280x720**

[image]

**640x480**

[image]

Another possible approach would be to use **YOLO** object detection to generate bounding boxes on the RGB image and then exclude from the image the contents of all the boxes except the one containing the area of the image that you want to keep.

https://docs.ultralytics.com/guides/object-cropping/

monajalal, 2024-08-23:
@MartyG-RealSense I am still working on this. I will close the issue when I get a proper understanding on my end and some sort of resolution. Thanks a lot for your time and help explaining. From my limited search, you cannot change the FoV of a camera, right? But like you can fool it with a newly defined (artificial intrinsics). My goal is not to lose image quality (like what happens when you crop) but to just narrow the field of view and look more to the straight ahead of you rather than looking at such a wide field of view.

MartyG-RealSense, 2024-08-23:
If you are only using this project with your own camera then perhaps you could place a small black piece of card over the RGB sensor with a hole in it on the outside of the camera and make the hole a little smaller than the diameter of the RGB sensor lens.  It should then obscure the edges of the RGB sensor's field of view so that it can only see the amount of the scene that the diameter of the card's hole allows.

Yes, you can define your own intrinsics with Python code.  An example of this is at https://github.com/IntelRealSense/librealsense/issues/4061

closed 2024-08-23 by monajalal