realsenseai/librealsense #13277: Use C++ to write a program: extract the timestamp information of color and depth in multi-frame（at least 100 frames） data

monson-li, 2024-08-21:
I want to use C++ to write a program to extract the color and depth timestamp information from multi-frame data. 

Here I want to ask as a user how to create a new example and run it?

Looking for answers from the bosses.

Thanks！

MartyG-RealSense, 2024-08-21:
Hi @monson-li  There is a C++ script for retrieving metadata at https://github.com/IntelRealSense/librealsense/issues/9191#issuecomment-861484767

It uses depth and infrared though, so you would have to change the infrared stream configuration line from this:

`rs2_config_enable_stream(config, RS2_STREAM_INFRARED, 1, WIDTH, HEIGHT, RS2_FORMAT_Y8, FPS, &e);
`
to this:

`rs2_config_enable_stream(config, RS2_STREAM_COLOR, WIDTH, HEIGHT, RS2_FORMAT_RGB8, FPS, &e);
`
If you are building a new project yourself in Linux then you will need to ensure that the **CMakeLists.txt** file has the necessary linkages to the librealsense library.  https://github.com/IntelRealSense/librealsense/issues/12907#issuecomment-2100263487 has information that is relevant to doing so.

https://github.com/IntelRealSense/librealsense/issues/6797 is also a good reference for compiling with CMake.

If a project is being built in Visual Studio on Windows, a RealSense user shared a guide to setting up a Visual Studio project using three .**props** property sheet files provided by the RealSense SDK..

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

monson-li, 2024-08-22:
Hi @MartyG-RealSense I will try and answer you later