realsenseai/librealsense #13278: RealSense D435 image doesn't take the entire secondary display width and height using cv2

monajalal, 2024-08-21:
```
        cv2.namedWindow("RealSense", cv2.WINDOW_NORMAL) #working resized window
        # cv2.namedWindow("RealSense",  cv2.WND_PROP_FULLSCREEN)
        cv2.moveWindow("RealSense", second_display_x-1, second_display_y-1) #working resized window
        cv2.resizeWindow("RealSense", second_display_width, second_display_height) # Resize to screen dimensions

        cv2.setWindowProperty("RealSense", cv2.WND_PROP_FULLSCREEN, cv2.WINDOW_FULLSCREEN)#working resized window
```

So when I use the following, it takes most of the display except the first few top rows and the first few bottom rows. Any method I could enforce it to take the entire display?

The display I am enforcing this onto is:

[image]

monajalal, 2024-08-23:
@MartyG-RealSense thank you so much this fixed the problem.

In order to fix the op, I tried bunch of things. Before, I was able to select the trash can and then show it in entire screen as the image above, now, with all my new changes combined, after cv2.selectROI it doesn't zoom it out to capture the entire screen automatically, do you have any guess what may be going wrong or what am I missing?

```
## super resolution for post-cropping is very slow
# sr = cv2.dnn_superres.DnnSuperResImpl_create()
# path = "EDSR_x4.pb"
 
# sr.readModel(path)
 
# sr.setModel("edsr",4)
config.enable_stream(rs.stream.color, 1920, 1080, rs.format.bgr8, 30)
cv2.namedWindow("RealSense",  cv2.WND_PROP_FULLSCREEN)
        cv2.moveWindow("RealSense", second_display_x-1, second_display_y-1) #working resized window
        cv2.resizeWindow("RealSense", second_display_width, second_display_height) # Resize to screen dimensions
        cv2.setWindowProperty("RealSense", cv2.WND_PROP_FULLSCREEN, cv2.WINDOW_FULLSCREEN)#working resized window

       new = cv2.rotate(color_img, cv2.ROTATE_90_CLOCKWISE)
       if not roi_selected:
            roi = cv2.selectROI('RealSense', new)
            roi_selected = True
            print('roi is: ', roi)
            if roi != (0, 0, 0, 0):
                x, y, w, h = roi

        # sr_roi_img = sr.upsample(roi_image) # commenting super resolution post cropping since it is very slow
        cv2.imshow('RealSense', roi_image)
        # cv2.imshow('RealSense', sr_roi_img)

```
One change I remember is installing super resolution package for opencv contrib (had to uninstall the previous version of open contrib and reinstall and upgrade). Unfortunately, I don't know what the previous version was. My goal is after I select the trash can using the selectROI from opencv, similar to original post I want it to take the entire screen in the 1920x1080 resolution you mentioned.

The reason I wanted to use super resolution is because when I select the ROI it gets cropped and then resize to full screen now it is very bad resolution. Super resolution works but it is too slow I realized the idea doesn't pan out. If you have a suggestion please let me know how we can get this fixed.

[image]

monajalal, 2024-08-23:
[image]

```
        roi_image_fullscreen = cv2.resize(roi_image, (second_display_width, second_display_height))
        cv2.imshow('RealSense', roi_image_fullscreen)
```

fixed it. I didn't have this piece of code in the previous working version.

MartyG-RealSense, 2024-08-23:
That's great news that you achieved a solution.  Thanks very much for the update and sharing the details of the solution!