realsenseai/librealsense #13279: IMU in Realsense d455 does not work on Android OS examples

brianchae, 2024-08-22:
| Required Info                         |                                                 |
|---------------------------------|-------------------------------- |
| Camera Model                       | {D455}                                     | 
| Firmware Version                   | 5.16.0.1                                    | 
| Operating System & Version |   Android OS                           | 
| Platform                                 | Samsung Galaxy S22              |
| SDK Version                           |  {2.55.1}                                   | 
| Language                               |  {Java/Kotlin}                           | 
| Segment			         |  {Smartphone}                        | 

### Issue Description
[Issue]
Hello, I am trying to connect Intel Realsense D455 to my smartphone Samsung Galaxy S22. What I want to do is to acquire 4 streams, acc, gyr, infrared0, and infrared1. I have referenced the Android example, stream_pipeline_callback, in this repo. 

        try(Config config  = new Config())
        {
            // streaming pipeline with default configuration or custom configuration and user callback
            // the user callback must be a class implements the FrameCallback interface

            // default configuration
            //try statement needed here to release resources allocated by the Pipeline:start() method
            //try(PipelineProfile pp = mPipeline.start(mFrameHandler)){}

            // custom configuration
            //config.enableStream(StreamType.DEPTH, 640, 480);
            //config.enableStream(StreamType.COLOR, 640, 480);
            config.enableStream(StreamType.ACCEL, StreamFormat.MOTION_XYZ32F);
            config.enableStream(StreamType.GYRO, StreamFormat.MOTION_XYZ32F);
            config.enableStream(StreamType.INFRARED, 1, 640, 480, StreamFormat.Y8, 30);
            config.enableStream(StreamType.INFRARED, 2, 640, 480, StreamFormat.Y8, 30);
            try(PipelineProfile pp = mPipeline.start(config, mFrameHandler)){}
        }

However, all the data does not seem to be work if I try to enable acc and gyr. If I disable acc and gyr, both the infrared0 and infrared1 work. This only happens in Android. 
From C++ or the Realsense Viewer in PC does work fine, but it only seems happen in Android.
Do I missing something here?

MartyG-RealSense, 2024-08-22:
Hi @brianchae  There was a case in July 2024 at https://github.com/IntelRealSense/librealsense/issues/13170 where a RealSense Android user had been able to access the IMU successfully with firmware 5.12.15.50 but not with 5.16.0.1.

The minimum recommended firmware for D455 is **5.13.0.50**, the firmware that came immediately after 5.12.15.50.  So it may be worth trying downgrading your camera firmware to 5.13.0.50 to see whether it resolves the IMU issue.

brianchae, 2024-08-22:
Thank you for your reply. I will try to downgrade my camera firmware as you suggested. About the firmware, I cannot find the version 5.13.0.50 from the website https://www.intel.com/content/www/us/en/download/19242/firmware-for-intel-realsense-d400-product-family.html. Is there any other link where I can find the firmware you recommanded?

brianchae, 2024-08-23:
@MartyG-RealSense I found the firmware version 5.13.0.50 though some searching.
https://support.intelrealsense.com/hc/en-us/community/posts/6456192756371

I have tried downgrading the firmware into 5.13.0.50 as well as the SDK version into 2.50.0. However, the problem still remains the same with D455&Android. When I try with my D435i, it works with the same code. 
It seems D455 in Android somehow needs some support.

MartyG-RealSense, 2024-08-23:
You are very welcome, @brianchae - it's great to hear that you achieved a working solution.  Thanks very much for the update!

closed 2024-08-30 by MartyG-RealSense