realsenseai/librealsense #13281: Feature Request: nuget package

electro-logic, 2024-08-22:
Would be nice to have an official nuget package for the librealsense for Visual Studio / C# developers.
There are actually few unofficial, but not really updated

Thanks

electro-logic, 2024-08-22:
Here Dot Net Wrapper v2.55.1 compiled
[realsense2.55.1.zip](https://github.com/user-attachments/files/16720035/realsense2.55.1.zip)

MartyG-RealSense, 2024-08-23:
Hi @electro-logic  The best route to an official up-to-date nuget package would likely be for a member of the RealSense user community to create it and then submit it as a Pull Request so that the Intel RealSense team can consider merging it into the SDK.

https://github.com/IntelRealSense/librealsense/pulls

The latest 2.56.1 SDK is currently an interim beta release that does not provide Windows executables, Linux Debians, Python wheels, etc.  It just has a downloadable source code zip-file.  

It should be possible to use that code to build the 2.56.1 SDK on Windows from this source file if necessary by using the instructions at the link below if you have not attempted to do so already.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

electro-logic, 2024-08-23:
Hi @MartyG-RealSense 

Someone from Intel needs to create a GitHub Action to automate the nuget package upload.
https://www.meziantou.net/publishing-a-nuget-package-following-best-practices-using-github.htm

I compiled myself v2.55.1 and it's fine, but would be nice for me to avoid to recompile the wrapper at every release.

Thank you

MartyG-RealSense, 2024-08-23:
I will pass your suggestion about a GitHub Action for a nuget package to my Intel RealSense colleagues.

MartyG-RealSense, 2024-08-24:
Hi @electro-logic  My colleagues informed me that the SDK’s release flow and content were changed, and C# wrapper collaterals have been removed since librealsense SDK version 2.54.1 last year.  This minimizes the file size of the release and simplifies it.  They advise that you continue to build from source as you are already doing.

electro-logic, 2024-08-24:
Hello @MartyG-RealSense 

Nuget is a separate repository for Dot Net libraries and the C# wrapper can be compiled and released there with automated actions. There is no need to add the wrapper to the GitHub SDK release and increase the file size.

Anyway thank you for asking the team

MartyG-RealSense, 2024-08-30:
Hi @electro-logic  Do you require further assistance with this case, please?  Thanks!

closed 2024-08-30 by electro-logic

vicero, 2025-01-16:
@MartyG-RealSense the removal of the dotnet dll and the unavailability of them in NuGet created a big headache for me.  I'm not able to compile the dotnet bindings on my mac because of the CMake dependency on Visual Studio.

I ended up having to go back in time to find a release that didn't have the dotnet dll removed.  It is not mentioned by name in your release notes, [searching](https://github.com/search?q=repo%3AIntelRealSense%2Flibrealsense+intel.realsense.dll&type=issues) for Intel.RealSense.dll brought up https://github.com/IntelRealSense/librealsense/issues/12100

I had to install the SDK on a personal computer to extract the dlls and copy them over to my work machine after digging through the repository.  There's no mention of this in the C# readme either, the [documentation](https://dev.intelrealsense.com/docs/csharp-wrapper) says it is possible to install the dll by installing the sdk.

This has been a bit of torture to get to this point.  If you're not going to include it in the installer anymore, could you at least update the docs, publish the dlls as portable build artifacts in a .zip file, or publish them to NuGet?

Thank you!

MartyG-RealSense, 2025-01-17:
@electro-logic  Thanks so much for your help!

@vicero  It is likely that the approach that you suggested would be a low development priority.  Could the method kindly provided by @electro-logic be suitable for you instead?