realsenseai/librealsense #13284: Manually installing SDK possible for Debian 12.6?

h-a-n-n-e-s, 2024-08-23:
| Camera Model                       | D456 | 
| Operating System & Version |  Linux Debian 12.6  | 
| Kernel Version (Linux Only)    | 6.1.0                                      | 
| Platform                                 | PC |
| SDK Version                            |  2.55.1                    | 

### Issue Description
Would there be a chance to install the sdk from source manually for Debian 12.6 if the Linux kernel version would be right, like 6.2 or 6.5?
Or is the distribution name also important, like it has to be Ubuntu?

Currently I use kernel 6.1 and get the error
`Unsupported distribution bookworm, kernel version 6.1.0-23-amd64 . The patches are maintained for Ubuntu 16/18/20 LTS`

MartyG-RealSense, 2024-08-23:
Hi @h-a-n-n-e-s  If you are building the SDK from source code then you can patch the kernel for 6.2 or 6.5 and Ubuntu 20.04 or 22.04 using the patch script **./scripts/patch-realsense-ubuntu-lts-hwe.sh**

If you are unable to change your kernel version then you can bypass the kernel and avoid the need to apply a patch for it by building from source code with the **libuvc backend** or the **RSUSB backend** installation method.  The SDK will be built in RSUSB = true mode if the flag **-DFORCE_RSUSB_BACKEND=TRUE** is added to the CMake build instruction.

Instructions for an RSUSB installation can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

The libuvc backend method, which works similarly to RSUSB, is easier to use as there is a pre-made build script available for it.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Another advantage of the libuvc and RSUSB backends is that they are not dependent on a particular Linux version, so it is easier to use a non-Ubuntu Linux with it.

h-a-n-n-e-s, 2024-08-23:
Thanks a lot @MartyG-RealSense !
The libuvc method worked like a charm! (See screenshot blow).

By pypassing the kernel, are there any performance differences, like gpu usage?

[image]
nce differences, like gpu usage?

MartyG-RealSense, 2024-08-23:
You're very welcome.  I'm pleased that I could help.  :)

There are some differences between a libuvc build and a kernel-patched build.  For a comparison of libuvc backend with a patch-based librealsense build, please visit https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604 and scroll down through the linked-to comment to the section headed **What are the advantages and disadvantages of using libuvc vs patched kernel modules?**

GPU performance should not be negatively impacted by choosing the libuvc method though.

h-a-n-n-e-s, 2024-08-24:
Thanks again for the detailed info @MartyG-RealSense !
I tried the Viewer 2.55.1 and streaming seems to work well.

For our system, we are mainly interested in the depth data and I need to get this data into a Rust program. Can I directly use the libuvc for this? Or do I need to write some Rust bindings to the function in the RealSense API? Any hints are greatly appreciated! Thanks again!

MartyG-RealSense, 2024-08-24:
There is an actively maintained RealSense Rust binding available outside of the RealSense SDK.  The most recent SDK version that it has been updated for is 2.54.2, the one before 2.55.1, so it will likely work with 2.55.1.

https://gitlab.com/tangram-vision/oss/realsense-rust

closed 2024-08-30 by h-a-n-n-e-s