realsenseai/librealsense #13285: Selecting a display for Intel RealSense D435 camera so there would be no lag

monajalal, 2024-08-23:
I am using an Ubuntu 22.04 system (open to use Windows if it may change anything). 

I am using a secondary display to show the real-time feed of RealSense D435 camera on the display.

I would like to know if the secondary display does need to have certain criteria for the RGB feed of this camera to meet no lag and be exactly real-time.

For example, does the display need to be 60Hz refresh rate? Does it need to have response time <16ms?

I ask since it looks FPS of RGB feed is 30 FPS.

Any recommendation on this end is really appreciated. 

What hardware specs do I need to consider for the display I am planning to buy if I want to make sure the video stream shown from Intel RealSense D435 is not laggy at all and indeed real-time? Assume, budget is not a concern.

P.S.: Have you seen people facing this lag problem?

MartyG-RealSense, 2024-08-23:
Hi @monajalal  You should not need a special display screen.

The D435 model has a fast global shutter on its depth sensors but a slower rolling shutter on its RGB sensor. This can cause the RGB to lag behind depth if fast motion is being observed.

A workaround is to set depth to 30 FPS and set RGB to 848x480 resolution and 60 FPS.  When RGB is 60 FPS, the resultant lag reduction compensates for the slow shutter to equalise the performance of the two shutter types.

 Another method of reducing lag on the RGB sensor is to disable RGB auto-exposure and set a manual RGB exposure value of 78 and an RGB FPS speed of 6 (six, not sixty).

The D455 type cameras (D455, D456, D457, etc) have a fast global shutter on both the RGB and depth sensors if you want to consider that option in future so that you do not need to compensate for a slow shutter on the RGB sensor,

monajalal, 2024-08-23:
We need to use the 1920x1080 resolution for the RGB camera. Also, for this application, we don't need the depth. Do you think if I only need to use 30FPS RGB feed in that resolution, it's gonna be laggy?

MartyG-RealSense, 2024-08-23:
It depends on how fast the camera will be moving or, if the camera is stationary, how quickly people and objects observed by the camera are moving.  If there is motion faster than a wave of the hand then there may be lag / blurring on the RGB.  The D415 camera model also has the rolling shutter on its RGB and is affected by high speed motion in the same way.  https://github.com/IntelRealSense/librealsense/issues/3554 has more information about this.

If the camera will be stationary or observing motion slower than a hand wave then RGB lag will likely not be an issue.  The computer itself should be well specced though because streaming 1920x1080 at 30 FPS will have more of a processing burden than using a lower resolution.

closed 2024-08-23 by monajalal