realsenseai/librealsense #13287: No Frames Received for Intel RealSense D435 via RealSense Viewer

monajalal, 2024-08-23:
I need to be occasionally testing both FRAMOS D435e and Intel RealSense D435 for my applications on the same Ubuntu 22.04 laptop.

RS D435 camera was working but my FRAMOS stopped working which means I had to install the updated firmware. However, now, my Intel RS D435 is not working.

I am getting this error in $ realsense-viewer GUI

[image]

[image]

```
$ realsense-viewer
 23/08 13:08:54,240 WARNING [132948055557696] (model-views.cpp:4229) Exception caught trying to detect presets: could not open directory: /home/mona/Documents/librealsense2/presets/
 23/08 13:09:20,397 WARNING [132947659195968] (backend-v4l2.cpp:1132) Frames didn't arrived within 5 seconds

```

Upon opening realsense-viewer, I get this error

[image]

MartyG-RealSense, 2024-08-23:
Hi @monajalal  The camera firmware driver that should ideally be used with SDK 2.50 is **5.13.0.50**, whilst you have 5.15.1.0 installed.

The newer firmwares after 5.13.0.50 contained a change to the **Auto Gain Limit Toggle** feature that could cause an error like the one that you encountered when using those firmwares with older SDK versions such as 2.50.  

Firmware 5.16.0.1 can work with SDK 2.50, so updating again to that firmware should hopefully resolve your issue.

monajalal, 2024-08-23:
Thanks for your notes. 

So, I did this:

```
2024-08-23 13:31:50 [mona@DOS Signed_Image_UVC_5_16_0_1]$ rs-fw-update -r -f Signed_Image_UVC_5_16_0_1.bin 

Update to FW: Signed_Image_UVC_5_16_0_1.bin

No recovery devices were found!

```

But the firmware wasn't updated. Did I do it wrongly?

[image]

I got the bin file from here https://dev.intelrealsense.com/docs/firmware-releases-d400

exact file link https://www.intelrealsense.com/download/23422/?tmstv=1713899242&_ga=2.187605931.286611782.1724434091-1350462228.1723576954

MartyG-RealSense, 2024-08-23:
If your camera has not had a firmware update failure that puts it into Recovery mode then you do not need to include the -r parameter in the rs-fw-update command.

monajalal, 2024-08-23:
BTW I just remembered about this StarTech.com USB extension device. I used it and problem was resolved however, I am still getting those errors

https://github.com/IntelRealSense/librealsense/issues/11973#issuecomment-1631188216

[image]

monajalal, 2024-08-23:
Removed the -r flag and it seems working. Should I be concerned about not using USB3 and having to use USB2?
```
2024-08-23 14:06:26 [mona@DOS Signed_Image_UVC_5_16_0_1]$ rs-fw-update -f Signed_Image_UVC_5_16_0_1.bin 

Warning! the camera is connected via USB 2 port, in case the process fails, connect the camera to a USB 3 port and try again

Updating device: 
Name: Intel RealSense D435, serial number: 819112070232, update serial number: 815213022437, firmware version: 05.15.01.00, USB type: 2.1
 23/08 14:06:46,287 ERROR [132530506251328] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error

Firmware update started

Firmware update progress: 100[%]

Firmware update done

Device 819112070232 successfully updated to FW: 05.16.00.01
```

Also, confirming that with new firmware, I still get an error upon opening realsense-viewer.

[image]

[image]

closed 2024-08-23 by monajalal

MartyG-RealSense, 2024-08-23:
I know that firmware 5.16.0.1 can work with librealsense 2.50.0 in the RealSense ROS wrapper.  This may not be the case with the Viewer though.  As mentioned earlier, firmware 5.13.0.50 is meant to be used with 2.50.0.

librealsense 2.16.0.1 can also sometimes work with the ROS1 wrapper 2.3.2 even though it is not meant to work.

If you are happy to use the 2.50.0 Viewer and you also need 2.50.0 for the 2.3.2 ROS1 wrapper then using firmware 5.13.0.50 is likely to solve your problems.