realsenseai/librealsense #13289: help

m0-hayashimanato, 2024-08-24:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2024-08-24:
Hi @m0-hayashimanato  This error usually occurs when attempting to use the **Add Source** button of the RealSense Viewer tool to access the RGB of a non-RealSense camera such as a USB webcam or the built-in webcam on a laptop,  Not all non-RealSense webcams are compatible with this feature, unfortunately.

Are you using the Add Source button, please?

MartyG-RealSense, 2024-08-30:
Hi @m0-hayashimanato  Do you require further assistance with this case, please?  Thanks!

m0-hayashimanato, 2024-08-30:
Thanks for your reply. Sorry for the late reply.

I used the add source button. 「Some RealSense users also use this button to attempt to access their RealSense camera if it is not listed in the Viewer's options side-panel when the Viewer is launched.」My situation is exactly like this one you pointed out. 

What do I do if it is not listed in the Viewer's options side-panel ?

MartyG-RealSense, 2024-08-30:
Which RealSense camera model are you using, please?  Some models may not appear in the Viewer's side-panel if they have been retired and support for them has been removed.

For example, support for the L515 and T265 camera models was removed in version 2.54.1, so if you have one of those cameras then version 2.51.1 should be used.

m0-hayashimanato, 2024-08-30:
I use the L515 camera model.
How can I use version 2.51.1?

MartyG-RealSense, 2024-08-30:
If you download and run the 2.51.1 version of the **Intel.RealSense.Viewer.exe** Windows file in the 'Assets' file list at the bottom of the page at the link below then the L515 should be detected. 

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

[image]

m0-hayashimanato, 2024-08-30:
Thank you very much. It was really helpful. I will try it.

m0-hayashimanato, 2024-09-05:
For now, it's okay. Thank you very much.

MartyG-RealSense, 2024-09-05:
You are very welcome.  As you are not having problems at this time, I will close the issue.  Thanks very much for the update!

closed 2024-09-05 by MartyG-RealSense