realsenseai/librealsense #13295: Intel RealSense Viewer v2.55.1 can't find RealSense L515

Mendopower, 2024-08-27:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|

I connected my RealSense L515 camera to my laptop but the Viewer.exe can't find my camera. 
I can see L515 in the device list:
[image]

MartyG-RealSense, 2024-08-27:
Hi @Mendopower  Support for L515 was removed in RealSense version 2.54.1 as L515 is a retired camera model.  If you download and run the **2.51.1** version of the **Intel.RealSense.Viewer.exe** file at the link below  then you should be able to access your camera in the Viewer.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

Mendopower, 2024-08-27:
Thank you for your help, it’s working now.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

MartyG-RealSense, 2024-08-27:
You are very welcome.  I'm pleased that I could help!

closed 2024-09-05 by MartyG-RealSense