realsenseai/librealsense #13304: out of bound access in rs2_project_color_pixel_to_depth_pixel

wsmlby, 2024-08-28:
https://github.com/IntelRealSense/librealsense/blob/e1688cc318457f7dd57abcdbedd3398062db3009/src/rs.cpp#L4120

this doesn't seems to assume points are 0-indexed and allow y = height which will cause a out of bound access and crash.

MartyG-RealSense, 2024-08-29:
Hi @wsmlby  An out of bounds access error is one that does not usually occur.  If you are using C++, do you have any problems if you test the script at https://github.com/IntelRealSense/librealsense/issues/2982 which uses **rs2_project_color_pixel_to_depth_pixel** please?

wsmlby, 2024-08-29:
I mean it is pretty simple if in any case the passed in value get mapped to a p[1]= height here, it will cause a out of bounds access. We are seeing this in production. This is a very simple bug I cannot believe Intel is releasing this code to public.

MartyG-RealSense, 2024-08-29:
This is not an issue that is currently being reported by other RealSense users, but of course this does not mean that it is not happening in your particular case.  You are very welcome to submit a PR containing your fix.

wsmlby, 2024-08-29:
added a PR. please take a look.

we are currently working around this by allocate one more line for the depth buffer but this should not be needed.

MartyG-RealSense, 2024-08-30:
Thanks so much @wsmlby for the PR at https://github.com/IntelRealSense/librealsense/pull/13307

I have added an Enhancement label to this issue to signify that it shoukd be kept open whilst your PR is active.  Thanks again!

MartyG-RealSense, 2026-10-25:
Case closed due to the Pull Request (PR) associated with it having been already closed.

closed 2026-10-25 by MartyG-RealSense