realsenseai/librealsense #13308: Misalignment Between Depth and RGB Streams on Intel RealSense D435

tangyipeng100, 2024-08-30:
I'm encountering a misalignment between the depth and RGB streams when using the Intel RealSense D435 camera. The two streams do not align accurately in both absolute and relative positions, as shown in the attached image.
[image]

Has anyone experienced a similar issue or knows how to fix this? Any advice would be appreciated.**

MartyG-RealSense, 2024-08-30:
Hi @tangyipeng100  The RealSense Viewer does not have alignment in its 2D mode.  Depth and RGB can be mapped together if you click the '3D' option in the top corner of the Viewer window to enter 3D pointcloud mode and then enable depth first and RGB second.

Without alignment, there will be differences between the depth and RGB images because the position of the RGB sensor on the front of the camera is horizontally offset from the left infrared sensor (which is the origin of depth).

Also on the D435 model, depth has a larger field of view size than the RGB field of view size.  So when depth is aligned to color, the outer edges of the aligned image will be cut off.

Furthermore, by default the Viewer has a post-processing filter called the **Decimation** filter enabled that 'downsamples' the depth resolution by a factor of 2.  For example, a 640x480 depth image will be reduced to 320x240 resolution.  To turn this filter off and display depth at the resolution that you selected, go to **Stereo Module > Post Processing**, expand open the list of filters and left-click on the blue icon beside Decimation Filter to turn the icon black (Off).

tangyipeng100, 2024-08-31:
thx. I am using the Intel RealSense SDK to stream and align depth and color frames. After aligning the frames using align.process(frames), I noticed that the depth frame (depth_frame.get_data()) contains only zeros, rendering the depth stream unusable. How to solve it?

```
import pyrealsense2 as rs
import numpy as np
import cv2

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()

# Get device product line for setting a supporting resolution
pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

found_rgb = False
for s in device.sensors:
    if s.get_info(rs.camera_info.name) == 'RGB Camera':
        found_rgb = True
        break
if not found_rgb:
    print("The demo requires Depth camera with Color sensor")
    exit(0)

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

if device_product_line == 'L500':
    config.enable_stream(rs.stream.color, 960, 540, rs.format.bgr8, 30)
else:
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Create alignment object
align_to = rs.stream.color
align = rs.align(align_to)

# Start streaming
pipeline.start(config)

try:
    while True:
        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        frames = align.process(frames) # if do not comment this code, then depth_image_aligned only contains zero
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()

        depth_image_aligned = np.asanyarray(depth_frame.get_data())
        if np.count_nonzero(depth_image_aligned) == 0:
            print("Aligned depth frame contains only zeros")

MartyG-RealSense, 2024-08-31:
Hi @tangyipeng100  If you are getting zeroes like the case at https://github.com/IntelRealSense/librealsense/issues/3631 then a suggestion for correcting this is to convert the numpy array to a **list** because of how numpy summarizes when printing large arrays.  The link below descibes how such a conversion might be performed.

https://numpy.org/doc/stable/reference/generated/numpy.ndarray.tolist.html

tangyipeng100, 2024-08-31:
> Hi @tangyipeng100 If you are getting zeroes like the case at #3631 then a suggestion for correcting this is to convert the numpy array to a **list** because of how numpy summarizes when printing [...]

thx, I have used np.unique() to observe the value, and it equals to 0. Besides, if I comment 'frames = align.process(frames) ', then I can get the depth stream correctly.

MartyG-RealSense, 2024-08-31:
The common way to program alignment, as demonstrated by the librealsense SDK's official Python depth-color alignment program **align_depth2color.py**, is to give the aligned frames their own individual variable name such as 'aligned_frames' instead of using 'frames'.  That way, you are not overwriting the contents of the original 'frames' variable at `frames = pipeline.wait_for_frames()`

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

```
frames = pipeline.wait_for_frames()

aligned_frames = align.process(frames)
aligned_depth_frame = aligned_frames.get_depth_frame()
color_frame = aligned_frames.get_color_frame()

 depth_image = np.asanyarray(aligned_depth_frame.get_data())
 color_image = np.asanyarray(color_frame.get_data())
```

tangyipeng100, 2024-09-01:
thx, In fact, I am using [opencv_viewer_example](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py, ), and I try to follow your strategy, and it works as before, like the image shows:
[image]
If I do not use `align.process(frames), then I can get:  

[image]
`

when I run [align-depth2color.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py), I get:
[image]

MartyG-RealSense, 2024-09-02:
Does the alignment work if you use a modified version of the align_depth2color.py script at https://github.com/IntelRealSense/librealsense/issues/11329

tangyipeng100, 2024-09-02:
I have changed to another Windows10 PC, it still does not work.
If I set `align_to = rs.stream.depth`, then get:
[image]

if I set `align_to = rs.stream.color`, then get:
[image]
And my Depth Scale is:  0.0010000000474974513

MartyG-RealSense, 2024-09-02:
This is an unusual problem, though is unlikely to be a hardware issue if `align_to = rs.stream.depth` works.    With this instruction, **color to depth** alignment is being performed instead of the more commonly used depth to color alignment.  The aligned images created by the two methods are similar, so if color to depth alignment works then it's fine to use that method.

tangyipeng100, 2024-09-02:
Although setting align_to = rs.stream.depth works, the color image does not display correctly. Instead, it looks like this:
[image]
The RGB image appears to contain only zeros, and I am still unable to obtain the correct RGB-depth pair alignment.How can I resolve this issue to get the correct alignment between the RGB and depth images?

tangyipeng100, 2024-09-02:
If resolving this issue proves too challenging, could I instead use the intrinsic and extrinsic parameters to achieve the alignment?

MartyG-RealSense, 2024-09-02:
You could try resetting your camera to its factory-new default calibration in the RealSense Viewer using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 to see whether that makes a difference.

https://github.com/IntelRealSense/librealsense/issues/12966#issuecomment-2181627379

In regard to using intrinsic and extrinsic parameters, https://github.com/IntelRealSense/librealsense/issues/1274#issuecomment-370228022 has some suggestions for methods of alignment other than using align.process.

ozgur-kurt, 2024-09-07:
@tangyipeng100 I have faced same issue for Realsense D435i camera. 
I am trying to improve this code
`import pyrealsense2 as rs
import numpy as np
import cv2

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
profile = pipeline.start(config)
frame = 10000
try:
    while True:
        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()

        if not depth_frame or not color_frame:
            continue

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Align depth frame to color frame
        align = rs.align(rs.stream.color)
        aligned_frames = align.process(frames)
        aligned_depth_frame = aligned_frames.get_depth_frame()
        aligned_depth_image = np.asanyarray(aligned_depth_frame.get_data())

        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(aligned_depth_image, alpha=0.03), cv2.COLORMAP_JET)
        images = np.hstack((color_image, depth_colormap))

        cv2.namedWindow('Aligned Depth and Color', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Aligned Depth and Color', images)
        # cv2.imwrite(str(frame)+"a.png",images)
        frame =  frame +1
        # print("TEST")

        if cv2.waitKey(1) & 0xFF == ord('q'):
            break

finally:
    # Stop streaming
    pipeline.stop()`

tangyipeng100, 2024-09-07:
@ozgur-kurt 
Thanks for your reply, I have tried to use your code, but I still get the same result as the fig shows:
[image]
Have you solved this problem?

tangyipeng100, 2024-09-07:
@ozgur-kurt  
I am trying to modify the code  
`align = rs.align(rs.stream.color)`  
to  
`align = rs.align(rs.stream.depth)`  
so that I can synchronize the depth and RGB streams.  
[image]  
However, I noticed in your code, the RGB frame isn't using the aligned frame, as shown below:

```python
 # Convert images to numpy arrays
depth_image = np.asanyarray(depth_frame.get_data())
color_image = np.asanyarray(color_frame.get_data())

# Align depth frame to color frame
align = rs.align(rs.stream.depth)
aligned_frames = align.process(frames)
aligned_depth_frame = aligned_frames.get_depth_frame()
aligned_depth_image = np.asanyarray(aligned_depth_frame.get_data())
```

In this snippet, the depth frame is aligned, but the RGB frame isn't, which leads to misalignment between the depth and RGB streams, as shown in the image I’ve provided. Could you provide guidance on how to ensure both frames are properly synchronized?

MartyG-RealSense, 2024-09-13:
Hi @tangyipeng100  Do you have an update about this case that you can provide, please?  Thanks!

tangyipeng100, 2024-09-14:
@MartyG-RealSense Thank you for your response. I've been a bit busy recently, but I plan to use the intrinsic parameters to align the depth and RGB streams next. I’ll share any updates as soon as I have them.

tangyipeng100, 2024-09-17:
@MartyG-RealSense Hi, I am trying to manually acquire the depth-to-RGB extrinsic parameters to align the depth and RGB streams using the following code:

```python
import pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

profile = pipeline.start(config)

depth_to_color_extrinsics = profile.get_stream(rs.stream.depth).as_video_stream_profile().get_extrinsics_to(
    profile.get_stream(rs.stream.color))

print("Rotation:", depth_to_color_extrinsics.rotation)
print("Translation:", depth_to_color_extrinsics.translation)

pipeline.stop()
```

However, I am getting the following results:

```
Rotation: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
Translation: [0.0, 0.0, 0.0]
Extrinsics from depth to color are all zeros.
```

It seems like there might be an issue with the extrinsic parameters for my device. Could this be causing the `rs.align` function to not align the depth and RGB streams correctly? Is there any way to resolve this issue or correctly acquire the extrinsic parameters?

Thanks!

MartyG-RealSense, 2024-09-17:
This particular question is outside of my programming knowledge, unfortunately.  I do apologize.

However, https://github.com/IntelRealSense/librealsense/issues/10180#issuecomment-1019452802 has a Python demonstration of getting depth to color extrinsics.

```
depth_frame = frames.get_depth_frame()
color_frame = frames.get_color_frame()

depth_to_color_extrin = depth_frame.profile.get_extrinsics_to(color_frame.profile)
```

tangyipeng100, 2024-09-18:
Hello,

I've been following the provided code but I continue to get extrinsics parameters as all zeros, as shown in the image below:

[image]

I'm wondering if it's possible to obtain the correct extrinsics parameters using the official software "Intel RealSense Dynamic Calibrator"? Unfortunately, when I tried using the Dynamic Calibrator, I encountered the following error:

[image]

Any guidance on how to resolve this issue or alternative methods to correctly retrieve extrinsics parameters would be greatly appreciated. Thanks in advance!

MartyG-RealSense, 2024-09-18:
The error indicates that the Dynamic Calibration tool is having problems with the 'calibration table' inside the camera which stores the camera's calibration parameters.

When the Dynamic Calibrator tool is installed, it also installs a text-based program called **CustomRW** that can be used to perform a 'gold reset' of the camera to create a new calibration table inside the camera to replace one that is possibly corrupted.

On Windows you can activate the gold reset to reset the camera to its factory-new default calibration by inputting the text command below into the Windows Command Prompt text interface to launch CustomRW in gold reset mode.

`Intel.Realsense.CustomRW.exe -g`

tangyipeng100, 2024-09-19:
I attempted to run the following commands, but encountered an issue:

```
C:\Users\TYP>Intel.Realsense.CustomRW.exe -g
CustomRW for Intel RealSense D400, Version: 2.13.1.0

  Device PID: 0B07
  Device name: Intel RealSense D435
  Serial number: 216122070043
  Firmware version: 05.15.01.00
  Type: 3.2
  Physical port: \\?\usb#vid_8086&pid_0b07&mi_00#7&33298fb&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global

Calibration on device successfully reset to default gold factory settings.

C:\Users\TYP>Intel.Realsense.CustomRW.exe -r
CustomRW for Intel RealSense D400, Version: 2.13.1.0

  Device PID: 0B07
  Device name: Intel RealSense D435
  Serial number: 216122070043
  Firmware version: 05.15.01.00
  Type: 3.2
  Physical port: \\?\usb#vid_8086&pid_0b07&mi_00#7&33298fb&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global

Error (9901): calibration tables on device are not supported. Please try again with the latest Dynamic Calibrator software. 
If you are already using the latest software, please re-calibrate the device with the latest OEM Calibration tools and try again.
Check CustomRW tool.
```

I also tried installing a previous firmware version, but the error persists. It seems like I'm encountering a similar issue to [this one](https://github.com/IntelRealSense/librealsense/issues/7300). 

Is this a known issue with the D435? If so, are there any recommended solutions or workarounds for this error?

MartyG-RealSense, 2024-09-19:
As you observed, this issue about calibration tables not being supported has occurred a small number of times in the past.  There is not a known fix for it though, unfortunately.

The error messages suggests "If you are already using the latest software, please re-calibrate the device with the latest OEM Calibration tools and try again".  That OEM version of the Dynamic Calibration software is only available though to purchasers of Intel's $1500 USD calibration target board product, so this may not be an option for you.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

An alternative that may be worth exploring is to try installing the Dynamic Calibration software on Ubuntu using the instructions on page 14 onwards of the tool's user guide.  Those instructions install the earlier **2.11.0.0** version of the tool.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

MartyG-RealSense, 2024-09-26:
HI @tangyipeng100  Do you have an update about this case that you can provide, please?  Thanks!

tangyipeng100, 2024-09-27:
@MartyG-RealSense I am now trying to calibrate the depth and RGB manually and have obtained initial results. Recently, I have been quite busy, but I will update you with my findings after this period. Thanks!

MartyG-RealSense, 2024-10-28:
HI @tangyipeng100  Do you require further assistance with this case, please?  Thanks!

tangyipeng100, 2024-10-28:
@MartyG-RealSense 

I apologize for my delayed response. I've been busy with my paper and interview preparations recently. I have tried manually calibrating the RGB and depth streams but haven’t yet evaluated additional cases to confirm if my manual extrinsic parameters are accurate. Once I’ve completed my current tasks, I’ll continue with this project and update you on the results. Thank you for your patience.

MartyG-RealSense, 2024-10-28:
You are very welcome. It's no problem at all.  I look forward to your next report.  Good luck!

MartyG-RealSense, 2025-01-06:
Case closed due to no further comments received.

closed 2025-01-06 by MartyG-RealSense

MelodyAI, 2025-03-28:
> [@MartyG-RealSense](https://github.com/MartyG-RealSense)
>
> I apologize for my delayed response. I've been busy with my paper and interview preparations recently. I have tried manually calibrating the [...]

嘿兄弟，这个问题解决了吗？我和你出现了一样的问题，都是相机外参为0，导致无法使用align.process来对齐rgb和深度图，图形数组输出也是全为0的矩阵，官方提供的动态校准工具（dynamic calibration）也是报9901的错误，只能够使用on chip-calibration来校准深度内参，但是无法校准rgb内参，我用matlab得到了rgb的内参，但是不知道导入相机固件里面的方法，只有手动算出对齐矩阵来对齐，如果有好的方法，麻烦分享一下，谢谢。

MelodyAI, 2025-03-28:
@tangyipeng100 嘿兄弟，这个问题解决了吗？我和你出现了一样的问题，都是相机外参为0，导致无法使用align.process来对齐rgb和深度图，图形数组输出也是全为0的矩阵，官方提供的动态校准工具（dynamic calibration）也是报9901的错误，只能够使用on chip-calibration来校准深度内参，但是无法校准rgb内参，我用matlab得到了rgb的内参，但是不知道导入相机固件里面的方法，只有手动算出对齐矩阵来对齐，如果有好的方法，麻烦分享一下，谢谢。

tangyipeng100, 2025-03-28:
> [@tangyipeng100](https://github.com/tangyipeng100) 嘿兄弟，这个问题解决了吗？我和你出现了一样的问题，都是相机外参为0，导致无法使用align.process来对齐rgb和深度图，图形数组输出也是全为0的矩阵，官方提供的动态校准工具（dynamic calibration）也是报9901的错误，只能够使用on chip-calibration来校准深度内参，但是无法校准rgb内参，我用matlab得到了rgb的内参，但是不知道导入相机固件里面的方法，只有手动算出对齐矩阵来对齐，如果有好的方法，麻烦分享一下，谢谢。

我目前也是用标定板来标定，没有再输入固件中了，直接获取数据后手动对齐。

MelodyAI, 2025-03-29:
@tangyipeng100 请问一下，内参外参的类型你是如何定义的，手动对齐需要realsense的这三个函数rs2_deproject_pixel_to_point，rs2_transform_point_to_point，rs2_project_point_to_pixel，主要将depth的数据映射到rgb上，我两个内参和一个外参都得到了，但是不清楚这几个函数的内外参数据类型，你是把这几个函数也重新编写了吗？

tangyipeng100, 2025-11-04:
> 你好，请问解决这个问题了吗，我也遇到了这个问题，尝试了很多方法这个depth to rgb的外参还是全部是0

搞不定，我买的二手的，感觉是有些问题。可以去搜一个别人的realsense d435的外参，直接用，差的不大，实在不行只能自己标了。