realsenseai/librealsense #13309: Saving IMU calibration result successful or failed?

HappySamuel, 2024-08-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   |  5.16.0.1 | 
| Operating System & Version |   Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  5.15.0-119-generic                                         | 
| Platform                                 |  PC  |
| SDK Version                            |   2.55.1                           | 
| Language                            |  python                          | 
| Segment			|  Robot                           | 

### Issue Description
Originally, i faced issue that gyro readings are wrong when using realsense-ros as per post at here [Angular Velocities Output goes wrong with D435i & D455 Camera #3195](https://github.com/IntelRealSense/realsense-ros/issues/3195#issue-2495998937) . Therefore, I decided to do calibration first, so i followed the guidance `Intel® RealSenseTM Depth Camera D435i IMU Calibration` to perform callibration. However i received this output as below, which makes me unsure about the calibration result saved successfully or not.

```
Writing files:
accel_d435i.txt
gyro_d435i.txt
[-0.00063041 -0.0059655  -0.00089412]
[1000 1000 1000 1000 1000 1000]
using 6000 measurements.
[[ 1.02838524e+00 -1.18005699e-02 -4.06653076e-02]
 [ 8.82452241e-04  1.00761932e+00 -1.19618954e-02]
 [-1.96185968e-03  4.66613172e-03  1.02031606e+00]
 [ 7.51304689e-02  4.11106294e-01  1.60809909e-01]]
residuals: [11.15314642 19.75435653 69.71640366]
rank: 4
singular: [440.17937651 433.75173032 419.12008205  77.21076618]
norm (raw data  ): 9.635639
norm (fixed data): 9.805722 A good calibration will be near 9.806650
rs-imu-calibration.py:703: DeprecationWarning: tostring() is deprecated. Use tobytes() instead.
  outfile.write(imu_calib_table.astype('f').tostring())
Would you like to write the results to the camera? (Y/N)Y
Writing calibration to device.
  Device PID:  0B3A
  Device name:  Intel RealSense D435I
  Serial number:  017322071591
  Firmware version:  5.16.0.1
SUCCESS: saved calibration to camera.
Done.
Segmentation fault (core dumped)
```

MartyG-RealSense, 2024-08-30:
Hi @HappySamuel  If the **SUCCESS: saved calibration to camera.  Done** message is displayed before the segmentation fault occurs then it can be trusted that the calibration has been successfully saved to the camera.

HappySamuel, 2024-08-30:
OK, thanks for the clarification.

closed 2024-08-30 by HappySamuel