realsenseai/librealsense #13310: Malformed entry error appears when following installation instructions in a docker container.

ahiyantra, 2024-08-30:
I found a similar issue reported by someone else but it seems to have already been resolved. Their solution didn't work for me.

[link to the mentioned thread](https://github.com/IntelRealSense/librealsense/issues/4779)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Operating System & Version |  Windows 10 subsystem for Linux (Ubuntu 22) | 
| Kernel Version (Linux Only)    |  5.15.90.1-microsoft-standard-WSL2+ | 
| Platform                                 | PC |
| Language                            | bash | 

----------------------------------------------------------------------------------------------------

### Issue Description.

terminal output messages:
```
# apt-get update
E: Malformed entry 1 in list file /etc/apt/sources.list.d/librealsense.list (Component)
E: The list of sources could not be read.
```

ahiyantra, 2024-08-30:
It seems that some important linux packages ("lsb-release" & "tee") weren't working properly in the docker container. Reinstalling them fixed this issue.

closed 2024-08-30 by ahiyantra

MartyG-RealSense, 2024-08-30:
I'm pleased to hear that you found the solution.  Thanks very much for the update!