realsenseai/librealsense #13314: v.2.56.1 support for Ubuntu 24.04 / Linux 6.8 missing although promised

SammyRamone, 2024-09-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 24.04.| 
| Kernel Version (Linux Only)    |  6.8                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.56.1                          | 
| Language                            |  ROS 2 Wrapper| 
| Segment			|  Robot                          | 

### Issue Description
The current [pre-release](https://github.com/IntelRealSense/librealsense/releases/tag/v2.56.1) (v.2.56.1) claims to support Ubuntu 24.04 with kernel 6.8. However, when I tried to install it on my Ubuntu 24.04 with kernel 6.8 it fails when running the `./scripts/patch-realsense-ubuntu-lts-hwe.sh` script with:
`Unsupported distribution noble, kernel version 6.8.0-41-generic . The patches are maintained for Ubuntu 16/18/20 LTS`

There is an open PR draft https://github.com/IntelRealSense/librealsense/pull/13090 which addresses part of the issue, however it also did not work on my machine since `../scripts/realsense-metadata-noble-master.patch` is missing.

MartyG-RealSense, 2024-09-03:
Hi @SammyRamone  Some changes have just been made today in a PR at https://github.com/IntelRealSense/librealsense/pull/13236#issuecomment-2326207776 that include updates of kernel 6.8 support and the manual patch script for Ubuntu 24 support.

SammyRamone, 2024-09-03:
Thanks for the quick reply :)
Yes, you are right, the mentioned PR resolves this issue. I think I did not find it due to searching for "24.04" and not for "noble".
As soon as this PR is merged, this issue can be closed.

IRONCROWY, 2024-09-06:
Ok, it seems that i'm not the only one who meets this problem: `./scripts/patch-realsense-ubuntu-lts-hwe.sh  Unsupported distribution noble, kernel version 6.8.0-41-generic . The patches are maintained for Ubuntu 16/18/20 LTS`, i wonder when will the bug-fixed librealsense sdk version released?

IRONCROWY, 2024-09-06:
Oh, i found this modified branch working, nice, [https://github.com/IntelRealSense/librealsense/tree/aa0be527737a90728d65447054d34294ebb547b6](url)

IRONCROWY, 2024-09-06:
There is one more problem is that, this librealsense version is 2.56.0, which is not compatiable with the latest realsense-ros version.
`CMake Warning at CMakeLists.txt:121 (find_package):
  Could not find a configuration file for package "realsense2" that is
  compatible with requested version "2.56.1".

  The following configuration files were considered but not accepted:

    /usr/local/lib/cmake/realsense2/realsense2Config.cmake, version: 2.56.0

CMake Error at CMakeLists.txt:126 (message):

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases`

MartyG-RealSense, 2024-09-06:
Hi @IRONCROWY  Are you attempting to install the ROS wrapper from packages with a command such as `sudo apt install ros-humble-realsense2-*` please?

If you are, have you tried building the wrapper from source code with **colcon build** for use with your 2.56.0 librealsense version?

efagerberg, 2024-09-06:
Hello, we are also using realsense, we have new machines we would like to use with kernel 6.8 so we are hitting this same issue.

IRONCROWY, 2024-09-07:
> Hi @IRONCROWY Are you attempting to install the ROS wrapper from packages with a command such as `sudo apt install ros-humble-realsense2-*` please?
> 
> If you are, have you tried building the wrapper from source code with **colcon build** for use with your 2.56.0 librealsense version?

Oh, my ros2 version is jazzy, so, emmm, humble version cannot be installed, (maybe i should `sudo apt install ros-jazzy-realsense2-*`?, this command seems okay and the camere node does woking) this failure just happened when using colcon build the realsense-ros using 2.56.0 librealsense.

By the way, at first i use the 2.56.1 beta librealsense because it supports ubuntu 24 but actually not, so i applied [#13236 ] to install librealsense and eventually i saddly found that its version is 2.56.0......

MartyG-RealSense, 2024-09-07:
Ubuntu 24.04 / Kernel 6.8 users can try building the librealsense SDK from source code with CMake with the **RSUSB Backend** method.  RSUSB bypasses the kernel and so is not dependent on a particular Linux version or kernel version and does not require a patch script to be applied to the kernel.

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/13065#issuecomment-2217435785 succeeded with an RSUSB source code build on Ubuntu 24.04.

Instructions for an RSUSB installation can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

IRONCROWY, 2024-09-07:
> Ubuntu 24.04 / Kernel 6.8 users can try building the librealsense SDK from source code with CMake with the **RSUSB Backend** method. RSUSB bypasses the kernel and so is not dependent on a particular Linux version or kernel version and does not require a patch script to be applied to the kernel.
> 
> A RealSense user at [#13065 (comment)](https://github.com/IntelRealSense/librealsense/issues/13065#issuecomment-2217435785) succeeded with an RSUSB source code build on Ubuntu 24.04.
> 
> Instructions for an RSUSB installation can be found at [#9931 (comment)](https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692)

@MartyG-RealSense Thanks, both the sdk and ros are working after i remove all the old sdk and ros then reinstall them using RSUSB installation on Ubuntu 24.04.

MartyG-RealSense, 2024-09-07:
@IRONCROWY  That's great to hear that an RSUSB build on 24.04 succeeded for you!

MartyG-RealSense, 2024-09-09:
The kernel 6.8 / Ubuntu 24.04 support updates in PR https://github.com/IntelRealSense/librealsense/pull/13236 have now been merged into the **development** branch of the librealsense SDK.

https://github.com/IntelRealSense/librealsense/tree/development

MartyG-RealSense, 2024-09-13:
Case closed due to solutions achieved, updates merged into the SDK and no further comments received.

closed 2024-09-13 by MartyG-RealSense