realsenseai/librealsense #13317: I have a Realsense D435. It does not shows any RGB image (no frames received) and shows USB2.1 instead of USB 3 even after I tried all the method of troubleshooting found on the internet.

ChiroJoy101, 2024-09-03:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|746612070097|
|**Firmware Version**|5.16.0.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|736613020024|
|**Firmware Update Id**|736613020024|
|**Dfu Device Path**||

Please provide a description of the problem

ChiroJoy101, 2024-09-04:
Of course i did. none of this made any difference. I was getting depth image before. But as of today when I tried connecting it is also gone.

ChiroJoy101, 2024-09-04:
Update: The depth images is back. But the RGB is not

MartyG-RealSense, 2024-09-04:
Are you using the official 1 meter USB cable supplied with the camera or a cable of your own choice, please?

If you go to the Windows **Device Manager** interface and expand open the 'Cameras' category whilst the camera is plugged in, are there two RealSense drivers (Depth and RGB) listed there, or only the Depth driver?

[image]

ChiroJoy101, 2024-09-04:
No i am using my own cable, which worked fine up untill 4 days ago. And yes there are two realsense drivers listed. 
[image]

ChiroJoy101, 2024-09-04:
No. As i said the cable worked fine for months without any issue. I also have used multiple other cable from my colleagues but now them worked. I tried both in windows and ubuntu. Same result

MartyG-RealSense, 2024-09-05:
If you open the Windows 'Camera' app (findable by typing **camera** into the search box at the bottom of your Windows screen) and then click the 'Change Camera' icon when it opens to switch to the RGB sensor of your D435, does an RGB image display?

[image]

If the D435's RGB displays in the Camera app then it would demonstrate that the RGB sensor of the camera is functioning correctly and the No Frames Received problem is therefore likely related to a RealSense software issue and not a camera hardware fault.

ChiroJoy101, 2024-09-05:
i use a desktop, so there is no camera attached. However when i open the camera app it shows this. 
[image]

ChiroJoy101, 2024-09-05:
New development: In the device manager now it shows no RGB device also it is named as 430 instead of 435
[image]

MartyG-RealSense, 2024-09-05:
If a camera is shown as 430 instead of 435 then it typically means that the camera's firmware driver has not been able to detect the presence of the RGB sensor hardware component inside the camera.

It can though also occur if the firmware driver itself is damaged.  You could test this by downgrading the firmware to the previous version **5.15.1.0** to see if the RGB is then detected.

https://dev.intelrealsense.com/docs/firmware-releases-d400

MartyG-RealSense, 2024-09-13:
Hi @ChiroJoy101  Do you require further assistance with this case, please?  Thanks!

ChiroJoy101, 2024-09-13:
Currently i giving my finals, will contact with you soon. Thank You

MartyG-RealSense, 2024-09-13:
I look forward to your next report.  Good luck with the finals!

ChiroJoy101, 2024-09-19:
Hi, I am suing a new expensive usb 3.2 cable (type c to type c) and it is showing that the camera is using usb 3.2. But it is still D430 not d435, and in the device manager there is no rgb camera 
[image]

MartyG-RealSense, 2024-09-19:
Thanks very much for the update.

I will raise the issue with my Intel RealSense colleagues.  Thanks very much again for your patience.

MartyG-RealSense, 2024-09-26:
Hi again @ChiroJoy101  Regarding the comment above, could you let me know which country you are located in please?  Thanks!

ChiroJoy101, 2024-09-26:
Bangladesh

MartyG-RealSense, 2024-09-26:
Thanks very much.  I will pass the information to my Intel RealSense colleagues.

MartyG-RealSense, 2024-09-26:
Please send me an email at the address below so that we can discuss the next step in resolving your D430 problem.  Thanks!

martyx.grover@intel.com

ChiroJoy101, 2024-09-30:
Hi again, I have emailed you as you instructed. Is there any updates?

MartyG-RealSense, 2024-09-30:
I have not received the email, unfortunately.  Please try sending again to martyx.grover@intel.com

It does not have to be more than a few words long, just so that I have your email address so that I can reply with the update that I have about your case.

ChiroJoy101, 2024-09-30:
Hi, I have sent you the mail. Please confirm if you received it. Thanks

MartyG-RealSense, 2024-09-30:
Thanks so much, I have received it and responded.  Please reply by email.

closed 2024-10-20 by MartyG-RealSense