realsenseai/librealsense #13319: Using rs-convert to get frames (png) of the infrared streams of both cameras

RyuseiiSama, 2024-09-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D455 } | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |   Ubuntu 22.04 | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC |
| SDK Version                            |  v2.55.1                         | 
| Language                            |  python }                          | 
| Segment			|  others }                          | 

### Issue Description
Hello. I used the `realsense-viewer` to record 2 infrared streams on my D455 camera, specifically the left and right camera. Saved it as a `.bag` file. I'm trying to convert this bag file into images, using the rs-convert. Here is my input:
`rs-convert -p test -i 1.bag`
This only converts the stream of ONE camera into frames, while the other stream is completely ignored. The streams have to be synced, so recording twice is not an option. Any ideas? Do note I'm not the best at code so detailed explanations would help!

MartyG-RealSense, 2024-09-05:
Hi @RyuseiiSama  The alternative bag extraction tool **rs-bag2image** only extracts the left infrared stream too.

https://github.com/UnaNancyOwen/rs_bag2image/tree/master/rs_bag2image

I suspect that rs-convert uses the default stream profile of the camera, which would be depth, RGB and Infrared 1 (left).  To add support for Infrared 2 (Right) you would likely have to add additional C++ 'cfg' camera configuration lines at line 146 of the rs-convert.cpp script and then compile the adjusted program to create an executable program.

Infrared 2 is not enabled in the default stream profile because that stream is unsupported on USB 2.1 connections and so is only available on USB 3.

If there are only infrared streams recorded in your bag then the code to configure which streams are accessed by the rs-convert.cpp script  would look like this:

```
 rs2::config cfg;
cfg.enable_stream(RS2_STREAM_INFRARED, 1);
cfg.enable_stream(RS2_STREAM_INFRARED, 2);
cfg.enable_device_from_file(inputFilename.getValue());
pipe->start(cfg);
```

RyuseiiSama, 2024-09-10:
Hello @MartyG-RealSense. To implement the changes you suggested, i included `cfg.enable_stream(RS2_STREAM_INFRARED, 1); cfg.enable_stream(RS2_STREAM_INFRARED, 2);` after line 146. I made these changes after cloning the github repositery, and subsequently I recompiled and rebuilt the entire librealsense package. However, the output i get from `rs-convert -p test -i 1.bag` still only provide me with stream from one camera. For your information, the images come out in such a format: `test_Infrared 1.<numbers>.png`

I'm recording with `realsense-viewer` as seen below.
[image]

Are there any other changes i should make on `rs-convert.cpp` such that it would take the 2nd infrared stream into account?

MartyG-RealSense, 2024-09-10:
My recommendation would have been to try inserting those two cfg lines as you have already done.

Have you confirmed that the bag file contains both the left and right streams?  You can play it back by drag and dropping the bag file into the center panel of the RealSense Viewer.

RyuseiiSama, 2024-09-10:
I just tested it out. Recorded a new bag file, showed both infrared streams, converted using rs-convert, but only got infrared 1 stream of pictures and its meta data. I hope that my method of inserting the two cfg lines are correct.

MartyG-RealSense, 2024-09-10:
Your method of inserting the two cfg lines is appropriate.  I am not aware though of any past cases where both IR streams have been extracted, just the left IR.

MartyG-RealSense, 2024-09-16:
Hi @RyuseiiSama  Do you require further assistance with this case, please?  Thanks!

RyuseiiSama, 2024-09-19:
hi @MartyG-RealSense, sorry i havent gone about fixing it yet. you can go ahead and close it!

MartyG-RealSense, 2024-09-19:
It's no problem at all.  As you suggested, I will close the issue.  Thanks very much for the update!

closed 2024-09-19 by MartyG-RealSense