realsenseai/librealsense #13320: No Frames Received!

min577, 2024-09-05:
[image]
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|026322071679|
|**Firmware Version**|5.16.0.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|950323050899|
|**Firmware Update Id**|950323050899|
|**Dfu Device Path**||

Please provide a description of the problem

MartyG-RealSense, 2024-09-05:
Hi @min577  Please first try unplugging the micro-sized end of the USB cable from the side of the D435i camera, reversing the connector's orientation and plugging it back into the camera (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

HanaaFariz, 2024-09-10:
I had the same problem with the D455 when using the supplied cable. Changing the cable fixed it.

closed 2024-09-24 by MartyG-RealSense