realsenseai/librealsense #13323: D435i detected as USB 2.1 after firmware update, original USB3.1

ChiaXinLiang, 2024-09-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.16.0.1 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  v2.55.1                      | 
| Language                                | C++|
| Segment			| Robot                       | 

### Issue Description
D435i detected as USB 2.1 after firmware update, original USB3.1
Also, I try to downgrade my d435i to lower firmware also show usb2.1

MartyG-RealSense, 2024-09-06:
Hi @ChiaXinLiang  Have you tried unplugging the micro-sized end of the USB cable from the side of the camera, reversing the connector's orientation and plugging it back into the side of the camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

MartyG-RealSense, 2024-09-06:
Is the camera detected at all as 3.1 if you unplug the USB cable (the large end) from the computer and then re-insert it with a firm, quick motion and not a slow insertion?  A quick insertion can increase the chances of a successful USB 3 detection.

closed 2024-09-24 by MartyG-RealSense