realsenseai/librealsense #13324: Bug in number of frames extracted with pyrealsense2 Version 2.55.1.6486

checolag, 2024-09-06:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |          | 
| Platform                                 | PC |
| SDK Version                            |    2.55.1.6486                 | 
| Language                            |  Python | 
| Segment			|                 | 

### Issue Description
I am encountering a bug with the pyrealsense2 library, specifically version 2.55.1.6486. The issue is related to the returned frame number extracted from the RealSense camera through the function `get_frame_number`.
With the new version of the library, in the while loop (while:True), the library extracts the same frame two times since the frame number is actually the same. 
Instead with the old version 2.54.2.5684, it doesn't happen.
This is crucial as I'm using an if-statement to get out of the loop once all the images are extracted. 

example code:

`
 pipeline = rs.pipeline()
          config = rs.config()
          config.enable_device_from_file(bag_file)
          profile = pipeline.start(config)
          profile.get_device().as_playback().set_real_time(False)
          cur_frame_number=-1
          frames_list = []
          align = rs.align(rs.stream.color)
          while True:
              frames=pipeline.wait_for_frames()
              frameset = align.process(frames)
              color_frame= frames.get_color_frame()
              depth_frame = frames.get_depth_frame()
              depth_align_frame = frameset.get_depth_frame()
        
        if cur_frame_number < color_frame.get_frame_number():
            cur_frame_number=color_frame.get_frame_number()
            frames_list.append(cur_frame_number)
            color_image = np.asanyarray(color_frame.get_data())
            color_image = cv2.cvtColor(color_image, cv2.COLOR_BGR2RGB)

            depth_image = np.asanyarray(depth_frame.get_data())
            colorized_depth = np.asanyarray(depth_align_frame.get_data())
            colorized_align_depth = cv2.cvtColor(np.asanyarray(rs.colorizer().colorize(depth_align_frame).get_data()), cv2.COLOR_BGR2RGB)

            cv2.imwrite(str(color_dir / f"{cur_frame_number}.png"), color_image)
        else:
            break

`

MartyG-RealSense, 2024-09-06:
Hi @checolag  If you are experiencing repeating frame numbers then this may not necessarily be due to a bug but because of a recovery function of the librealsense sdk.  If it encounters a 'bad' frame then it will go back to the last known good frame and continue onwards from that point.  This recovery jump-back results in a duplicated frame number.  

So it is possible that your program is not performing as well in the 2.55.1 version as it did in 2.54.2 and bad frames are being produced that results in duplicated frame numbers.

After you upgraded to librealsense 2.55.1, have you updated your camera's firmware driver version to 5.16.0.1 (the recommended firmware for 2.55.1).

checolag, 2024-09-06:
Hi Marty, thanks for the fast response. I tried the same code using the two versions of the library and we the newer one I'm having this problem. how can i understand if it's due to the recovery function or a bug on the library?
Thank you again.

MartyG-RealSense, 2024-09-13:
Hi @checolag  Do you require further assistance with this case, please?  Thanks!

checolag, 2024-09-13:
No, thank you. I solved just downgrading the library.
Thanks again for the help!

closed 2024-09-13 by checolag

MartyG-RealSense, 2024-09-13:
You are very welcome.  It's great to hear that you achieved a solution.   Thanks very much for the update!