realsenseai/librealsense #13325: Using the callback function to obtain the IR left and right images of the realsense d455 camera results in an error.

robofans, 2024-09-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux (Ubuntu 20)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description

I use the callback method in the following code to obtain the left and right infrared images of the realsense d455 camera, but it get error.

```

RsD455Device::RsD455Device(bool manual_exposure, 
                           int skip_frames,
                           int webp_quality, 
                           double exposure_value):
      rsCmmonDevice("rsD455"){       

  std::cout << "open realsense d455" << std::endl;
  
  rsCmmonDevice::manual_exposure = manual_exposure;
  rsCmmonDevice::skip_frames = skip_frames;
  rsCmmonDevice::webp_quality = webp_quality;

  rs2::log_to_console(RS2_LOG_SEVERITY_ERROR);
  pipe = rs2::pipeline(context);

  config.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F);
  config.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F);
  config.enable_stream(RS2_STREAM_INFRARED, 1, RS2_FORMAT_Y8);    
  config.enable_stream(RS2_STREAM_INFRARED, 2, RS2_FORMAT_Y8);  

  // config.disable_stream(RS2_STREAM_DEPTH);

  if (context.query_devices().size() == 0) {
    std::cout << "Waiting for device to be connected" << std::endl;
    rs2::device_hub hub(context);
    hub.wait_for_device();
  }

  for (auto& s : context.query_devices()[0].query_sensors()) {
    std::cout << "Sensor " << s.get_info(RS2_CAMERA_INFO_NAME)
              << ". Supported options:" << std::endl;

    for (const auto& o : s.get_supported_options()) {
      std::cout << "\t" << rs2_option_to_string(o) << std::endl;
    }
  }
  auto device = context.query_devices()[0];
  device.hardware_reset();

  std::cout << "Device " << device.get_info(RS2_CAMERA_INFO_NAME) << " connected" << std::endl;
  sensor = device.query_sensors()[0];

  if (manual_exposure) {
    std::cout << "Enabling manual exposure control" << std::endl;
    sensor.set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
    sensor.set_option(rs2_option::RS2_OPTION_EXPOSURE, exposure_value * 1000);
  }
  // 
  std::cout << "finish open camera" << std::endl;
}

void RsD455Device::start() {

  auto callback = [&](const rs2::frame& frame) {

    if (auto fp = frame.as<rs2::motion_frame>()) {
      auto motion = frame.as<rs2::motion_frame>();
      if (motion && motion.get_profile().stream_type() == RS2_STREAM_GYRO &&
          motion.get_profile().format() == RS2_FORMAT_MOTION_XYZ32F) {
      } 

      else if (motion &&
                 motion.get_profile().stream_type() == RS2_STREAM_ACCEL &&
                 motion.get_profile().format() == RS2_FORMAT_MOTION_XYZ32F) {
        }
    } 
    else if (auto fs = frame.as<rs2::frameset>()) {
      std::vector<rs2::video_frame> vfs;
      for (int i = 0; i < NUM_CAMS; ++i) {
        rs2::video_frame vf = fs[i].as<rs2::video_frame>();
        if (!vf) {
          std::cout << "Weird Frame, skipping" << std::endl;
          return;
        }
        vfs.push_back(vf);
      }
      for (int i = 1; i < NUM_CAMS; ++i) {
        if (vfs[0].get_timestamp() != vfs[i].get_timestamp()) {
          return;
        }
      }
      if (frame_counter++ % skip_frames != 0) {
        return;
      }

      for (int i = 0; i < NUM_CAMS; i++) {
        const auto& vf = vfs[i];

        int64_t t_ns = vf.get_timestamp() * 1e6;
      }
    }

    // std::cout << "frame num = " << frame_num << std::endl;
    // frame_num += 1;
    return;
  };

  

  profile = pipe.start(config, callback);
}

void RsD455Device::stop() {
  std::cout << "stop d455" << std::endl;
}

bool RsD455Device::setExposure(double exposure) {
  if (!manual_exposure) return false;
  sensor.set_option(rs2_option::RS2_OPTION_EXPOSURE, exposure * 1000);
  return true;
}
void RsD455Device::setSkipFrames(int skip) { skip_frames = skip; }

void RsD455Device::setWebpQuality(int quality) { webp_quality = quality; }

int main(void){

  RsD455Device d455_device(false, 1, 90, 10.0);
  

  d455_device.start();

    while(1);
    return 0;
}

```

the error is 
[image]

if I only use "profile = pipe.start(config)" how can i get two ir frame sametime?

think you

MartyG-RealSense, 2024-09-07:
Hi @robofans  As the error reports **No such device**, I would recommand first trying commenting out the `device.hardware_reset();` line.  This is because when a camera reset is initiated, the camera disconnects and then attempts to reconnect, a process that takes 2 to 3 seconds to complete.  But a camera is not always detected during the re-connection attempt after disconnection.

MartyG-RealSense, 2024-09-13:
Hi @robofans  Do you require further assistance with this case, please?  Thanks!

robofans, 2024-09-13:
it run ok after  trying commenting out the device.hardware_reset(), Thanks!

closed 2024-09-13 by robofans