realsenseai/librealsense #13326: D455 No realsense devices were found!

hong-cr, 2024-09-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455  | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Ubuntu 22.04  | 
| Kernel Version (Linux Only)    | 5.15                                         | 
| Platform                                 |  NVIDIA Jetson Orin Nano  |
| SDK Version                            |  2.55                          | 
| JetPack                           |       6.0                 | 
| L4T		| 36.2                        | 

### Issue Description

Hello~, I faced few problems. And i couldn't fixed it yet

<Project PurPose : Object Recognition with Orin Nano (Jetpack 6.0) Using YOLOv8, OpenCV with Cuda,  Ros2 Humble + rviz2  and RealSense>

<Procedure Sector>

■ Install Jetpack 6.0 on Jetson

■ Install Ros2 Humble

■ install Pytorch and torchvision

■ Build OpenCV with Cuda

■ Install Realsense - Ros Wrapper(Install Realsense SDK (use buildscript.sh))

■ Install Realsense - Yolov8 Wrapper

Ok. Let's Start.

1. Install Jetpack 6.0 on Jetson

<> Success

2. Install Ros2 Humble

--> I follow this WebPage "https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debs.html"

locale  # check for UTF-8

sudo apt update && sudo apt install locales
sudo locale-gen en_US en_US.UTF-8
sudo update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
export LANG=en_US.UTF-8

locale  # verify settings

sudo apt install software-properties-common
sudo add-apt-repository universe

sudo apt update && sudo apt install curl -y
sudo curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg

echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | sudo tee /etc/apt/sources.list.d/ros2.list > /dev/null

sudo apt update
sudo apt upgrade
sudo apt install ros-humble-desktop

source /opt/ros/humble/setup.bash

+ Test : Talker - Listener test also finished

<Co> Success

[... 163 lines cut ...]

--> Error / Solve
[image]
[image]

● Source environment

ROS_DISTRO=humble
source /opt/ros/humble/setup.bash 
cd ~/ros2_ws
. install/local_setup.bash

--> GO connect to realsense camer D455 to Jetson Orin Nano

● Usage

ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 pointcloud.enable:=true

--> Error
[image]
[image]
[image]
[image]

So... What is problems?? plz Help Me, i'm desperate. No realsense devices....

MartyG-RealSense, 2024-09-08:
Hi @hong-cr  The current latest librealsense SDK release **2.56.1** introduces official support for JetPack 6.  Because 2.56.1 is currently an interim beta release, it can only be compiled from source code.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.56.1

If you would prefer to use 2.55.1, a problem with using JetPack 6 is that it does not support a function called **hiddraw** and so RealSense cameras equipped with an IMU - such as D455 - will not work correctly with it, often not being detected.

IMU support can be added by building the **MIPI driver** at the link below.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

I would not recommend using this method though due to the complexity of the installation process.

*******

An alternative method is to install librealsense with the **libuvc backend** and make modifications to the build to build with CUDA, like you described above.  The librealsense SDK should then be able to bypass the kernel.

I not though that before you performed the libuvc backend procedure, you used the installation instructions at installation_jetson.md to install librealsense from packages by installing librealsense2-utils and librealsense2-dev.  This should not be done.

The reason is that you are creating two separate installations of librealsense on the same computer, one installed from packages (the installation_jetson instructions) and another installed from source code (the libuvc backend instructions).  These two separate versions of librealsense could conflict with each other.  There is also the possibility of creating two sets of **udev** device handling rules that conflict with each other.

I would recommend uninstalling the Jetson package version of librealsense using the command below to see whether your libuvc backend source code installation of the SDK is then able to work.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

hong-cr, 2024-09-08:
Thank you! then if i just use

 "sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE"

"sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u"

"sudo apt-get install librealsense2-utils"
"sudo apt-get install librealsense2-dev"

"wget https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh"

and open that .sh file, modify line 46 for add '-DFORCE_RSUSB_BACKEND=ON' '-DBUILD_WITH_CUDA=true', and then "./libuvc_installation.sh" 

is it right?

MartyG-RealSense, 2024-09-08:
You only need to use the libuvc_installation.sh script and edit line 46.  The package installation is not necessary for installation on Jetson if you are using the libuvc script.

If you need to install and use the realsense-viewer tool then please use the CMake build command below.

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=TRUE DBUILD_GRAPHICAL_EXAMPLES=TRUE -DFORCE_RSUSB_BACKEND=ON -DBUILD_WITH_CUDA=ON`

hong-cr, 2024-09-08:
i did like that, before use "sudo ./libuvc_installation.sh", i did "dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge" and then use "sudo ./libuvc_installation.sh" , but this err comes

CMake Deprecation Warning at CMakeLists.txt:16 (MESSAGE):
  FORCE_LIBUVC, FORCE_WINUSB_UVC and ANDROID_USB_HOST_UVC are deprecated, use
  FORCE_RSUSB_BACKEND instead

-- Info: REALSENSE_VERSION_STRING=2.55.1
-- Setting Unix configurations
-- Info: Building with CUDA requires CMake v3.8+
-- The CUDA compiler identification is unknown
CMake Error at CMake/cuda_config.cmake:3 (enable_language):
  No CMAKE_CUDA_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CUDACXX" or the CMake cache entry CMAKE_CUDA_COMPILER to the full
  path to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMake/global_config.cmake:63 (include)
  CMakeLists.txt:46 (global_set_flags)

-- Configuring incomplete, errors occurred!
See also "/root/librealsense_build/librealsense-master/build/CMakeFiles/CMakeOutput.log".
See also "/root/librealsense_build/librealsense-master/build/CMakeFiles/CMakeError.log".

I try to find solution, usually this problem is related with PATH, so 
ubuntu:~/Downloads$ sudo nano ~/.bashrc

and then i added 

export PATH=$PATH:/usr/local/cuda-12.2/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/cuda-12.2/lib64
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
export CUDACXX=/usr/local/cuda/bin/nvcc
export CMAKE_CUDA_COMPILER=/usr/local/cuda/bin/nvcc

@ubuntu:~$ nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Aug_15_22:08:11_PDT_2023
Cuda compilation tools, release 12.2, V12.2.140
Build cuda_12.2.r12.2/compiler.33191640_0

.. still that error comes, i really don't know what is problems...

hong-cr, 2024-09-08:
and i saw "With librealsense2-dev package installed, you can compile an application with librealsense using g++ -std=c++11 filename.cpp -lrealsense2 or an IDE of your choice. To get started with RealSense using CMake check out [librealsense/examples/cmake](https://github.com/IntelRealSense/librealsense/tree/master/examples/cmake)" this comment in web

should i do something? or i don't need to care about that

hong-cr, 2024-09-08:
What the Fck, i just used "./libuvc_installation.sh"... it works!, i can't understand, why if i use "sudo ./libuvc_installation.sh" it didn't work... Ha...i wanna die. Now i'm just waiting build the .sh file

hong-cr, 2024-09-08:
OK..build success... and but when i run the code "ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=1280x720x30 pointcloud.enable:=true" below this errors comes... how can i solve it?

"[realsense2_camera_node-1]  09/09 08:38:10,973 WARNING [281473248258272] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  09/09 08:38:10,984 WARNING [281473248258272] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [WARN] [1725838717.037569521] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1] [WARN] [1725838720.544333081] [camera.camera]: No stream match for pointcloud chosen texture Process - Color"

MartyG-RealSense, 2024-09-09:
Im pleased to hear that you were successful with your build.  The **libuvc_installation.sh** build script is often successful on Jetsons that cannot detect IMU-equipped RealSense cameras (which are called 'HID' devices) when using JetPack 6 due to its lack of 'hiddraw' support.

**control_transfer returned error, index: 768, error: Resource temporarily unavailable** indicates a communication problem between the camera and computer, typically related to the USB port or USB cable.  For example, new frames might not be arriving from the camera fast enough.

**No stream match for pointcloud chosen texture Process - Color** means that RGB color frames are missing, likely because of frame drops occurring.  This could be related to the communication problem indicated by the 'control_transfer returned' warning.

Do the number of warnings / errors reduce if you use a lower resolution for the deptjh profile such as 640x480 or change the 1280x720 FPS from 30 to 5

Sometimes when these warnings occur, it is still possible to display the data successfully in RViz.

hong-cr, 2024-09-10:
yes! i saw https://github.com/IntelRealSense/librealsense/issues/13130. and it really succeess to display the data in rviz2! even that erros comes, I can ignore!.. So thank you for ur reply! Now i can operate the camera. It's really pleasure! Thank you very much!

closed 2024-09-10 by hong-cr

MartyG-RealSense, 2024-09-10:
You are very welcome.  It is excellent to hear that you have succeeded.  Thanks very much for the update!

SnoopyGeeek, 2024-11-25:
> Hi @hong-cr The current latest librealsense SDK release **2.56.1** introduces official support for JetPack 6. Because 2.56.1 is currently an interim beta release, it can only be compiled from source code.
> 
> https://github.com/IntelRealSense/librealsense/releases/tag/v2.56.1
> 
> If you would prefer to use 2.55.1, a problem with using JetPack 6 is that it does not support a function called **hiddraw** and so RealSense cameras equipped with an IMU - such as D455 - will not work correctly with it, often not being detected.
> 
> IMU support can be added by building the **MIPI driver** at the link below.
> 
> https://github.com/IntelRealSense/realsense_mipi_platform_driver
> 
> I would not recommend using this method though due to the complexity of the installation process.
> 
> An alternative method is to install librealsense with the **libuvc backend** and make modifications to the build to build with CUDA, like you described above. The librealsense SDK should then be able to bypass the kernel.
> 
> I note though that before you performed the libuvc backend procedure, you used the installation instructions at installation_jetson.md to install librealsense from packages by installing librealsense2-utils and librealsense2-dev. This should not be done.
> 
> The reason is that you are creating two separate installations of librealsense on the same computer, one installed from packages (the installation_jetson instructions) and another installed from source code (the libuvc backend instructions). These two separate versions of librealsense could conflict with each other. There is also the possibility of creating two sets of **udev** device handling rules that conflict with each other.
> 
> I would recommend uninstalling the Jetson package version of librealsense using the command below to see whether your libuvc backend source code installation of the SDK is then able to work.
> 
> `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

I solved it by using the command: 
> `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

Many Thx!

MartyG-RealSense, 2024-11-25:
You are very welcome, @SnoopyGeeek - I'm pleased to hear that you achieved a solution.  Thanks very much for the update!