realsenseai/librealsense #13333: Need to simulate RGBD Realsense camera in Gazebo 11 (with ROS2 humble)

girishsmurthyIntel, 2024-09-11:
Hi @MartyG-RealSense ,
     I would like to simulate realsense RGBD camera in Gazebo11 (with ROS2 Humble). Do you suggest any plugins or details which I can use for the same. 

Rgds,
Girish

MartyG-RealSense, 2024-09-11:
There is a RealSense Gazebo plugin but it has not been updated since ROS2 Foxy.

https://github.com/pal-robotics/realsense_gazebo_plugin/tree/foxy-devel

https://github.com/introlab/rtabmap_ros/issues/1070 has a discussion about Gazebo simulation with Humble and a D435i camera.

MartyG-RealSense, 2024-09-16:
Hi @girishsmurthyIntel  Do you have an update about this case that you can provide, please?  Thanks!

girishsmurthyIntel, 2024-09-17:
Hi @MartyG-RealSense, We are continuing to work on this. I will provide an update based on these experiments.

MartyG-RealSense, 2024-09-17:
Thanks very much, @girishsmurthyIntel - I look forward to your next report.  Good luck!

girishsmurthyIntel, 2024-10-15:
Hi @MartyG-RealSense, We did not continue to use the realsense gazebo plugin and used the default depth cameras of Gazebo alternatively. Hence I have no further updates. Request you to kindly close this thread. Thank you.

closed 2024-10-15 by girishsmurthyIntel