realsenseai/librealsense #13340: D415: RuntimeError: hwmon command 0x80( 8 3 0 8 ) failed (response -7= HW not ready)

Hasnain1997-ai, 2024-09-13:
I am have a realsense d415 camera
I want to auto calibrate it

I am using 
https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/depth_auto_calibration_example.py
Code for calibration 

WHen I run the code I am getting the error

Starting UCAL...
Starting On-Chip calibration...
	Speed:	medium
	Scan:	intrinsic
Traceback (most recent call last):
  File "depth_auto_calibration_example.py", line 285, in <module>
    main()
  File "depth_auto_calibration_example.py", line 87, in main
    run_on_chip_calibration(args.onchip_speed, args.onchip_scan)
  File "depth_auto_calibration_example.py", line 121, in run_on_chip_calibration
    table, health = adev.run_on_chip_calibration(json_content='', timeout_ms=5000)
RuntimeError: hwmon command 0x80( 8 3 0 8 ) failed (response -7= HW not ready)

I am new and did not know how to solve it 
please help

MartyG-RealSense, 2024-09-13:
Hi @Hasnain1997-ai  Which RealSense SDK (librealsense) version and camera firmware driver version are you using, please?

The Intel calibration guide linked to below provides information about how an error can occur if you are calibrating a D415 against a white wall. 

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

 I have quoted the relevanr section below.

********

We developed a special calibration mode for the case of pointing the D400 Series cameras at a white wall that has no texture while the laser pattern projector is turned on.  It turns out that this special case can be troublesome due to the semi-regular laser pattern of the D415 projector. 

This mode of operation can be selected under the “speed” settings and is called “white wall”.  It is also highly recommended that the “High Accuracy depth settings” be used during any white-wall calibration run. If not, then the ASIC will most likely return an error. 

However, to be clear, we do not recommend using a white wall for self-calibration for the D415.  A textured surface with the projector turned off using medium speed is the recommended method for on-chip calibration of D415 cameras. 

**********

The on-chip calibration function can also be accessed in the RealSense Viewer tool if you are able to run the tool.  You can find it by going to the **More** option near the top of the Viewer's options side-panel and selecting **On-Chip Calibration** from its drop-down menu.

Do you also experience this RuntimeError in the Viewer?

As an alternative to performing calibration, you can reset the camera to its factory-new default calibration in the Viewer using instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

Hasnain1997-ai, 2024-09-14:
Using this code the info i am getting is
`import pyrealsense2 as rs

def get_realsense_info():
    try:
        # Create a context object
        ctx = rs.context()
        
        # Get the list of connected devices
        devices = list(ctx.query_devices())
        
        if len(devices) == 0:
            print("No RealSense devices connected.")
            return
        
        print(f"Number of connected RealSense devices: {len(devices)}")
        
        # Print info for each connected device
        for i, dev in enumerate(devices):
            print(f"\nDevice {i+1} info:")
            print(f"    Name: {dev.get_info(rs.camera_info.name)}")
            print(f"    Firmware Version: {dev.get_info(rs.camera_info.firmware_version)}")
            print(rs.__version__)
    except Exception as e:
        print(f"An error occurred: {str(e)}")

if __name__ == "__main__":
    get_realsense_info()`
    
    
Number of connected RealSense devices: 1

Device 1 info:
    Name: Intel RealSense D415
    Firmware Version: 5.16.0.1
and pyrealsense2 version is 2.55.1.6486

Hasnain1997-ai, 2024-09-14:
Also upon selecting wall i get the same error

Starting UCAL...
Starting On-Chip calibration...
	Speed:	wall
	Scan:	intrinsic
Traceback (most recent call last):
  File "auto_calib.py", line 285, in <module>
    main()
  File "auto_calib.py", line 87, in main
    run_on_chip_calibration(args.onchip_speed, args.onchip_scan)
  File "auto_calib.py", line 121, in run_on_chip_calibration
    table, health = adev.run_on_chip_calibration(json_content='', timeout_ms=5000)
RuntimeError: hwmon command 0x80( 8 3 0 8 ) failed (response -7= HW not ready)

Hasnain1997-ai, 2024-09-14:
Also, I commented the on-chip Self-Calibration and run_tare_calibration from the code, and only run the run_focal_length_calibration
and It runs successfully with the below output
`Starting UCAL...
Starting Focal-Length calibration...
	Target Size:	(175, 100)
	Side Adjustment:	right_only
Focal-Length calibration finished
	Ratio:	0.04020035266876221
	Angle:	-7.758955001831055
UCAL finished successfully
`

But When I un-commented the  OnChip Self Calibration and ran the code it is giving the same error

MartyG-RealSense, 2024-09-14:
On some computers the self-calibration tool will only work up until librealsense version **2.49.0** and not in 2.50.0 or newer, where the **HW not ready** error that you experienced occurs.  This is due to a change in the design of the tool's interface in 2.50.0.

If the On-Chip tool will not work for you then there is an alternative tool called **Dynamic Calibration** that can be installed on Ubuntu using instructions on page 14 onwards of the Dynamic Calibration user guide.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

Hasnain1997-ai, 2024-09-14:
[image]
I am unable to downgrade pyrealsense2

I want to do calibration using python, because my use case require python based calibration

MartyG-RealSense, 2024-09-15:
Did you install librealsense 2.49.0 before attempting a pip install of pyrealsense2 2.49.0.3474 please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.49.0