realsenseai/librealsense #13341: LibRealsense 2.56 and JetsonPack 6.0

djaniel, 2024-09-13:
Hello

I am trying to setup my Realsense camera on my Jetson nano Orin board. Here is the info about my system:

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   | 5.16.0.1   | 
| Operating System & Version |  Ubuntu 22, jetsonpack 6.0  | 
| Kernel Version (Linux Only)    |  5.15.136-rt-tegra               | 
| Platform                                 | NVIDIA Jetson Orin Nano  |
| SDK Version                            |   2.56.1                           | 
| Language                            |    Python/C                       | 
| Segment			|  Robot                          | 

### Issue Description
According to the docs, it is first necessary to patch the kernel. And so I tried to patch the kernel using the script finishing with Linux for Tegra (L4T).

```bash
./scripts/patch-release-ubuntu-L4T.sh
```
but I get an error:

```bash
Copying the patched modules to destination
...librealsense-2.56.1
Move the modified modules into the modules tree
Insert the modified kernel modules
Unload kernel modules
Load modified kernel modules
modprobe: ERROR: could not insert 'kfifo_buf': Exec format error
Failed to reload module kfifo_buf. error type 1. Operation is aborted.
```
I understand kfifo_buf is part of the Linux kernel, but being no kernel expert I decided to ask first.

The compilation of the Realsense Library goes ok, but when I launch `rs-enumerate-devices` I don't see my camera. In contrast, `lsusb` lists my camera.
 
Can you please tell me what may be wrong with my setup?

Thank you in advance,

MartyG-RealSense, 2024-09-13:
Hi @djaniel  RealSense cameras equipped with an IMU, such as D435i, will not work correctly with JetPack 6 because that JP version removed an instruction called **hiddraw**.

A solution that is often successful on Jetsons with JP6 is to build the librealsense SDK from source code with the **libuvc backend** script at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

This method bypasses the kernel, meaning that a kernel patch script does not need to be applied to the kernel.

When using the **libuvc_installation.sh** script on a Jetson, edit line 46 to add the build flag **-DBUILD_WITH_CUDA=ON** so that the SDK is built with CUDA support for Jetson.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

djaniel, 2024-09-16:
Hello @MartyG-RealSense 

Thank you for your fast reply. Everything worked fine after I used `FORCE_LIBUVC=true` when calling cmake.
Unexpectedly, I downloaded the release ZIP file with version 2.56.1, and after installing the library I see:
```bash
$pkg-config --modversion realsense2
2.55.1
```
Thanks again.

MartyG-RealSense, 2024-09-16:
You are very welcome.  I'm pleased to hear that you succeeded.  Thanks very much for the update!

2805651606, 2024-09-24:
能打开realsense-viewer后，出现
Timestamp: 1727182773055.368408
Severity: Error
Description: Motion Module force pause   我的板子是orin nx，jetdonpack 6.0,内核5.15.136-rt-tegra，ubuntu22,ros-humble

MartyG-RealSense, 2024-09-24:
Hello @2805651606  If you need to install librealsense with support for graphics and install example programs with graphics such as realsense-viewer then please add additional build flags to line 46.

`-DBUILD_EXAMPLES=TRUE -DBUILD_GRAPHICAL_EXAMPLES=TRUE`

***********

你好@2805651606 如果您需要安装支持图形的 librealsense 并安装示例程序（如 realsense-viewer），那么请在第 46 行添加额外的构建标志。

-DBUILD_EXAMPLES=TRUE -DBUILD_GRAPHICAL_EXAMPLES=TRUE

2805651606, 2024-09-24:
> Hello @2805651606 If you need to install librealsense with support for graphics and install example programs with graphics such as realsense-viewer then please add additional build flags to line [...]

但是用realsense-viewer后，出现
imestamp: 1727182773055.368408
Severity: Error
Description: Motion Module force pause

2805651606, 2024-09-24:
[image]不知道怎么回事，

MartyG-RealSense, 2024-09-24:
Usually a Motion Module Force Pause in realsense-viewer is an annoyance rather than a serious problem, because enabling the Motion Module again after the force pause typically makes the IMU activate normally.

Does the IMU start if you enable Motion Module again after the error please?

*********

通常，realsense-viewer 中的运动模块强制暂停是一种烦恼，而不是严重的问题，因为强制暂停后再次启用运动模块通常会使 IMU 正常激活。

如果在错误发生后再次启用运动模块，IMU 是否会启动？

2805651606, 2024-09-25:
> Usually a Motion Module Force Pause in realsense-viewer is an annoyance rather than a serious problem, because enabling the Motion Module again after the force pause typically makes the IMU activate [...]
不会，会出现黑屏
[image]
[image]

VunkersChenping, 2024-09-25:
Hello @MartyG-RealSense I've been looking at this thread and I have a similar problem in the file ./scripts/patch-release-ubuntu-L4T.sh.
I have a problem with Module.symvers

make: *** No rule to make target 'mrproper'. Stop.
cp: cannot stat '/usr/src/linux-headers-5.15.0-1016-nvidia-tegra-igx-ubuntu22.04_aarch64/kernel-5.10/Module.symvers': No such file or directory

I have updated the kernel and tried other solutions in my /usr/src there is no directory similar to linux-headers-5.15.0-1016-nvidia-tegra-igx-ubuntu22.04_aarch64
I only got theses:
linux-headers-5.15.0-1015-nvidia-tegra-igx
linux-headers-5.15.0-1016-nvidia-tegra-igx
linux-headers-tegra-oot-jetson-5.15.0-1015-nvidia-tegra-igx
linux-headers-tegra-oot-jetson-5.15.0-1016-nvidia-tegra-igx
linux-nvidia-tegra-igx-headers-5.15.0-1015
linux-nvidia-tegra-igx-headers-5.15.0-1016

i had found a solution for that editing  ./scripts/patch-release-ubuntu-L4T.sh. putting my own path like that
#Reuse existing module.symver                                                                                          
kernel_ver=`uname -r`                                                                                                   
LINUX_HEADERS_NAME="linux-headers-${kernel_ver}-ubuntu18.04_aarch64"                                                   
if [ "5.0.2" = "$PATCHES_REV" ]; 
then      
LINUX_HEADERS_NAME="linux-headers-${kernel_ver}-ubuntu22.04_aarch64"                                                  
fi                                                                                                                      
cp /usr/src/linux-headers-tegra-oot-jetson-5.15.0-1016-nvidia-tegra-igx/hwpm/Module.symvers .                          
 cp /usr/src/linux-headers-5.15.0-1015-nvidia-tegra-igx/Makefile . 

but now i got this error

make: *** No rule to make target 'mrproper'.  Stop.
Makefile:1084: scripts/Makefile.extrawarn: No such file or directory
make: *** No rule to make target 'scripts/Makefile.extrawarn'.  Stop.

Im using intel realsense 415, ubuntu 22.04LTS.

MartyG-RealSense, 2024-09-25:
@2805651606  Are the depth and RGB streams in the realsense-viewer working correctly or do you just get a black screen with thise too?

realsense-viewer 中的深度和 RGB 流是否正常工作，还是也只出现黑屏？

MartyG-RealSense, 2024-09-25:
Hi @VunkersChenping  If you install librealsense using the libuvc_installation.she script then you do not need to apply the L4T patch script to the kernel, as a libuvc build of librealsense bypasses the kernel.

VunkersChenping, 2024-09-26:
@MartyG-RealSense thx for you help i successfully installed id

MartyG-RealSense, 2024-09-26:
@VunkersChenping  It's great to hear that you were successful - thanks very much for the update about your particular situation!

MartyG-RealSense, 2024-10-03:
@2805651606   Do you require further assistance with this case please?  Thanks!

您需要进一步协助解决此案吗？谢谢！

closed 2024-10-20 by MartyG-RealSense