realsenseai/librealsense #13342: Is their any way to get the camera pose information per framw for D457 with ubantu20 kernel 5.3 on PC?

Dehaoq, 2024-09-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-09-14:
Hi @Dehaoq  "Pose" is usually described as the cameras rotation **and** its position in the real world.  You can obtain the rotation from the camera's IMU, but analyzing its own position in the real world is not a supported feature.

If it is only the camera rotation that you need and you require it to update continuously then the C++ **rs-motion** example program may be helpful.

https://github.com/IntelRealSense/librealsense/tree/master/examples/motion

If you have installed the librealsense SDK with tools and examples included then you can look in the Ubuntu folder location **usr/local/bin** for a pre-made executable version of rs-motion.

*********

The rotation can also be checked in the **RealSense Viewer** tool by enabling its 'Motion Module' in the Viewer's options side-panel and hovering the mouse cursor over the panel that is displaying the 'Gyro' rotation angle.

Dehaoq, 2024-09-14:
Hi Marty @MartyG-RealSense , thank you so much for your reply. I do not need the exact position in the realword, relative position is good, like what we can get from T265. I have get the info from the blow code. 
        accel_data = accel_frame.as_motion_frame().get_motion_data()
        gyro_data = gyro_frame.as_motion_frame().get_motion_data() 
But if I generate the pose only based on the IMU data, it is not accurate. I checked that there are some methods which combine the IMU data and rgbd data. Do you have any suggestions that I can get accurate pose information?

MartyG-RealSense, 2024-09-14:
The only method that I know of for directly obtaining a relative pose of a RealSense 400 Series camera is with ROS.  At the link below, a RealSense user created an adaptation of Intel's D435i SLAM guide that added localization with RTABMAP and obtaining the relative position of the camera from a recorded pointcloud map.

https://shinkansan.github.io/2019-UGRP-DPoom/SLAM

Alternatively, if you know the distance of an observed object from the camera and its height above / below the camera and can calculate its height - as described at https://github.com/IntelRealSense/librealsense/issues/9297 - then you may be able to work out the camera's 3D real-world position relative to the real-world position of that observed object.

If you have a T265, you could also place the D457 and T265 on a mount together and use the T265's pose stream to provide the position of the D457 (since they will have the same position because they are sharing a mount).

Dehaoq, 2024-09-14:
Thank you, Marty! Your answer also remind me another question. Can D456 get the real distance, not just a depth color map?

MartyG-RealSense, 2024-09-14:
Yes, all RealSense 400 Series cameras can output the real-world distance in meters.  The RealSense SDK's example program **rs-hello-realsense** demonstrates this by outputting the distance in meters as a plain text value.

closed 2024-10-03 by MartyG-RealSense