realsenseai/librealsense #13343: raspberry4 cant get the data from l515 with ubuntu22

chengfengpiao, 2024-09-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  | 
| Firmware Version                   | 1.5.8.1 | 
| Operating System & Version |  Ubuntu 22  | 
| Kernel Version (Linux Only)    |  5.15.0-1062-raspi                                      | 
| Platform                                 |Raspberry Pi 4B 8GB|
| SDK Version                            |  2.54.2       | 
### Issue Description
I follow the instruction with this  https://raw.githubusercontent.com/IntelRealSense/librealsense/master/scripts/libuvc_installation.sh .But when I connect the l515 with raspberry (before this I try the same cable with PC and it works well)and launch the realsense-viewer it will show 

(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11(or 0 or b)
...

because I want to collect the data from l515,so I cant choose a newer version
could you help me sovle the question?

MartyG-RealSense, 2024-09-14:
Hi @chengfengpiao  The libuvc_installation.sh script will install the latest version of the librealsense SDK.  Support for the L515 was removed in 2.54.1 as it is a retired camera model.  If you use **2.51.1** then you should be able to access the camera.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

In regard to how you could install the old librealsense version with the libuvc script instead of the latest one, a possibility may be to first download and unzip the source code for 2.51.1.  Then copy the script file into the 'root' directory of the source code folder, remove lines 37-39 from the script and then launch the script.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L37-L39

The 'root' directory of the source code folder is the one with the large list of folders and files.
 
[image]

chengfengpiao, 2024-09-16:
thanks for your reply.In fact, i have installed **2.54.2** in my PC and nvidia jetson orin (both are Ubuntu 22.0.4), they all works well, and I can collect data through realsense-ros 4.54.1 , the problem is that my raspberry 4's system is also **Ubuntu 22.0.4** ,but the 2.54.2 can't work wel l,and the 2.51.1 can't be installed because of the Ubuntu22. I don't know the difference between the orin and raspberry .  I will try the ubuntu 20 with 2.51.1 next week :)

MartyG-RealSense, 2024-09-16:
RealSense cameras typically have more problems on Raspberry Pi boards than with Nvidia Jetson.  For example, usually the best that you can get from a Pi is two streams (such as depth and RGB) with no alignment / pointclouds / post-processing, and the IMU may also not work.  There can also be problems with getting the ROS wrapper to work correctly.  So I would recommend using the Jetson board if you a choice between the Pi and Orin.

closed 2024-09-23 by chengfengpiao