realsenseai/librealsense #13349: rs-server and realsense2-net install issue

kacper-olszewski, 2024-09-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435if | 
| Firmware Version                   |     -   | 
| Operating System & Version |   Win 10 /Raspbian  | 
| Kernel Version (Linux Only)    |     6.6.47+rpt-rpi-v8  | 
| Platform                                 | PC/Raspberry PI   |
| SDK Version                            |  2.53.1                         | 
| Language                            |  C++                       | 
| Segment			|  others                      | 

### Issue Description
Hello, I am trying to set up my Raspberry Pi to stream a RealSense camera image over ethernet and view the stream in the RealSense Viewer. I am connected to the Raspberry Pi via SSH and have the RealSense SDK installed on my PC. However, when I attempt to add a network device in the RealSense Viewer, I get the following message:

"To enable RealSense device over network, please build the SDK with CMake flag -DBUILD_NETWORK_DEVICE=ON. This binary distribution was built with network features disabled."

To address this, I tried building the SDK with the necessary flags to enable the network module using the following command on my Raspberry Pi:
cmake .. -DBUILD_NETWORK_DEVICE=ON -DFORCE_RSUSB_BACKEND=ON
However, during the CMake configuration, I encountered the following warning:

CMake Warning: Manually-specified variables were not used by the project: BUILD_NETWORK_DEVICE

As a result, the rs-server and realsense2-net module were not built, and I was unable to set up the network camera functionality.

MartyG-RealSense, 2024-09-16:
Hi @kacper-olszewski  In the information at the top of this issue, your librealsense version is listd as 2.53.1.  Can you confirm if you downloaded the source code folder for 2.53.1 from the link below and built from source with CMake on your Raspberry Pi using that folder, please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1

Otherwise, the latest version of librealsense would be built on your Pi.  The network device support was removed from librealsense in version 2.54.1.

kacper-olszewski, 2024-09-19:
Thank you this solved my issue

MartyG-RealSense, 2024-09-19:
You are very welcome, @kacper-olszewski - I'm pleased that I could help.  As you have achieved a solution, I will close this case.  Thanks very much for the update!

closed 2024-09-19 by MartyG-RealSense