realsenseai/librealsense #13351: Do I need to do a distortion correction after getting an aligned frame?

liuchangji, 2024-09-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515/ D435i } | 
| Firmware Version                   | (2.55.1) | 
| Operating System & Version |   Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  5.15.0-107-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            | pyrealsense2   2.48.0.3381  | 
| Language                            |  python | 
| Segment			|  Robot   | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I have one L515 and one D435i and I followed the [code example](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py) to get an aligned frame, the minimum code example is below:

`

       aligned_frames = align.process(frames)
        # Get aligned frames
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = aligned_frames.get_color_frame()
`

So my question is 
1. Do I need to do a distortion correction to the aligned color frame?
2. If I need to do a distortion correction to the aligned color frame, I also need to a distortion correction to the aligned depth map. This is an odd operation.
3. Is it the same in L515 and D435i. The coeffs I got from L515 are [0.16058915853500366, -0.5034521818161011, -0.0007438811007887125, -0.0008255860302597284, 0.45545297861099243].  But coeffs I got from D435i are [0.0, 0.0, 0.0, 0.0, 0.0]?

I have already searched the issues and  document but I didn't find a direct answer from the official source.

MartyG-RealSense, 2024-09-18:
Hi @liuchangji  thanks very much for your questions.

1 and 2. If you are using **align.process()** then you do not need to perform a distortion correction.  The alignment calculations are performed automatically by the librealsense SDK's **align processing block** component.

3.  In the early models in the RealSense 400 Series range such as D435i, all coefficients were set to zero - as described at - https://github.com/IntelRealSense/librealsense/issues/1430 - whilst in later RealSense 400 Series models and L515, coefficients can be non-zero.  So your coefficient results are normal.

liuchangji, 2024-09-18:
@MartyG-RealSense Thank you for your reply!

nbubis, 2024-10-27:
@MartyG-RealSense - Can you please confirm that the distortion coefficients are correct? Irrespective of alignment, if only using the color sensor, applying the intrinsic distortions via opencv (for both the L515 and zeros given by the D435) will give an undistorted image? or is additional calibration needed to find it?

MartyG-RealSense, 2024-10-27:
Hi @nbubis  Even though some camera models such as D435 have their distortion coefficients set to zero, the depth, color and infrared streams (except for Y16 format) do have a distortion model applied to them. 

The RGB color stream has an **Inverse Brown-Conrady** distortion model applied to it.

[image]

Y16 format is not distorted because it is used for camera calibration.

If you want to undistort in OpenCV a RealSense image that has had a distortion model applied to it then OpenCV's **undistort()** function is likely to be the best way to do it.

nbubis, 2024-10-28:
I'm not sure I understand. On the D435, the following appears:

```
 Intrinsic of "Color" / 1280x720 / {YUYV/RGB8/BGR8/RGBA8/BGRA8/Y8/Y16}
  Width:      	1280
  Height:     	720
  PPX:        	649.18603515625
  PPY:        	357.792572021484
  Fx:         	914.134887695312
  Fy:         	912.908752441406
  Distortion: 	Inverse Brown Conrady
  Coeffs:     	0  	0  	0  	0  	0  
  FOV (deg):  	69.99 x 43.04
```
Does this mean all formats (including Y16) are fully rectified, as in if I use fx, fy, ppx, ppy without undistortion, all lines should be straight? If so, why would I need to apply the undistort() function? Also, for the L515 does the opencv undistort accept the "Brown Conrady" parameters, or is it expecting "Inverse Brown Conrady"?

MartyG-RealSense, 2024-10-28:
Y16 is definitely unrectified.  It is one of the few formats that does not have a distortion model applied to it.  Almost all of the RealSense stream formats that are accessible by users are rectified.

There is not an explanation available regarding how rectification changes a raw stream, so I cannot offer advice about your question regarding lines, unfortunately.

https://github.com/IntelRealSense/librealsense/issues/3880 is the only reference available about using OpenCV's undistort() on a RealSense stream.

The discussion at the link below may also be helpful to you.

https://support.intelrealsense.com/hc/en-us/community/posts/21654882261779/comments/34220325970835

closed 2024-11-08 by liuchangji