realsenseai/librealsense #13355: pipeline.start(config) RuntimeError: Couldn't resolve requests when increasing depth stream resolution to 1920x1080

monajalal, 2024-09-18:
When I change the resolution of depth camera, I get this error, however, I need the resolution of my depth camera to match resolution of my color camera. Could you please provide a fix?

```
# Get device product line for setting a supporting resolution
pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))
 
found_rgb = False
for s in device.sensors:
    if s.get_info(rs.camera_info.name) == 'RGB Camera':
        found_rgb = True
        break
if not found_rgb:
    print("The demo requires Depth camera with Color sensor")
    exit(0)
 
config.enable_stream(rs.stream.depth, 1920, 1080, rs.format.z16, 30)
 
config.enable_stream(rs.stream.color, 1920, 1080, rs.format.bgr8, 30)
# Start streaming
pipeline.start(config)
 
 
 
    pipeline.start(config)
RuntimeError: Couldn't resolve requests
```

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-09-18:
Hi @monajalal  RealSense cameras do not support a depth resolution of 1920x1080 - it is only supported for RGB.  This is why the error **RuntimeError: Couldn't resolve requests** occurs, because the requested depth resolution cannot be provided.  The maximum depth resolution available is 1280x720.  So there is no solution available that will provide 1920x1080 depth, unfortunately.  

Is it possible for you to use 1280x720 for RGB, please?

monajalal, 2024-09-18:
That works but for my application I need 1920x1080 for color (I only need to match depth so that I know if a person is far away or too close).

monajalal, 2024-09-18:
@MartyG-RealSense I am not sure what that means, can you please show an example?

I need to show the color image to user. I only need the depth feed to tell how far the person is

MartyG-RealSense, 2024-09-18:
Here are some example programs demonstrating depth to color alignment.

**C++**
https://github.com/IntelRealSense/librealsense/tree/master/examples/align

**Python**
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

If you installed the librealsense SDK on Linux with the example programs included in the installation then you should be able to find a pre-built executable version of rs-align in the folder **usr/local/bin**

closed 2024-10-03 by MartyG-RealSense