realsenseai/librealsense #13357: depth confidence frame/map in D435

monajalal, 2024-09-18:
```
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)

config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)

config.enable_stream(rs.stream.confidence, 1280, 720, rs.format.raw8, 30)
# Start streaming
pipeline.start(config)
```

Error:

```
    pipeline.start(config)
RuntimeError: Couldn't resolve requests
```

Related link:

https://github.com/IntelRealSense/librealsense/issues/9037

Could you please help how I can get the confidence value for each pixel in depth map for the resolution of the 1280x720 for depth map?

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-09-18:
Hi @monajalal  1280x720 for both depth and color should be supported on a D435, unless your camera is being detected as being on a USB 2.1 connection instead of USB 3.2.  On a USB 2.1 connection, 1280x720 resolution is only supported at 6 FPS instead of 30 and requesting 30 FPS would cause the **RuntimeError: Couldn't resolve requests** error.

RealSense 400 Series cameras do not have a confidence map but you can do something similar with the Advanced Mode **Second Peak Threshold** parameter, as advised at https://github.com/IntelRealSense/librealsense/issues/3185#issuecomment-460540961

monajalal, 2024-09-18:
Thanks for mentioning this. Turns out I cannot use rs.stream.confidence since this is D400 series, right?

As for the other option, how can I set these in the python code? I can access them in realsense-viewer GUI. 

My intention is when a person is detected by GUI and is further than 1.5m, do not show the bounding box surrounding it

[image]

MartyG-RealSense, 2024-09-19:
The confidence stream is a feature only supported on the RealSense T265 Tracking Camera.

Tthe Second Peak Threshold value can be defined in a custom json camera configuration file or with Python using Advanced Mode scripting, as described at https://github.com/IntelRealSense/librealsense/issues/10224#issuecomment-1060412603

Using Advanced Mode code to set it will be complicated.  It would be much easier to define the value in a json file and then import the json into your program so that the configuration settings stored in the file are applied.

closed 2024-10-03 by MartyG-RealSense