realsenseai/librealsense #13358: Frames didn't arrived within 5 seconds in D435

yedouu, 2024-09-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   ubuntu 20.04.6 LTS | 
| Kernel Version (Linux Only)    |           Linux 5.15.0-122-generic           | 
| Platform                                 | mini PC (13th Gen Intel® Core™ i7-13620H × 16 ) |
| SDK Version                            |   2.55.1                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
when i first time to try to D435,install SDK and librealsense-ros and D435 work successfully.
then i try the same method to install it in my mini PC,at first the realsense-viewer always tip "Frames didn't arrived within 5 seconds".
then i uninstall it use
```dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge```
and install use
```git clone https://github.com/IntelRealSense/librealsense```
``` cd librealsense```
```sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/```
```sudo udevadm control --reload-rules && udevadm trigger```
```
mkdir build
cd build
cmake ../ -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true
sudo make uninstall
make clean && make -j8
sudo make install
```
and test it in realsense-viewer and ros,but it still tips "Frames didn't arrived within 5 seconds"
```
uav@uav-M9-PRO:~/realsense/ws_ros$ roslaunch realsense2_camera rs_rgbd.launch  
... logging to /home/uav/.ros/log/a985c2ec-7646-11ef-931e-8150280cc979/roslaunch-uav-M9-PRO-45102.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://uav-M9-PRO:39681/

SUMMARY
========

PARAMETERS
 * /camera/color_rectify_color/queue_size: 5
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
[... 117 lines cut ...]
[ INFO] [1726723134.292091138]: Sync Mode: On
[ INFO] [1726723134.292182720]: Device Sensors: 
[ INFO] [1726723134.293837277]: Stereo Module was found.
[ INFO] [1726723134.295110185]: RGB Camera was found.
[ INFO] [1726723134.295148122]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1726723134.295250607]: num_filters: 1
[ INFO] [1726723134.295273781]: Setting Dynamic reconfig parameters.
[ INFO] [1726723134.352942428]: Done Setting Dynamic reconfig parameters.
[ INFO] [1726723134.353715876]: depth stream is enabled - width: 640, height: 480, fps: 15, Format: Z16
[ INFO] [1726723134.354371942]: color stream is enabled - width: 640, height: 480, fps: 15, Format: RGB8
[ INFO] [1726723134.356019957]: setupPublishers...
[ INFO] [1726723134.358360541]: Expected frequency for depth = 15.00000
[ INFO] [1726723134.385548934]: Expected frequency for color = 15.00000
[ INFO] [1726723134.407496084]: Expected frequency for aligned_depth_to_color = 15.00000
[ INFO] [1726723134.425364291]: setupStreams...
[ INFO] [1726723134.434779225]: SELECTED BASE:Depth, 0
[ INFO] [1726723134.443720614]: RealSense Node Is Up!
 19/09 13:18:59,435 WARNING [139838632228608] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
 19/09 13:18:59,439 WARNING [139838623835904] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
 19/09 13:19:04,439 WARNING [139838632228608] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
 19/09 13:19:04,443 WARNING [139838623835904] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
 19/09 13:19:09,443 WARNING [139838632228608] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
 19/09 13:19:09,447 WARNING [139838623835904] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
```
how can i work out it?

MartyG-RealSense, 2024-09-19:
Hi @yedouu  I would first recommend going back to your librealsense source code folder's Build directory and inputting the command below whilst in that directory to uninstall your current source code build of the SDK.

**sudo make uninstall && make clean**

*******

Next, uninstall the ROS1 2.3.2 wrapper.  If you built it from packages then running the 'purge' command again should do that.  If you built it from source code then simply delete the entire **/catkin_ws/src** catkin workspace folder to remove the wrapper.

**********

Reinstall the librealsense SDK from source code using the **libuvc backend** method, which has a pre-made build script called **libuvc_installation.sh** at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

If you need to use the realsense-viewer tool then edit line 46 of the libuvc_installation.sh script.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=TRUE -DBUILD_GRAPHICAL_EXAMPLES=TRUE`

*******

Finally, build the ROS1 2.3.2 wrapper again from source code.

yedouu, 2024-09-20:
thanks for your help!
i try the method you provide,uninstall the SDK and use the libuvc_installation.sh to install SDK ,then i use it in ROS and open the rqt_image_view,

yedouu, 2024-09-20:
Thank you for your suggestion. I tried deleting the original SDK and installing a new SDK using the script you mentioned. It has been proven to have some effect, but not much. After starting the ROS node, I tried to use rqt_image_view to view the image and found that only in the first few seconds could I see the image, and the screen froze and no longer updated. In addition, I opened the readme file of the ROS package and found this sentence,I don't know if it's a version issue that affects the camera's operation (but this is not a problem on another computer)：
```
LibRealSense2 supported version: v2.50.0 (see [realsense2_camera release notes](https://github.com/IntelRealSense/realsense-ros/releases))
```

My ROS node startup process is as follows:
```
uav@uav-M9-PRO:~/realsense/ws_ros$ roslaunch realsense2_camera rs_rgbd.launch 
... logging to /home/uav/.ros/log/33d4a262-772b-11ef-bdb2-69e3c7ac0f03/roslaunch-uav-M9-PRO-5297.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://uav-M9-PRO:32787/

SUMMARY
========

PARAMETERS
 * /camera/color_rectify_color/queue_size: 5
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
[... 169 lines cut ...]
 20/09 16:34:57,637 ERROR [139862159169280] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 130
 20/09 16:34:58,027 WARNING [139862951196416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 20/09 16:34:58,038 WARNING [139862951196416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 20/09 16:34:58,049 WARNING [139862951196416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 20/09 16:34:58,060 WARNING [139862951196416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 20/09 16:34:58,071 WARNING [139862951196416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 20/09 16:34:59,084 WARNING [139862951196416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 20/09 16:34:59,096 WARNING [139862951196416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 20/09 16:34:59,106 WARNING [139862951196416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[... cut ...]
 20/09 16:34:59,149 WARNING [139862951196416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 20/09 16:35:00,302 ERROR [139862159169280] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 130
^C[camera/points_xyzrgb_hw_registered-5] killing on exit
[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[camera/color_rectify_color-4] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

MartyG-RealSense, 2024-09-20:
Ideally, SDK 2.50.0 and firmware 5.13.0.50 should be used with the 2.3.2 ROS1 wrapper, though the configuration that you are using (librealsense 2.55.1 and firmware 5.16.0.1) does work for some people even though it should not.

The  **control_transfer returned** warning, when repeating continuously, indicates a communication problem between the camera and computer, typically a USB related issue with the USB cable or USB port.  I note that the launch is detecting your camera as being on a slow-speed USB 2.1 connection instead of USB 3.

Is there any improvement if you reset the camera at launch by adding **initial_reset:=true** to your roslaunch instruction?

yedouu, 2024-09-21:
After trying the method you mentioned and adding the restart code, the camera still worked for one or two seconds before getting stuck again, just like before; Regarding the issue with USB, I am using a USB port that should be 3.0, and I think it may be a problem with the connection cable.
Here are the results of my run and pictures of the connecting wires:
```
uav@uav-M9-PRO:~/realsense/ws_ros$ roslaunch realsense2_camera rs_rgbd.launch initial_reset:=true
... logging to /home/uav/.ros/log/cffe8934-77e2-11ef-9cd6-2920681f627e/roslaunch-uav-M9-PRO-7952.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://uav-M9-PRO:35539/

SUMMARY
========

PARAMETERS
 * /camera/color_rectify_color/queue_size: 5
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
[... 170 lines cut ...]
 21/09 14:29:16,888 WARNING [140580217759488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/09 14:29:16,900 WARNING [140580217759488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/09 14:29:16,911 WARNING [140580217759488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/09 14:29:16,921 WARNING [140580217759488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/09 14:29:16,932 WARNING [140580217759488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/09 14:29:16,943 WARNING [140580217759488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/09 14:29:16,954 WARNING [140580217759488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/09 14:29:16,964 WARNING [140580217759488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/09 14:29:16,975 WARNING [140580217759488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 21/09 14:29:20,015 WARNING [140580217759488] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
^C[camera/points_xyzrgb_hw_registered-5] killing on exit
[camera/color_rectify_color-4] killing on exit
[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
[image]

MartyG-RealSense, 2024-09-21:
Have you installed support for using the rs_rgbd launch file with Noetic before using the launch file please?  Support for this launch file is not included in the ROS wrapper by default and needs to be first installed with the command below.

`sudo apt-get install ros-noetic-rgbd-launch`

yedouu, 2024-09-22:
Thank you for your suggestion. I have installed this software package, but it didn't work. However, I bought a new USB 3.0 cable yesterday and found that the camera can work completely normally. Thank you for your advice these past few days!

MartyG-RealSense, 2024-09-22:
You are very welcome.  It's great to hear that a new cable resolved your issue.  Thanks very much for the update!

closed 2025-01-06 by MartyG-RealSense