realsenseai/librealsense #13362: v2.5.6.1 Beta Python Install Error

stu68, 2024-09-19:
I am getting the following when trying to install the latest beta release into python 3.12/ PyCharm:

Processing c:\users\admin\documents\github\librealsense-development\wrappers\python
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [22 lines of output]
      Traceback (most recent call last):
        File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Admin\AppData\Local\Temp\pip-build-env-kikdptcn\overlay\Lib\site-packages\setuptools\build_meta.py", line 332, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Admin\AppData\Local\Temp\pip-build-env-kikdptcn\overlay\Lib\site-packages\setuptools\build_meta.py", line 302, in _get_build_requires
          self.run_setup()
        File "C:\Users\Admin\AppData\Local\Temp\pip-build-env-kikdptcn\overlay\Lib\site-packages\setuptools\build_meta.py", line 503, in run_setup
          super().run_setup(setup_script=setup_script)
        File "C:\Users\Admin\AppData\Local\Temp\pip-build-env-kikdptcn\overlay\Lib\site-packages\setuptools\build_meta.py", line 318, in run_setup
          exec(code, locals())
        File "<string>", line 5, in <module>
        File "C:\Users\Admin\Documents\GitHub\librealsense-development\wrappers\python\pyrealsense2\__init__.py", line 2, in <module>
          from .pyrealsense2 import *
      ModuleNotFoundError: No module named 'pyrealsense2.pyrealsense2'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

MartyG-RealSense, 2024-09-19:
Hi @stu68  The interim beta release of 2.56.1 does not have Python wheels published for this version.

[image]

MartyG-RealSense, 2024-09-26:
Hi @stu68  Do you require further assistance with this case, please?  Thanks!

stu68, 2024-09-28:
Is there an ETA for Python 3.12 compatibility. I purchased a realsense camera and have been unable to use it.

Thank you,
Stu

MartyG-RealSense, 2024-09-28:
2.56.1 is already compatible with Python 3.12.  At the moment though, the pyrealsense2 wrapper needs to be built from source code instead of using PyPi (pip install pyrealsense2) when installing the wrapper with 2.56.1.

If you need to use a whl package and there are not packages available for a particular librealsense version then https://github.com/IntelRealSense/librealsense/issues/12601#issuecomment-1905554344 describes a workaround method provided by my Intel RealSense colleagues for creating your own whl.

stu68, 2024-10-08:
I tried all the approaches/ procedures I could find, but none of them seem to work. I can get it built from the source code but not installed. Is there a complete, updated procedure for that?

Thank you,
Stu

If you need to use a whl package and there are not packages available for a particular librealsense version then #12601 (comment)<https://github.com/IntelRealSense/librealsense/issues/12601#issuecomment-1905554344> describes a workaround method provided by my Intel RealSense colleagues for creating your own whl.

MartyG-RealSense, 2024-10-09:
Official .whl pyrealsense2 package files for use with librealsense 2.56.1 are not currently available, so source code installation of pyrealsense2 is currently the only option.

The most straightforward method of installing librealsense + pyrealsense2 together from source code may be to use the **libuvc backend** procedure at the link below, which has a build script called 'libuvc_installation.sh'.  This will install librealsense 2.55.1 instead of 2.56.1, but it will not matter.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

To install the pyrealsense2 wrapper at the samw time as librealsense, line 46 of the build script should be modified so that it uses the line below:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_PYTHON_BINDINGS:bool=true`

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

Because pyrealsense2 is being built from source code, it should work with Python 3.12.

MartyG-RealSense, 2024-11-04:
Case closed due to no further comments received.

stu68, 2025-02-13:
Just getting back to this. Please let me know if I should start a new issue. I was able to get the package installed with 2.5.3b and it shows up. However, the import statement still says no module found.

[image]

MartyG-RealSense, 2025-02-14:
It's fine to continue with the existing issue.  I have reopened it.

I note that you are using librealsense 2.55.1 and Python 3.13.  Support for Python 3.13 was only added 5 days ago to the **development** branch of librealsense, as described at https://github.com/IntelRealSense/librealsense/pull/13747#event-16238171356

https://github.com/IntelRealSense/librealsense/tree/development

However, the problem may be related to your PYTHONPATH parameter.  Please try inputting the command below into the terminal before launching your pyrealsense2 script.

`export PYTHONPATH=$PYTHONPATH:/usr/local/lib`

stu68, 2025-02-15:
I'm not sure if I mentioned it but this is on a Windows PC

stu68, 2025-02-15:
I'm getting this when I try to install 2.55.1:

Defaulting to user installation because normal site-packages is not writeable
Processing c:\users\admin\documents\github\asap_pupil_detector\librealsense\wrappers\python
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'

  error: subprocess-exited-with-error
  
  python setup.py egg_info did not run successfully.
  exit code: 1
  
  [17 lines of output]
  Traceback (most recent call last):
    File "<string>", line 2, in <module>
      exec(compile('''
      ~~~~^^^^^^^^^^^^
      # This is <pip-setuptools-caller> -- a caller that pip uses to run setup.py
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      ...<31 lines>...
      exec(compile(setup_py_code, filename, "exec"))
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      ''' % ('C:\\Users\\Admin\\Documents\\GitHub\\ASAP_Pupil_Detector\\librealsense\\wrappers\\python\\setup.py',), "<pip-setuptools-caller>", "exec"))
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "<pip-setuptools-caller>", line 34, in <module>
    File "C:\Users\Admin\Documents\GitHub\ASAP_Pupil_Detector\librealsense\wrappers\python\setup.py", line 5, in <module>
      from pyrealsense2._version import __version__
    File "C:\Users\Admin\Documents\GitHub\ASAP_Pupil_Detector\librealsense\wrappers\python\pyrealsense2\__init__.py", line 2, in <module>
      from .pyrealsense2 import *
  ModuleNotFoundError: No module named 'pyrealsense2.pyrealsense2'
  [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

Encountered error while generating package metadata.

See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

MartyG-RealSense, 2025-02-16:
If you can use pip on your Windows PC then all that should be necessary to do in order to install pyrealsense2 is to input the 'pip install pyrealsense2' command.

It is not necessary to run the 'setup.py' file, as this is just used by the RealSense team to create the pyrealsense2 pip package.  But when using 2.55.1 the package is already pre-built by the team and so you only need to use 'pip install pyrealsense2' to install from the 2.55.1 version of the pip package for Windows.

At https://github.com/IntelRealSense/librealsense/issues/6402#issuecomment-629763906 I described how to install the pip tool on Windows if it is not installed and working already.

stu68, 2025-02-16:
This is what I get when I try to install via pip:
PS C:\Users\Admin\Documents\GitHub\ASAP_Pupil_Detector> pip install pyrealsense2
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2
PS C:\Users\Admin\Documents\GitHub\ASAP_Pupil_Detector>

MartyG-RealSense, 2025-02-16:
Which Python 3 version are you using, please?  Pip install of pyrealsense2 in librealsense 2.55.1 on Windows supports Python 3.7 to 3.11.  If you have a newer version of Python 3 than 3.11 then the **No matching distribution found for pyrealsense2** error message will occur and pyrealsense2 wrapper needs to be compiled from source code instead.

stu68, 2025-02-19:
I was able to get it working by rolling back to 3.11. Thank you!

MartyG-RealSense, 2025-02-20:
That's great to hear, @stu68 - thanks very much for the update about your success!

closed 2025-02-28 by MartyG-RealSense